esp8266/Arduino #6376: Migrate all scripts, symlinks, and installed tools to Python3

earlephilhower, 2019-08-02:
Python 2 has less than 5 months left to live (or at least, until all support ends)
https://pythonclock.org/

#6364 had an issue with a symlink to Python 2 being installed in the tools directory, which is one part but not all.

We need to:
. Update all script refs from `/usr/bin/python` to `/usr/bin/python3`
. Update the tools python-wrapper.tgz to point to python3
. Update the Windows standalone installation to python3
. Update platform.txt/etc. to ensure it's using the right python verison
. Validate things still work.

It's not as simple as changing scripts to `#/usr/bin/env python`, either, as that only works if there is a Python2 installation somewhere.  Newer OS releases may not install Python2 by default.

TD-er, 2019-08-03:
And what about using virtualenv as wrapper?
Then you can also have a very well contained set of included libraries.
Not sure how easy virtualenv is setup in all OS'es (e.g. Windows), but on Linux I use it all the time.

This also allows to create a stand-alone Python 2.7 environment if conversion takes more time than expected.

earlephilhower, 2019-08-03:
I think the point is that Python 2.7 is dead as of January 1, 2020, so virtualenv's not really a solution.  Without the python 2 executable, the virtualenv would run under the Python 3 interpreter, in fact.  I don't think there's any issue with scripts we're using, they should already run fine under Py3.

The standalone Python we're using for Windows is kind of a virtualenv.  We install the needed libraries right in the core directories, and it only uses PYCs from inside the installation dir, not system-wide ones (assuming there's some system-wide Python installed on the user's Windows).

commit 0a031ce9 (2019-08-28): Move all scripts and documentation to Python3 (#6378)

* Move all scripts and documentation to Python3

Python 2 EOL is Jan 1, 2020.  Migrate scripts to run under Python 3.

Under Windows, we're already running Python 3.7, by dumb luck.  The
oddness is that the Windows standalone executable for Python 3 is called
"python" whereas under UNIX-like OSes it's called "python3" with
"python" always referring to the Python 2 executable.  The ZIP needs to
be updated to include a Python3.exe (copy of Python.exe) so that we can
use the same command lines under Linux and Windows, and to preserve my
sanit [... cut]