esp8266/Arduino #6387: Setting HTTPCLIENT_1_1_COMPATIBLE to 0 and using HTTPS client causes Exception 9

mmmilan, 2019-08-06:
- Hardware:			ESP-8266
- Core Version:		2.5.2
- Development Env:	Arduino IDE 1.8.9
- Operating System:	Windows
- Module:  			Lolin(Wemos) D1 mini Pro
(all remaining settings are the defaults for the board)

Issuing a HTTPS request causes Exception 9.

```
#define HTTPCLIENT_1_1_COMPATIBLE 0

#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "mobiledos";
const char* wifiPassword = "mobiledos1";

void setup()
{
  Serial.begin(115200);
  Serial.println();
  while (Serial.available())
    Serial.read();
  // wifi
  Serial.println(F("connecting wifi"));
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, wifiPassword);
  while (!WiFi.isConnected())
    delay(0);
  Serial.println(F("wifi connected"));
}

void loop()
{
  Serial.println(F("begin loop"));
  if (WiFi.isConnected())
  {
    BearSSL::WiFiClientSecure wifiClient;
    wifiClient.setInsecure();
    HTTPClient http;
    String url = F("https://graph.facebook.com/v3.3/me/messages?access_token=noidfortesting");
    if (http.begin(wifiClient, url))
    {
      http.addHeader(F("Content-Type"), F("application/json"));
      String recipient = F("12345678");
      String entry = F("testmessage");
      String data = String(F("{\"recipient\": { \"id\": \"")) + recipient + F("\" }, \"message\": { \"text\": \"") + entry + F("\" }}");

      int httpCode = http.POST(data);
      if (httpCode != 200)
        Serial.println(String(F("messenger post http code ")) + httpCode);
      if (httpCode >= 0)
      {
        int sent = http.writeToStream(&Serial);
        if (sent < 10)
          Serial.println(String(F("messenger writeToStream error ")) + sent);
      }
    }
    http.end();
  }
  Serial.println(F("end loop"));
}
```

The output is:
```
connecting wifi
wifi connected
begin loop

Exception (9):
epc1=0x40204fd6 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0047004f depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffba0 end: 3fffffc0 offset: 01a0
3ffffd40:  3ffe85b0 00000000 3ffffecc 4020501c  
3ffffd50:  00000000 3ffffdc0 3ffffecc 40205059  
3ffffd60:  00000005 00000000 3ffffdcc 3fffff4c  
3ffffd70:  3ffef78c 00000047 3ffffdc0 40205e32  
3ffffd80:  3ffe85b0 00000000 3ffffdc0 40205e90  
3ffffd90:  00000000 3ffef78c 3ffffdc0 40206050  
3ffffda0:  3ffe860a 3ffffecc 3ffffdcc 3fffff4c  
3ffffdb0:  00000000 3ffffecc 3ffffdcc 40205269  
3ffffdc0:  3ffef934 0047004f ffffff4c 70747468  
3ffffdd0:  00000073 00fef734 3fffff4c 40206050  
3ffffde0:  00000000 00000000 3ffee90c 3ffee984  
3ffffdf0:  3fffdad0 00000000 3ffee90c 40201149  
3ffffe00:  402090f8 00000000 00003a98 00000001  
3ffffe10:  00000000 00000000 00000000 00000000  
3ffffe20:  00000000 00000000 00000000 00000000  
3ffffe30:  00000000 00000000 00000000 00000000  
3ffffe40:  00000000 00000000 00000000 00000000  
3ffffe50:  00000000 00000000 00000000 00000000  
3ffffe60:  00004145 00000255 00000000 00000000  
3ffffe70:  00000000 00000000 00000000 00000000  
3ffffe80:  00000000 00000000 00000001 00000000  
3ffffe90:  00000077 0000007e 3ffee984 00000030  
3ffffea0:  00000000 00000000 00000000 00000000  
3ffffeb0:  3ffee800 00000000 00000000 00000030  
3ffffec0:  00000000 00000000 3ffee984 00000000  
3ffffed0:  00000000 3ffffe00 00000000 00000000  
3ffffee0:  00fee8d8 000001bb 3f001388 00000000  
3ffffef0:  00000000 00509ffe 00000000 00000000  
3fffff00:  00626f6d 00000000 00000000 00000000  
3fffff10:  3ffef89c 0011001f fffef6b4 00000000  
3fffff20:  00000000 00000000 00000000 00000000  
3fffff30:  00000000 ffffffff 00000000 3ffe000a  
3fffff40:  00000000 00000000 00fee90c 3ffef78c  
3fffff50:  00000000 ff000064 3ffef57c 40205875  
3fffff60:  3ffe85ae 3ffee90c 0000000d 402055bb  
3fffff70:  3ffef05c 402015e8 3ffee90c 402058f0  
3fffff80:  00000000 3ffee8d8 3ffef734 0047004f  
3fffff90:  ff000000 3ffee8d8 3ffee90c 402010a8  
3fffffa0:  3fffdad0 00000000 3ffee954 40206e2c  
3fffffb0:  feefeffe feefeffe 3ffe8510 40100461  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld
```

And the exception decoded:

```
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
PC: 0x40204fd6: std::unique_ptr   >::reset(StreamString*) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/StreamString.h line 27
EXCVADDR: 0x0047004f

Decoding stack results
0x4020501c: HTTPClient::clear() at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 142
0x40205059: HTTPClient::beginInternal(String, char const*) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 280
0x40205e32: String::reserve(unsigned int) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 148
0x40205e90: String::copy(char const*, unsigned int) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 211
0x40206050: String::operator=(String const&) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 259
0x40205269: HTTPClient::begin(WiFiClient&, String) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266HTTPClient\src\ESP8266HTTPClient.cpp line 178
0x40206050: String::operator=(String const&) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\WString.cpp line 259
0x40201149: loop() at C:\project\arduino\HTTPSClientTest/HTTPSClientTest.ino line 33
0x40205875: Print::write(char const*) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Print.h line 60
0x402055bb: HardwareSerial::write(unsigned char) at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/HardwareSerial.h line 154
0x402015e8: ESP8266WiFiSTAClass::status() at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x402058f0: Print::println() at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\Print.cpp line 178
0x402010a8: setup() at C:\project\arduino\HTTPSClientTest/HTTPSClientTest.ino line 22
0x40206e2c: loop_wrapper() at C:\Users\mmilan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 125
```

d-a-v, 2019-08-06:
You cannot set this defines in sketch.
The worst way is to modify `ESP8266HTTPClient.h`
If you use Arduino IDE you can create a file named `platform.local.txt` next to `platform.txt` (search for it) and put inside:
```
compiler.c.extra_flags=-DSTASSID="yourwifiap" -DSTAPSK="yourwifipassword" -DHTTPCLIENT_1_1_COMPATIBLE=0
compiler.cpp.extra_flags={compiler.c.extra_flags}
```

d-a-v, 2019-08-06:
With this define, you can also try to
```
    http.useHTTP10(true);
```

mmmilan, 2019-08-06:
Adding the define through platform.local.txt worked. Thank you!

closed 2019-08-06 by mmmilan