esp8266/Arduino #6397: Crash with basic ciphers chosen

CyanoFresh, 2019-08-08:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Wemos D1 mini
- Core Version:			https://github.com/esp8266/Arduino/commit/a8873c2364612cd1f8225cd038cc6d58a482bfd8
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			[qio|dio|other]?
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory || Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	[40Mhz]?
- CPU Frequency:		80Mhz || 160MHz
- Upload Using:		SERIAL
- Upload Speed:   921600 (serial upload only)

### Problem Description

When `Basic ciphers` selected in HelloServerBearSSL.ino example, crashes on every request

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
  HelloServerBearSSL - Simple HTTPS server example

  This example demonstrates a basic ESP8266WebServerSecure HTTPS server
  that can serve "/" and "/inline" and generate detailed 404 (not found)
  HTTP respoinses.  Be sure to update the SSID and PASSWORD before running
  to allow connection to your WiFi network.

  Adapted by Earle F. Philhower, III, from the HelloServer.ino example.
  This example is released into the public domain.
*/
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServerSecure.h>

#ifndef STASSID
#define STASSID "HUAWEI-D190"
#define STAPSK  "12307201"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

BearSSL::ESP8266WebServerSecure server(443);

static const char serverCert[] PROGMEM = R"EOF(
-----BEGIN CERTIFICATE-----
MIIDSzCCAjMCCQD2ahcfZAwXxDANBgkqhkiG9w0BAQsFADCBiTELMAkGA1UEBhMC
VVMxEzARBgNVBAgMCkNhbGlmb3JuaWExFjAUBgNVBAcMDU9yYW5nZSBDb3VudHkx
EDAOBgNVBAoMB1ByaXZhZG8xGjAYBgNVBAMMEXNlcnZlci56bGFiZWwuY29tMR8w
HQYJKoZIhvcNAQkBFhBlYXJsZUB6bGFiZWwuY29tMB4XDTE4MDMwNjA1NDg0NFoX
DTE5MDMwNjA1NDg0NFowRTELMAkGA1UEBhMCQVUxEzARBgNVBAgMClNvbWUtU3Rh
dGUxITAfBgNVBAoMGEludGVybmV0IFdpZGdpdHMgUHR5IEx0ZDCCASIwDQYJKoZI
hvcNAQEBBQADggEPADCCAQoCggEBAPVKBwbZ+KDSl40YCDkP6y8Sv4iNGvEOZg8Y
[... 125 lines cut ...]
0x40229968: br_ssl_hs_server_run at src/ssl/ssl_hs_server.c line 1735
0x4022a506: sendpld_flush at src/ssl/ssl_engine.c line 851
0x40221bcc: sendpld_ack at src/ssl/ssl_engine.c line 878
0x4022a524: clear_encrypt at src/ssl/ssl_engine.c line 951
0x40221cd0: jump_handshake at src/ssl/ssl_engine.c line 1081
0x402221b5: br_ssl_engine_recvrec_ack at src/ssl/ssl_engine.c line 1206
0x40207579: BearSSL::WiFiClientSecure::_run_until(unsigned int, bool) (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 523
0x40228994: br_ssl_hs_server_run at src/ssl/ssl_hs_server.c line 943
0x40228940: br_ssl_hs_server_init_main at /home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/sys/pgmspace.h line 77
0x4022126c: br_rsa_i15_private at src/rsa/rsa_i15_priv.c line 33
0x402077a8: BearSSL::WiFiClientSecure::_wait_for_handshake() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 547
0x40207c9f: BearSSL::WiFiClientSecure::_connectSSLServerRSA(BearSSL::X509List const*, BearSSL::PrivateKey const*, BearSSL::X509List const*) (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 1194
0x402211d8: br_rsa_i15_pkcs1_sign at src/rsa/rsa_i15_pkcs1_sign.c line 32
0x4020ee3a: printf at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/printf.c line 61
0x40207e20: BearSSL::WiFiClientSecure::WiFiClientSecure(ClientContext*, BearSSL::X509List const*, BearSSL::PrivateKey const*, int, int, BearSSL::X509List const*) (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 142
0x4020743c: BearSSL::WiFiClientSecure::~WiFiClientSecure() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 130
0x40208134: BearSSL::WiFiServerSecure::available(unsigned char*) (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiServerSecureBearSSL.cpp line 84
0x4020743c: BearSSL::WiFiClientSecure::~WiFiClientSecure() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 130
0x4020bf8c: WiFiClient::operator bool() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x402038e9: esp8266webserver::ESP8266WebServerTemplate ::handleClient() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 306
0x40203600: esp8266webserver::ESP8266WebServerTemplate ::send(int, char const*, String const&) (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h line 437
0x4020743c: BearSSL::WiFiClientSecure::~WiFiClientSecure() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src\WiFiClientSecureBearSSL.cpp line 130
0x402039fc: loop() at C:\Users\Work\Documents\Arduino\HelloServerBearSSL_dev/HelloServerBearSSL_dev.ino line 142
0x4020a314: loop_wrapper() (C:\Program Files at x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 140
```

earlephilhower, 2019-08-08:
I'm traveling without VPN home, but it looks like it's trying to start up an AES_GCM engine to do something.  But in basic mode, there shouldn't be any need for aes_gcm, per the allowed cipher list, so there are no aes_gcm providers installed (or built into the .BIN to save space).

To let you get on with #6247, you can do a 1-line hack.  Change line 950 in `libraries/ESP8266WiFi/src/WiFiClientSecure.cpp` from `#ifndef BEARSSL_SSL_BASIC` to `#if 1` to install all encryption/decryption (AES is fast, it's not slow like EC) options.

closed 2019-08-09 by earlephilhower

CyanoFresh, 2019-08-09:
Updated to the latest core. Same exmaple on 160MHz and basic ciphers.
Now delay is between 2 and 5 seconds... still long enough (

I am loading page from Chrome

```
00:32:32.435 -> SDK:2.2.2-dev(38a443e)/Core:2.5.2-106-gadfc28d7=20502106/lwIP:STABLE-2_1_2_RELEASE/glue:1.1-8-g2314329/BearSSL:89454af
00:32:32.435 -> scandone
00:32:32.435 -> 
00:32:32.435 -> wifi evt: 2
00:32:32.539 -> scandone
00:32:32.539 -> state: 0 -> 2 (b0)
00:32:32.539 -> state: 2 -> 3 (0)
00:32:32.539 -> state: 3 -> 5 (10)
00:32:32.539 -> add 0
00:32:32.539 -> aid 10
00:32:32.539 -> cnt 
00:32:32.573 -> 
00:32:32.573 -> connected with Solomaha, channel 3
00:32:32.573 -> dhcp client start...
00:32:32.573 -> wifi evt: 0
00:32:32.573 -> ip:192.168.1.178,mask:255.255.255.0,gw:192.168.1.1
00:32:32.608 -> wifi evt: 3
00:32:32.920 -> .
00:32:32.920 -> Connected to Solomaha
00:32:32.920 -> IP address: 192.168.1.178
00:32:32.989 -> HTTPS server started
00:32:36.087 -> WS:ac
00:32:36.087 -> :rn 517
00:32:36.087 -> WS:ac
00:32:36.087 -> :rn 517
00:32:36.087 -> :ref 1
00:32:36.087 -> :rd 5, 517, 0
00:32:36.087 -> :rdi 517, 5
00:32:36.087 -> :rd 512, 517, 5
00:32:36.087 -> :rdi 512, 512
00:32:36.087 -> :c0 512, 517
00:32:36.155 -> :wr 517 0
00:32:36.155 -> :wrc 517 517 0
00:32:36.155 -> :wr 435 0
00:32:36.155 -> :wrc 435 435 0
00:32:36.189 -> :ack 517
00:32:36.224 -> :ack 435
00:32:36.224 -> :rn 7
00:32:36.224 -> :rcl
00:32:36.224 -> :abort
00:32:36.224 -> :rd 5, 7, 0
00:32:36.224 -> :rdi 7, 5
00:32:36.224 -> :rd 2, 7, 5
00:32:36.224 -> :rdi 2, 2
00:32:36.224 -> :c0 2, 7
00:32:36.224 -> BSSL:_wait_for_handshake: failed
00:32:36.224 -> :ur 1
00:32:36.224 -> WS:dis
00:32:36.224 -> :del
00:32:36.224 -> WS:av
00:32:36.224 -> :ref 1
00:32:36.224 -> :rd 5, 517, 0
00:32:36.224 -> :rdi 517, 5
00:32:36.224 -> :rd 512, 517, 5
00:32:36.224 -> :rdi 512, 512
00:32:36.224 -> :c0 512, 517
00:32:36.259 -> :wr 517 0
00:32:36.259 -> :wrc 517 517 0
00:32:36.294 -> :wr 435 0
00:32:36.294 -> :wrc 435 435 0
00:32:36.328 -> :ack 517
00:32:36.363 -> :ack 435
00:32:36.363 -> :rn 7
00:32:36.363 -> :rcl
00:32:36.363 -> :abort
00:32:36.363 -> :rd 5, 7, 0
00:32:36.363 -> :rdi 7, 5
00:32:36.363 -> :rd 2, 7, 5
00:32:36.363 -> :rdi 2, 2
00:32:36.363 -> :c0 2, 7
00:32:36.363 -> WS:ac
00:32:36.363 -> :rn 517
00:32:36.363 -> BSSL:_wait_for_handshake: failed
00:32:36.363 -> :ur 1
00:32:36.363 -> WS:dis
00:32:36.363 -> :del
00:32:36.363 -> WS:av
00:32:36.363 -> :ref 1
00:32:36.363 -> :rd 5, 517, 0
00:32:36.363 -> :rdi 517, 5
00:32:36.363 -> :rd 512, 517, 5
00:32:36.363 -> :rdi 512, 512
00:32:36.363 -> :c0 512, 517
00:32:36.431 -> :wr 517 0
00:32:36.431 -> :wrc 517 517 0
00:32:36.431 -> :wr 435 0
00:32:36.431 -> :wrc 435 435 0
00:32:36.466 -> :ack 517
00:32:36.501 -> :ack 435
00:32:36.501 -> :rn 342
00:32:36.501 -> :rd 5, 342, 0
00:32:36.501 -> :rdi 342, 5
00:32:36.501 -> :rd 262, 342, 5
00:32:36.501 -> :rdi 337, 262
00:32:38.162 -> :rd 5, 342, 267
00:32:38.162 -> :rdi 75, 5
00:32:38.162 -> :rd 1, 342, 272
00:32:38.162 -> :rdi 70, 1
00:32:38.162 -> :rd 5, 342, 273
00:32:38.162 -> :rdi 69, 5
00:32:38.162 -> :rd 64, 342, 278
00:32:38.162 -> :rdi 64, 64
00:32:38.162 -> :c0 64, 342
00:32:38.162 -> :wr 6 0
00:32:38.162 -> :wrc 6 6 0
00:32:38.162 -> :wr 69 0
00:32:38.162 -> :wrc 69 69 0
00:32:38.162 -> WS:av
00:32:38.162 -> :ref 2
00:32:38.162 -> :ur 2
00:32:38.162 -> :ref 2
00:32:38.162 -> :ur 2
00:32:38.162 -> New client
00:32:38.162 -> :ref 2
00:32:38.162 -> :ur 2
00:32:38.196 -> :ack 6
00:32:38.231 -> :ack 69
00:32:38.231 -> :rn 536
00:32:38.231 -> :rch 536, 45
00:32:38.231 -> :rd 5, 581, 0
00:32:38.231 -> :rdi 536, 5
00:32:38.231 -> :rd 576, 581, 5
00:32:38.231 -> :rdi 531, 531
00:32:38.231 -> :c 531, 536, 581
00:32:38.231 -> :rdi 45, 45
00:32:38.231 -> :c0 45, 45
00:32:38.231 -> request: GET / HTTP/1.1
00:32:38.231 -> method: GET url: / search: 
00:32:38.231 -> headerName: Host
00:32:38.231 -> headerValue: 192.168.1.178
00:32:38.231 -> headerName: Connection
00:32:38.231 -> headerValue: keep-alive
00:32:38.231 -> headerName: Cache-Control
00:32:38.231 -> headerValue: max-age=0
00:32:38.231 -> headerName: Upgrade-Insecure-Requests
00:32:38.266 -> headerValue: 1
00:32:38.266 -> headerName: User-Agent
00:32:38.266 -> headerValue: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/76.0.3809.100 Safari/537.36
00:32:38.266 -> headerName: Sec-Fetch-Mode
00:32:38.266 -> headerValue: navigate
00:32:38.266 -> headerName: Sec-Fetch-User
00:32:38.266 -> headerValue: ?1
00:32:38.266 -> headerName: Accept
00:32:38.266 -> headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3
00:32:38.301 -> headerName: Sec-Fetch-Site
00:32:38.301 -> headerValue: none
00:32:38.301 -> headerName: Accept-Encoding
00:32:38.301 -> headerValue: gzip, deflate, br
00:32:38.301 -> headerName: Accept-Language
00:32:38.301 -> headerValue: ru-RU,ru;q=0.9,en-US;q=0.8,en;q=0.7
00:32:38.301 -> args: 
00:32:38.301 -> args count: 0
00:32:38.301 -> args: 
00:32:38.301 -> args count: 0
00:32:38.301 -> Request: /
00:32:38.301 -> Arguments: 
00:32:38.301 -> final list of key/value pairs:
00:32:38.301 -> :wr 133 0
00:32:38.301 -> :wrc 133 133 0
00:32:38.369 -> :ack 133
00:32:38.369 -> :wr 85 0
00:32:38.369 -> :wrc 85 85 0
00:32:38.369 -> :ack 85
00:32:38.369 -> :rcl
00:32:38.369 -> :abort
00:32:38.369 -> :ur 1
00:32:38.369 -> WS:dis
00:32:38.369 -> :del
```

PR #6402 (2019-08-09): Fix basic SSL server definitions
A typo was present in several ifdefs which would allow a server to negotiate
an EC connection even when in basic SSL mode.  When this happened, a crash
would occur (since there were no EC or advanced AES modes installed).

Fix the typo, fixes #6397
commit adfc28d7 (2019-08-09): Fix basic SSL server definitions (#6402)

A typo was present in several ifdefs which would allow a server to negotiate
an EC connection even when in basic SSL mode.  When this happened, a crash
would occur (since there were no EC or advanced AES modes installed).

Fix the typo, fixes #6397