esp8266/Arduino #6404: WPA Enterprise PEAP/TLS Failure

mahendrakumarshinde, 2019-08-09:
HW: ESP8285 
SW:
ESP SDK Version: 3.0.0-dev(c0f7b44)
ESP Boot Version: 31
ESP Core Version: 2_5_0
Other:
Enterprise mode AP and RADIUS server. 

I want to connect my ESP8285 based board to WPA Enterprise mode AP (802.1x EAP).
I have tried with sample application from lib./package to init and connect to AP in EAP-PEAP and EAP-TLS mode.

Observed:
EAP-PEAP connection is successful, got IP, but during mqtt comm. and other operation station(ESP8285) gets disconnected. When tries to reconnect, seeing crash. Exception 9. 
Want to know, what could be reason for crash with exception 9. I know exception 9 is for unaligned access. But my existing code (without EAP related changes) works fine to connect and do mqtt communication based on WPA-PSK mode AP. Only after adding EAP code support and during actual wifi data exchange seeing disconnect or crash.
Checked free heap, but not seeing it running out during actual execution.

For EAP-TLS: Also seeing crash with exception 9, 28. 
When checked with crash dump on arduino... show memory allocation failed. But I dont see any shortage on printing freeheap()..

code for EAP support:
    struct station_config sta_conf = { 0 }; 
    WiFi.mode(WIFI_STA);
 //   WiFi.config(staticIP, gateway, subnet);
    memset(&sta_conf, 0, sizeof(sta_conf));
    strcpy((char*)sta_conf.ssid, SSID);
    strcpy((char*)sta_conf.password, PASSWORD);
    wifi_station_set_config(&sta_conf);

    wifi_station_clear_enterprise_cert_key();
    wifi_station_clear_enterprise_ca_cert();    
    wifi_station_clear_enterprise_identity();
    wifi_station_clear_enterprise_username();
    wifi_station_clear_enterprise_password(); 

    char *username = WPA2_USERNAME;
    char *password = WPA2_PASSWORD;

    wifi_station_set_wpa2_enterprise_auth(1);
    if(wifi_station_set_enterprise_identity((u8 *)"testing", strlen("testing"))) { //This is an option. If not call this API, the outer identity will be "anonymous@espressif.com".
    }
    if (method == EAP_TLS) {
        if(wifi_station_set_enterprise_cert_key(client_cert, strlen((const char *)client_cert)+1, client_key, strlen((const char *)client_key)+1, NULL, 1)) {
        }
        if(wifi_station_set_enterprise_username((u8 *)"testing", strlen("testing"))) {
        }   
    }
    else if (method == EAP_PEAP || method == EAP_TTLS) {
        if(wifi_station_set_enterprise_username((u8 *)"testing", strlen("testing"))) {
        }
        if(wifi_station_set_enterprise_password((u8 *)password, strlen(password))) {
        }
    }

  wifi_station_connect();

  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(F("Free Heap:"));
    Serial.println(system_get_free_heap_size());  
  }
    Serial.print(F("\nESP8285 WiFi Client Connected to "));  
    Serial.println(SSID);
    Serial.print(F("\nIP: "));
    Serial.println(WiFi.localIP());

d-a-v, 2019-08-09:
Try adding `disable_extra4k_at_link_time();` in `setup()`.

Anyway when it does not work, we don't know why.
Ask Espressif.

https://github.com/esp8266/Arduino/search?q=ENTERPRISE&state=open&type=Issues
https://github.com/espressif/ESP8266_NONOS_SDK/search?q=ENTERPRISE&state=open&type=Issues
https://github.com/espressif/ESP8266_RTOS_SDK/search?q=ENTERPRISE&state=open&type=Issues

The most promising is support from RTOS SDK which is the only actively maintained one. Maybe if we remind them this feature request - which they planned (last link above) - they will not forget to propose us examples we can try.

mahendrakumarshinde, 2019-08-09:
Thanks David for quick help. Sure will try it out.

closed 2019-08-09 by devyte