esp8266/Arduino #6406: Cannot upload to ESPresso Lite 1.0 board using 2.5.2

SamuelPittet, 2019-08-09:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			    ESPresso Lite V2.0
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			ESPresso Lite V2.0
- Flash Mode:		N/A
- Flash Size: 		4MB
- lwip Variant:   	v1.4 Lower Higher Bandwidth
- Reset Method:		ck
- Flash Frequency:	40Mhz
- CPU Frequency:	80Mhz
- Upload Using:		SERIAL
- Upload Speed:   	115200

### Problem Description

I cannot upload a sketch to my ESPresso Lite 2.0 board using 2.5.1 or 2.5.2 core version. 2.5.0 works fine. Seems to be related to the new esptool.py. 
You can try the standard Blink sketch. I used the same Settings in the IDE, same board, same USB/TTL converter, same hardware, etc 
for both core versions and anything up to 2.5.0 works fine. 2.5.1 and 2.5.2 doesn't work.

f5soh, 2019-08-09:
Some details with the error ?
Which "Erase flash" option ?

Did you look at https://github.com/esp8266/Arduino/issues/6296 ?

SamuelPittet, 2019-08-12:
@f5soh No details other than it ends with a communication timeout at the end of the upload try. 

The "Erase Flash" option is set at "Only Sketch". So, I'm not sure it is related to #6296 .

mcer12, 2019-08-13:
@SamuelPittet i am 99.9% sure it is caused by #6296. Doesn't matter what erase flash option you choose, esptool is set up so that it restarts the device before upload and that's the issue. Old upload method wiped memory and went on with the upload without reset. 
If you want to use library v2.5.2, all you (should) need to do is to push reset button right after memory wipe finishes.

SamuelPittet, 2019-08-13:
@mcer12 I tried holding the reset button at various moment during the upload, and nothing happen. I joined the output of the Arduino debug window during a failed attempt to upload with 2.5.2.

[output.txt](https://github.com/esp8266/Arduino/files/3497855/output.txt)

earlephilhower, 2019-08-13:
You may have better luck at the upstream esptool.py repo with this issue.  We're just using the latest official uploader from Espressif, without any changes.  They may have some method to make your unique boards work properly:  https://github.com/espressif/esptool

earlephilhower, 2020-01-01:
#6429 changed the reset method to something we believe to be more compatible. Closing this issue, but if you have something similar even with the latest GIT or release, please open a new issue so we can track it.

closed 2020-01-01 by earlephilhower