esp8266/Arduino #6411: PROGMEM issues

mbm60, 2019-08-11:
When i try store a PROGMEM char in eeprom, program crash and Exception(3) happen. 

```cpp

const char message[] PROGMEM = "Message";
EEPROM.begin(512);
EEPROM.put(0, message);

```

earlephilhower, 2019-08-16:
EEPROM.put does not support PROGMEM sources, so this is expected.  You need to make the values sent in be located in ram (i.e. a local stack copy of the PROGMEM structure using `memcpy_P`).

closed 2019-08-16 by earlephilhower

mbm60, 2019-08-17:
But i use non structure with PROGMEM and working.
```cpp
const char new[] PROGMEM = "New";
EEPROM.put(0, FPSTR(new));
```
How can i use FPSTR for structure.

earlephilhower, 2019-08-17:
In general, you can't pass in flash addresses to EEPROM.  Copy it to heap or stack and write it from there.

marza-pane, 2019-12-11:
I am having a similar issue storing "nested" structures to flash. I made a small code that will make my ESP8266 crash and reboot:

`struct variable`
`{`   
    `char        name[30];`
    `uint32_t    value;`
`};`

`const   variable    var1    PROGMEM = { "temperature", 25 };`
`const   variable    var2    PROGMEM = { "pressure", 101325 };`
`const   variable    var3    PROGMEM = { "volume", 1 };`

`const variable  list[3] PROGMEM = `
`{`
    `var1, var2, var3`
`};`

`void setup(void)`
`{`
    `Serial.begin(115200);`
    `Serial.println(F("\n<BEGIN>\n"));`
`}`

`void loop(void) {}`

it looks like a PROGMEM struct containing another PROGMEM struct makes the ESP crash. If I just avoid declaring _var1_, _var2_ and _var2_ and simply put all in _list_, it works. This will not crash:

`struct variable`
`{`
    `char        name[30];`
    `uint32_t    value;`
`};`

`const variable  list[3] PROGMEM = `
`{`
    `{ "temperature", 25 },`
    `{ "pressure", 101325 },`
    `{ "volume", 1 }`
`};`

`void setup(void)`
`{`
`   Serial.begin(115200);`
`    Serial.println(F("\n<BEGIN>\n"));`
`}`

`void loop(void) {}`

I am just a noob when it comes to compile options and I have no idea what to do.
Thanks.