esp8266/Arduino #6417: SPIClass::transfer(void *buf, uint16_t count) transfers with wrong block structure

UlliBien, 2019-08-13:
concerns spi.cpp, lines 309 to 329

The current algorithm of void SPIClass::transfer(void *buf, uint16_t count) is
-  repeat until buf is 32bits-aligned: transmit a single bytes using SPIClass::transfer(uint8_t data), increment buf and decrements count. This could occur up to 3 times.
- transmits multiple of 4 bytes via SPIClass::transferBytes(const uint8_t * out, uint8_t * in, uint32_t size). transferBytes() splits data into 64 byte blocks.
- transmit the remaining bytes (up to 3) as single byte transfers using SPIClass::transfer(uint8_t data)

If you use hardware CS there is the following problem. Mention, you *must* use hardware CS if you use the altenate pin configuration (hspi overlap):

After each transfer CS goes HIGH and LOW again. So, using transfer(void *buf, uint16_t count) you dont have a single block transfer. There are lot of tranfers with different sizes. In worst case: 3 single byte transfers, some 64 Byte transfers, 3 single Byte transfers.

You one can not avoid that this happens after every 64 bytes, because the IO buffer is only that long. But up to 64 bytes only a single block should be transferred.

The call sequence is:
```
transfer(void *buf, uint16_t count)      (establishes 32 bit aligment)
 |
 +--> transferBytes
      |
      +--> transferBytes_    (establishes 32 bit aligment too)
	        |
		    +--> transferBytesAligned_
```
You see, there is no need for method "transfer" to make a 32 bit alignment. This is done by "transferBytes_" (with underline at the end) anyway later. The algorithm of "transferBytes_" is much better. It transfers in 64 Byte blocks.

Please replace the code if "transfer" to a simple call to "transferBytes":
```
void SPIClass::transfer(void *buf, uint16_t count) {
  transferBytes(reinterpret_cast<uint8_t*>(buf), nullptr, count);
}
```

thangktran, 2019-08-16:
i'll take a look at this issue

earlephilhower, 2019-08-16:
@UlliBien, is the current behavior with multiple chip selects actually illegal per the spec, or is it something that's just not nice to see on a scope?  I'm asking because things like the SD library now use the SPI library for larger transfers, and it doesn't seem to be having any issue.

UlliBien, 2019-08-17:
More complex slaves understand the lowering of the CS signal as the start of a new transmission. Often the transfer starts with some command bytes, then address bytes follow. Occasionally, dummy bytes are inserted. Then the data bytes follow (The ESP8266 hardware supports this type of transaction exactly.). If the CS signal is raised again before completion of the transfer, the current transfer is considered incomplete and is discarded.

I've spent over three hours trying to figure out how the SD library works. It does not look to me as if it uses the hardware handling of the CS signal. The CS signal is probably set conventionally by appropriate instructions in the code. The described problem only arises if you let the hardware drive the CS signal. So this library works properly.

----------------------------------------------

Finally, I'd like to say that the ESP8266 SPI implementation is one of the worst I've ever seen. 

- It contains a number of new methods, all undocumented. One has to study the implementation code of the methods to understand what they do. You can only do that if you deal with the only sparsely documented internal registers of the ESP8266 too. Some methods have hints in the .cpp file. But you should never have to look at this file as a user of a library. The only thing you have to look at is the header file.

- In combination, the new methodes do not work properly (e.g., "transfer16" using "setHwCs").

- Others are incompatible with the Arduino / Uno implementation. For example, "begin": ESP8266 sets the frequency to 1MHz, Arduino does not do that.

- Some are very inefficient. For example, the use of SPISettings. The determination of the "ClockDivider" may be very time-consuming. Instead of calculating, storing and then using the "ClockDivider" once at the construction of a "SPISettings" object, it is recalculated each time "beginTransaction" is called.

- The hardware supports up to three CS signals so that up to three slaves can be controlled by the hardware. The library does not support this. So using "setHwCs" there is only one slave.

- The "pins" method gives the impression that there is more than one SPI interface available. Also, the SPIClass constructor allows you to create multiple instances of the class. But all instances work with the same register set and thus influence each other. This is a very bad design. The Arduino implementation is not perfect too. But there is no choice to disable the creation of the access object to the interface, as it could be done in the ESP8266 library via
```
#if !defined(NO_GLOBAL_INSTANCES) && !defined(NO_GLOBAL_SPI)
extern SPIClass SPI;
#endif
```
Either the library should stay completely compatible with the Arduino version, then there are no problems. But if there are extensions, they should do there job properly.

earlephilhower, 2019-08-17:
@UlliBien, thanks for the writeup.

You've obviously got a need and have been looking at SPI extensively.  I don't think the current maintainers have much involvement in the SPI library, and FWIW it does seem to hobble along well enough for most other users.  We'd love to see a PR to fix any of the points you addressed, if you're up to it.