esp8266/Arduino #6418: Sketch uploading fails in verbose mode

sysprogs, 2019-08-13:
The ESP8266 Arduino package 2.5.2 contains an invalid upload command template that fails the upload when using verbose mode.

The following line in **platform.txt** causes the problem:

> tools.esptool.upload.pattern="{cmd}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" "{upload.verbose}" write_flash 0x0 "{build.path}/{build.project_name}.bin" --end

This expands to the following:

> <...>\tools\python\3.7.2-post1\python "<...>\hardware\esp8266\2.5.2/tools/upload.py" --chip esp8266 --port "COM5" --baud "115200" "-vv" version --end --chip esp8266 --port "COM5" --baud "115200" "-vv" write_flash 0x0 "<...>.ino.bin" --end

The "-vv" ends up being passed to upload.py instead of the Python itself, causing the following error:

>esptool: error: unrecognized arguments: -vv

Modifying the pattern as shown below solves the problem:

> tools.esptool.upload.pattern="{cmd}" "{upload.verbose}" "{runtime.platform.path}/tools/upload.py" --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" {upload.erase_cmd} --end --chip esp8266 --port "{serial.port}" --baud "{upload.speed}" write_flash 0x0 "{build.path}/{build.project_name}.bin" --end

earlephilhower, 2019-08-16:
I can't repro this, and `platform.txt` doesn't have `-vv`, it has `--trace`:

````
tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=--trace
tools.esptool.upload.params.quiet=
````

I'm seeing the following:
````
Creating BIN file "/tmp/arduino_build_21714/sketch_aug02a.ino.bin" using "/tmp/arduino_build_21714/sketch_aug02a.ino.elf"
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/python/python /home/earle/Arduino/hardware/esp8266com/esp8266/tools/signing.py --mode sign --privatekey /home/earle/Arduino/sketch_aug02a/private.key --bin /tmp/arduino_build_21714/sketch_aug02a.ino.bin --out /tmp/arduino_build_21714/sketch_aug02a.ino.bin.signed --legacy /tmp/arduino_build_21714/sketch_aug02a.ino.bin.legacy_sig
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-size -A /tmp/arduino_build_21714/sketch_aug02a.ino.elf
Sketch uses 369607 bytes (35%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34720 bytes (42%) of dynamic memory, leaving 47200 bytes for local variables. Maximum is 81920 bytes.
/home/earle/Arduino/hardware/esp8266com/esp8266/tools/python/python /home/earle/Arduino/hardware/esp8266com/esp8266/tools/upload.py --chip esp8266 --port /dev/ttyUSB0 --baud 921600 --trace version --end --chip esp8266 --port /dev/ttyUSB0 --baud 921600 --trace write_flash 0x0 /tmp/arduino_build_21714/sketch_aug02a.ino.bin --end 
esptool.py v2.6
2.6
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.001 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.101 Timed out waiting for packet header
....
....

Could you have a partial installation, or older platform.txt (or platform.local.txt)?

f5soh, 2019-08-17:
Looks like @sysprogs own a "ESPDuino (ESP-13 Module)" board, the only with "menu.UploadTool.esptool.upload.verbose"  option.
https://github.com/esp8266/Arduino/blob/6dd847401486e1c45cb6beb5fe3758b91aed866a/boards.txt#L770

earlephilhower, 2019-08-18:
Good eyes.  I wonder what the reason was for having a separate option.

f5soh, 2019-08-18:
Same for generator ;)
https://github.com/esp8266/Arduino/blob/6dd847401486e1c45cb6beb5fe3758b91aed866a/tools/boards.txt.py#L303

PR #6426 (2019-08-19): Fix espduino verbose upload
Fixes #6418
commit def85866 (2019-08-19): Fix espduino verbose upload (#6426)

Fixes #6418
closed 2019-08-19 by earlephilhower