esp8266/Arduino #6420: Panic after OTA

gmag11, 2019-08-14:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it. **It fails**
- [x] I have filled out all fields below.

#### Platform

- Hardware:					Wemos D1 Pro and ESP-12
- Core Version:			2.5.2 and latest git
- Development Env:		VisualMicro
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		4MB and 16MB for Pro
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		OTA
- Upload Speed:   n/a
- Espressif FW: tried both nonos-sdk 2.2.1+100 and 2.2.1 (legacy)

### Problem Description

I'm developing a sensor network system that communicates over ESP-NOW. All messaging interchange works smoothly.
Now I'm implementing OTA over ESP-NOW so that I can update nodes wirelessly without having to switch to STA WiFi.
I manage to send firmware binary in 222 bytes chunks to fulfil ESP-NOW limitation of max 250 bytes payload. I add MD5 calculation to ensure binary is transmitted without errors.

The problem comes after OTA flash has been finished and I call `ESP.restart()`or `ESP.reset()`. I get a panic and stack dump that I've not been able to decode.

OTA firmware is not loaded.

If I do not call restart function code runs fine. If I reboot ESP8266 manually old firmware runs.

Project is on https://github.com/gmag11/EnigmaIOT/tree/feature/OTA

### Code
Code that file is under [this line](https://github.com/gmag11/EnigmaIOT/blob/d408142c4afb839ac508ecdc4d4a1da84a734503/src/EnigmaIOTSensor.cpp#L915). 
It is called asynchronously after receiving last packet. All other 222 bytes long packets have been written one by one without issues.
Could this be the reason? Is there any problem to write firmware in such small chunks? The only that I've not tested is make an intermediate buffer so I call `Update.write()` less often.

```
if (Update.end (true)) {
    .....
    Update.printError (otaError);
    otaError.trim (); // remove line ending
    DEBUG_WARN ("OTA Finished");
    DEBUG_WARN("%s", otaError.c_str ());
    delay (1000);
    ESP.reset (); // <--- Panic happens here
} else {
    .....
    Update.printError (otaError);
    otaError.trim (); // remove line ending
    DEBUG_ERROR ("OTA Failed");
    DEBUG_ERROR ("%s", otaError.c_str ());
}
```
### Debug messages
I've enabled Updater.h debug messages, that seem ok.

```
W [1727] 37704 free (processOTACommand:820) OTA message #0
[... 356 lines cut ...]
3ffffed0:  00004bc6 00000000 00000000 fffffffe  
3ffffee0:  00000000 3fffc6fc 00000000 3fff1a48  
3ffffef0:  00000000 3fffdad0 3fff1a78 00000030  
3fffff00:  00000000 3fffdad0 3fff1a78 00000030  
3fffff10:  00000000 4bc6a7f0 29fbe76c 00000000  
3fffff20:  00000000 00000000 4bc6a7f0 00000000  
3fffff30:  3fff2934 3fff0fa8 401001f8 000113cf  
3fffff40:  00000000 00000004 3fff1b10 4020a931  
3fffff50:  013ee788 3fff0fa8 3fff1b10 40201315  
3fffff60:  00000000 00000000 00000002 3fff1a78  
3fffff70:  3fffdad0 3fff12a8 3fff0fa8 40201f3b  
3fffff80:  3fff2934 feef0814 feefeffe 3fff1a78  
3fffff90:  00000000 00000000 00000001 4020c105  
3fffffa0:  3fffdad0 00000000 3fff1a48 4020c236  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vadfc28d7
~ld
```

gmag11, 2019-08-14:
A second run gave a different error message with same code:

```
W [54269] 33816 free (processOTACommand:820) OTA message #1691
W [54269] 33816 free (processOTACommand:877) 222 bytes written. Total 375402
W [54298] 33816 free (processOTACommand:820) OTA message #1692
W [54299] 33816 free (processOTACommand:877) 222 bytes written. Total 375624
W [54603] 33816 free (processOTACommand:820) OTA message #1693
W [54678] 33816 free (processOTACommand:877) 8 bytes written. Total 375632
W [54679] 33816 free (processOTACommand:892) OTA end
W [54679] 33816 free (processOTACommand:894) OTA MD5 feed930466a499285368bc68ae9d4140
W [54687] 33816 free (processOTACommand:897) OTA MD5 check OK
.
MD5 Success: feed930466a499285368bc68ae9d4140
Staged: address:0x002A4000, size:0x0005BB50

Exception (9):
epc1=0x4020db64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000056 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffeac0 end: 3fffffb0 offset: 01a0
3fffec60:  3fff3d38 00100406 00000008 3fff1ad8  
3fffec70:  40251298 00000001 00000000 3fffec80  
3fffec80:  3fff1a48 3fff1ad8 3fffece0 4020d6ed  
3fffec90:  3fff1a48 3fff1ad8 3fffece0 4020f41a  
3fffeca0:  4020c69e 000001f4 3fff1a68 3fff1ad8  
3fffecb0:  3fff1a48 00000001 3fff1a68 402027d0  
3fffecc0:  64656566 34303339 34613636 38323939  
3fffecd0:  38363335 38366362 64396561 30343134  
3fffece0:  40219000 00000000 000003e8 40218222  
3fffecf0:  00000000 00000000 80ff1398 069d06ff  
3fffed00:  3fff0f00 00000020 0000d59f 00000008  
3fffed10:  858fdcdd 00000013 3fff1398 00000001  
3fffed20:  00000020 3fff0f00 ffffd8d8 40202996  
3fffed30:  7a702322 3fff25c4 3ffedcac 00000001  
3fffed40:  00000020 3fff0f00 3ffeb49f 40202abe  
3fffed50:  00000020 3ffed2e8 3ffed2e8 00000000  
3fffed60:  7a702322 03397217 3ffedd30 40102476  
3fffed70:  00000024 3fffee00 00000001 3ffeda50  
3fffed80:  ffffffdb 03397217 401028b6 00000000  
3fffed90:  3fff1000 3ffeb49f 3fff0f00 40202cf4  
3fffeda0:  3fffee00 00000024 00000001 401022a5  
3fffedb0:  3ffed310 03397217 50590000 00000000  
3fffedc0:  3ffeb498 00000000 3ffeb490 40202d35  
3fffedd0:  00000000 00000000 0000001f 40203252  
3fffede0:  033a2a10 401028b6 00000100 4022b69f  
3fffedf0:  402357c5 3ffea168 00000001 00000001  
3fffee00:  24c2ddbe 3fff9714 3ffeb490 40235307  
3fffee10:  3ffeb490 3fff21fc 000000d0 3ffeb478  
3fffee20:  00000000 0000001f 4010526d 4000050c  
3fffee30:  4022ce64 00000000 0000001f 40100bb9  
3fffee40:  00000000 00000000 0000001f 4010526d  
3fffee50:  4000050c 4020c210 3fffd9d0 00000000  
3fffee60:  40000f83 00000030 00000010 ffffffff  
3fffee70:  40000f3d 00000023 00000000 3ffe959c  
3fffee80:  3ffe959c 402287c8 00000010 40100950  
3fffee90:  00000000 40221748 0000d57e 402287c8  
3fffeea0:  00000000 4bc6a7f0 0000003b 00000000  
3fffeeb0:  40234e51 00000000 4bc6a7f0 00000000  
3fffeec0:  4022c7bc 4022c90b 3ffeb478 3ffed2e8  
3fffeed0:  3fff21fc 00000000 3fff2174 402216ed  
3fffeee0:  00f6903c 00000000 00000000 3ffeb482  
3fffeef0:  4023ceb0 0000004f 3ffeb46c 3fff25c4  
3fffef00:  3ffeb46c 00000000 0000003b 4023c7de  
3fffef10:  3ffed2e8 3ffeb46c 3fffdcc0 3ffe99d8  
3fffef20:  3ffeb488 3fff24a4 000000d0 00004ee0  
3fffef30:  00000010 3ffed2e8 00000000 3ffe8508  
3fffef40:  4023c0b7 3fffdab0 00000000 4020c09b  
[... 253 lines cut ...]
3fffff20:  00000000 00000000 4bc6a7f0 00000000  
3fffff30:  3fff288c 3fff0f00 401001f8 0000d54b  
3fffff40:  00000000 00000004 3fff1a68 4020a931  
3fffff50:  00f69efb 3fff0f00 3fff1a68 40201315  
3fffff60:  00000000 00000001 00000002 3fff19d0  
3fffff70:  3fffdad0 3fff1200 3fff0f00 40201f3b  
3fffff80:  3fff288c feef0814 feefeffe 3fff19d0  
3fffff90:  00000000 00000000 00000001 4020c105  
3fffffa0:  3fffdad0 00000000 3fff19a0 4020c236  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vadfc28d7
~ld
```

Exception decoder gives only this text: 
`Failed to resolve hardware exception from the monitor text`
`Exception:Error:9 -> LoadStoreAlignmentCause: Attempt to read/write memory with an unaligned address (for example, trying to read/write a 32-bit word at an address that is not a multiple of 4)
`

TD-er, 2019-08-14:
Can you try writing in chunks of a multiple of 4 bytes?
222 bytes isn't.

gmag11, 2019-08-15:
Thanks, I'll try to reduce chunk size to 220 bytes

gmag11, 2019-08-15:
Sadly, this gives the same result.

```
........
W [63962] 33648 free (processOTACommand:877) 220 bytes written. Total 375320
W [64034] 33648 free (processOTACommand:820) OTA message #1707
W [64034] 33648 free (processOTACommand:877) 220 bytes written. Total 375540
W [64047] 33648 free (processOTACommand:820) OTA message #1708
W [64141] 33648 free (processOTACommand:877) 92 bytes written. Total 375632
W [64142] 33648 free (processOTACommand:892) OTA end
W [64143] 33648 free (processOTACommand:894) OTA MD5 feed930466a499285368bc68ae9d4140
W [64150] 33648 free (processOTACommand:897) OTA MD5 check OK
.
MD5 Success: feed930466a499285368bc68ae9d4140
Staged: address:0x002A4000, size:0x0005BB50

User exception (panic/abort/assert)
Abort called

>>>stack>>>

ctx: sys
sp: 3fffeb20 end: 3fffffb0 offset: 0000
3fffeb20:  3fffece0 3ffe8300 00000040 4021ca99  
3fffeb30:  000000fe 00000000 00000000 00000000  
3fffeb40:  00000000 00000000 00000000 00000018  
3fffeb50:  000000b0 00000002 3fff10a8 000000b0  
3fffeb60:  0000ffff b49eb0bf 3fffece0 4020c5c6  
3fffeb70:  3fff10a8 00000019 3fff0fa8 4020c5dc  
3fffeb80:  00000020 42353030 0a303542 4020e4ad  
3fffeb90:  36f41e03 5660a210 ee6e713b b51263cf
..............
```

d-a-v, 2019-08-15:
Stack traces are useless when not decoded.
> stack dump that I've not been able to decode

What did you try ?

gmag11, 2019-08-17:
I was doing that using VisualMicro stack decoder. I gave no information.

Now I've used Arduino IDE and got expected information:

```
Decoding stack results
0x4020f04e: raise_exception() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_postmortem.cpp line 236
0x4020c89c: MD5Builder::calculate() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\MD5Builder.cpp line 76
0x4020ca98: Print::printf_P(char const*, ...) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Print.cpp line 97
0x4020c9d4: Print::printf(char const*, ...) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Print.cpp line 74
0x4020d016: UpdaterClass::printError(Print&) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Updater.cpp line 497
0x4020f126: __delay(unsigned long) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 57
0x402027d0: EnigmaIOTSensorClass::processOTACommand(unsigned char const*, unsigned char const*, unsigned char) at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 921
0x40218422: CFBCommon::decrypt(unsigned char*, unsigned char const*, unsigned int) at C:\Arduino\libraries\CryptoLegacy\src\CFB.cpp line 120
0x40202996: EnigmaIOTSensorClass::processControlCommand(unsigned char const*, unsigned char const*, unsigned int) at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 961
0x40202abe: EnigmaIOTSensorClass::processDownstreamData(unsigned char const*, unsigned char const*, unsigned int, bool) at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 1021
0x40202cf4: EnigmaIOTSensorClass::manageMessage(unsigned char const*, unsigned char const*, unsigned char) at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 1136
0x40202d35: EnigmaIOTSensorClass::rx_cb(unsigned char*, unsigned char*, unsigned char) at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 369
0x4020332e: Espnow_halClass::rx_cb(unsigned char*, unsigned char*, unsigned char) at C:\Arduino\libraries\EnigmaIOT\src\espnow_hal.cpp line 40
0x4020ead3: loop_task(ETSEvent*) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 148
0x4020ec48: loop_wrapper() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 132
0x402218ed: sys_timeout_abs at core/timeouts.c line 189
0x4020ead3: loop_task(ETSEvent*) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 148
0x4020f61e: _calloc_r(_reent*, size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\heap.cpp line 49
0x401009fc: calloc(size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 1704
0x4021ebd5: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x4020f61e: _calloc_r(_reent*, size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\heap.cpp line 49
0x4021ebd5: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x4020f61e: _calloc_r(_reent*, size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\heap.cpp line 49
0x401009fc: calloc(size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 1704
0x4021ebd5: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x4020f61e: _calloc_r(_reent*, size_t, size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\heap.cpp line 49
0x4021ebd5: _Balloc at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/mprec.c line 124
0x4021dfc1: _dtoa_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdlib/dtoa.c line 858
0x401009c4: malloc(size_t) at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 1677
0x4021f5aa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021f5aa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021b0c1: _printf_i at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 194
0x4021f5aa: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4021f4e4: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 180
0x4010027f: millis() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x40201f3b: EnigmaIOTSensorClass::handle() at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 353
0x4010026c: millis() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_wiring.cpp line 188
0x4020a935: CayenneLPP::~CayenneLPP() at C:\Arduino\libraries\CayenneLPP\src\CayenneLPP.cpp line 17
0x40201315: setup() at C:\Arduino\libraries\EnigmaIOT\examples\enigmaiot_sensor/enigmaiot_sensor.ino line 72
0x40201f3b: EnigmaIOTSensorClass::handle() at C:\Arduino\libraries\EnigmaIOT\src\EnigmaIOTSensor.cpp line 353
0x4020eb3d: esp_schedule() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 102
0x4020ec6e: loop_wrapper() at C:\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 142
```

gmag11, 2019-08-17:
Great! It was my fault, nothing related with update library. I was reusing a variable name on `EnigmaIOTSensor.cpp:921`.
Now, my problem is that new firmware is not applied, but I'll write it in a different issue.

Thanks

closed 2019-08-17 by gmag11