esp8266/Arduino #6431: Updating to 2.5.2 but compiler throws error about files inside 2.5.0 when such folder does not exist

kapyaar, 2019-08-21:
### Basic Infos

- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[2.5.2 release]
- Development Env:		[Arduino 1.6.5]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[dout(compatible)]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I just updated from 2.5.0 to 2.5.2 via board manager. After that, when I compile, I get an error saying "first defined here" inside a file under folder 2.5.0. I confirmed that the 2.5.0 folder in 
AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\
is non existent. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

ESP8266 AdvancedWebServer Example

```

### Debug Messages

```
Build options changed, rebuilding all
c:/users/dar w/appdata/roaming/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: {archive_file_path}(umm_malloc.cpp.o):C:\Users\Dar W\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc/umm_malloc.cpp:634: multiple definition of `umm_heap'; {archive_file_path}(umm_malloc.c.o):C:\Users\Dar W\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc/umm_malloc.c:632: first defined here

...
And a bunch of other files with the same error. 
```

earlephilhower, 2019-08-21:
The error message shows where your other copy is.  Delete the whole 2.5.0 tree there and give it another go.  `C:\Users\Dar W\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\2.5.0\cores\esp8266\umm_malloc/umm_malloc.c`

Closing as unrelated to the core.

closed 2019-08-21 by earlephilhower

kapyaar, 2019-08-22:
Yes offcourse! thats the first thing I tried to do. But the funny part is the folder 2.5.0 does not exist as I had mentioned.  I searched my entire computer for umm_malloc.c, nowhere to be found other than the one under 2.5.2. Which made sense as when I updated to 2.5.2, the folder would get overwritten.

Anyways, If it is just me, you can close this.