esp8266/Arduino #6434: WiFiClientSecure::available() blocks on dropped connection when TX buffer full

johnm545, 2019-08-22:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12E
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	[40Mhz]
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

I've come across some unwanted behaviour in the WiFiClientSecure::available() and _run_until methods.  This becomes apparent when sending MQTT messages of a few hundred bytes at relatively slow intervals (minutes) without regular keep-alive pings, and the connection drops somewhere between the access point and the server (i.e. wifi stays up but internet goes down).  

The issue is as follows: when messages are sent but not ACK-ed, the TCP buffers begin to fill.  When there is no longer enough space for a complete message, the write attempt will block for 10 seconds, but still report success.  No problem yet, because it remains buffered.  

At this point, subsequent calls to WiFiClientSecure::available() will block for 5-10 seconds, because the available() method calls _run_until, which then attempts to re-write the bytes  remaining in the bearssl engine's buffer to the underlying WiFiClient.  This will occour endlessly until a further write attempt is made from the application via WiFiClientSecure::write; at this point the connection will be closed after write timeout.

Example and debug log below.  I am using the new ArduinoMqttClient, which I much prefer over the old PubSubClient, but this issue will occour with any WiFiClientSecure connection

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include <ArduinoMqttClient.h>

const char* msg = R"(To be, or not to be, that is the question:
Whether 'tis nobler in the mind to suffer
The slings and arrows of outrageous fortune,
Or to take arms against a sea of troubles
And by opposing end them. To die—to sleep,
No more; and by a sleep to say we end
The heart-ache and the thousand natural shocks
That flesh is heir to: 'tis a consummation
Devoutly to be wish'd. To die, to sleep;
To sleep, perchance to dream—ay, there's the rub:
For in that sleep of death what dreams may come,
When we have shuffled off this mortal coil,
Must give us pause—there's the respect
That makes calamity of so long life)";

WiFiClientSecure wifiClient;
MqttClient mqttClient(wifiClient);

const char broker[] = "test.mosquitto.org";
int        port     = 8883;
const char topic[]  = "arduino/simple";

const long interval = 60000L; // send every minute
unsigned long previousMillis = 0;

[... 669 lines cut ...]
        }
        if (rlen > 0) {
          br_ssl_engine_recvrec_ack(_eng, rlen);
        }
        no_work = 0;
        continue;
      }
    }

    /*
       We can reach that point if the target RECVAPP, and
       the state contains SENDAPP only. This may happen with
       a shared in/out buffer. In that case, we must flush
       the buffered data to "make room" for a new incoming
       record.
    */
    br_ssl_engine_flush(_eng, 0);

    no_work++; // We didn't actually advance here
  }
  // We only get here if we ran through the loop without getting anything done
  return -1;
}

```

earlephilhower, 2019-08-22:
Thanks for the detailed description and logs and the proposed fix.  Does it also fail with John Donne, or is this a Shakespeare-only issue?  :)

Why not submit a PR and get your name in the git logs?  You've got a failing test and a proposed solution ready to go, which is really all you need!

johnm545, 2019-08-23:
Thanks, I will experiment some more with it over the weekend and submit a
PR if happy.

johnm545, 2019-08-26:
I have tested this some more and whipped up a simpler MCVE that does not require the MQTT client, just ugly HTTP String hacking.  To reproduce it, you have to disconnect your internet in the first couple of minutes, because the postman echo server will close the connection after a couple of posts.  Once the WiFiClient tx buffer fills, available() calls will block for 5 seconds.  

The 15 second block I was seeing in the previous example occoured because the MQTT client was calling connected() first, which calls available(), then calling available() twice.  The below example is a better test, but in practice the if connected() while available() pattern is common and results in long bocks.

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* host = "postman-echo.com";
int port = 443;  //https
WiFiClientSecure wifiClient;

unsigned long lastSend;
unsigned long sendInterval = 60000L;

void setup() {
  wifiClient.setInsecure(); // disable validation of server certificate
  Serial.begin(115200);
  delay(1000);

  WiFi.begin(/*ssid, password if needed */);     //Connect to your WiFi router
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("wifi connected");

  if (!wifiClient.connect(host, port)) {
    Serial.println("initial connection failed, restart to try again");
    while (1) delay(1);
  }
  Serial.println("Secure socket connected to server. Disconnect internet without disconnecting wifi to demonstrate bad behaviour");
  lastSend = millis();
}

void loop() {

  if (millis() - lastSend > sendInterval) {
    lastSend = millis();

    String msg = "POST /post HTTP/1.1\r\n";
    msg += "Host: " + String(host) + "\r\n";
    msg += "Content-Type: application/x-www-form-urlencoded\r\n";
    msg += "Content-Length: 121\r\n\r\n";
    msg += "thequickbrownfoxjumpsoverthelazydogfjaoranckaeioavrjkanvjknvaeruivnaivnrakjnvkanvkanuarwefjrieavnkjanrlsnbiakjncoanranven\r\n";
    msg += "Connection: keep-alive\r\n\r\n";

    int afw = wifiClient.availableForWrite();

    Serial.print("Sending a message, unencrypted message length: "); Serial.println(msg.length());
    Serial.print("Bytes available for write in WiFiClient tx buffer: "); Serial.println(afw);
    Serial.println(msg);

    int msgLength = msg.length();
    int written = wifiClient.print(msg);

    Serial.print("request sent, bytes accepted by WiFiClientSecure: "); Serial.println(written);
    Serial.println("\n\n");
  }

  unsigned long beforeAvailableCall = millis();
  int avail = wifiClient.available();
  unsigned long elapsed = millis() - beforeAvailableCall;

  if (elapsed > 10) {
    Serial.print("available() blocked for "); Serial.print(elapsed); Serial.println(" millis!");
  }
  for (int i = 0; i < avail; i++) {
    byte b = wifiClient.read();
    Serial.print( (char)b );
  }
  if (avail) Serial.println();  

delay(1000);
}

```

Getting my head around how the _write and available methods work took quite a bit of head scratching, so I have brain-dumped my understanding below before I forgot.

The WiFiClientSecure::_write loop first does a _run_until(BR_SSL_SENDAPP). Usually this will return success instantly.  But if there is unsent record (encrypted) data hanging around from last time (state & BR_SSL_SENDREC == true), WiFiClient::write is called.  If this data can not fit in the WiFiClient tx buffer, then it will block until space is available or timeout occours (default 5sec).  If there is still space in the bearssl sendapp buffer (state & BR_SSL_SENDAPP == true) then _run_until returns success; if the sendapp buffer is full it returns -1 and the _write loop is broken, signaling a write failure

WiFiClientSecure::_write then feeds the user's data into the bearssl sendapp buffer (as much as will fit), then calls br_ssl_engine_flush to force the engine to generate an encrypted record. Now state & BR_SSL_SENDREC == true but the data has not yet been written to the WiFiClient.  So _write then calls flush(), which calls WiFiClientSecure::flush(0), which does a _run_until(BR_SSL_SENDAPP), causing the record data to be written to WiFiClient.  WiFiClientSecure::flush then flushes the WiFiClient with the WiFiClient default timeout (300ms)

WiFiClientSecure::_write does not really care what happens in the flush() method; it returns the number of bytes transfered to bearssl sendapp buffer.  It only returns a fail (i.e. 0 or less than it was asked to send) when the bearssl sendapp buffer is full (state & BR_SSL_SENDAPP == false).  This is fine but means that writes on dropped connections will not fail until both the WiFiClient tx buffer and bearssl sendapp buffers are full; when the former is full but the later not, writes will block for 15sec*) but still return success.  It might be more elegant to avoid the blocking-but-apparently-successful write, but that's not simple.

WiFiClientSecure::available() calls _run_until(BR_SSL_RECVAPP, false), and this also triggers a blocking write attempt if there is still record data hanging around from previous write attempts (i.e. state & BR_SSL_SENDREC == true).  The fix is to avoid writing to the WiFiClient more data than will fit in its tx buffer when the blocking flag == false; I will submit a PR with this change.

*The 15sec block occours because WiFiClientSecure::_write has a call to connected() at the top, which in turn calls available().

devyte, 2019-09-17:
Closing via #6449.

closed 2019-09-17 by devyte