esp8266/Arduino #6437: Loop interface (127.0.0.1)

igvassor-git, 2019-08-23:
I have the Arduino IDE 1.8.9 and the core 2.5.2.
I want to connect a local client and a local server:

  WiFiServer Nserver(2222);
  WiFiClient Wclient;
   .....
  Nserver.begin();
   .....
  if(!Nclient.connected())
  {
    int T;
    T = Nclient.connect(IPAddress(127,0,0,1),2222);
    Console_write(String(T));
  }
  else
  {
    Console_write("Connected");
  }
    ....

To do this, I set the following in the file ..\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\tools\sdk\lwip2\include\lwipopts.h	:

/*
   * LWIP_NETIF_LOOPBACK==1: Support sending packets with a destination IP
   * address equal to the netif IP address, looping them back up the stack.
*/
      #if !defined LWIP_NETIF_LOOPBACK || defined __DOXYGEN__
      #define LWIP_NETIF_LOOPBACK             1
      #endif

The core was recompiled successfully.
But the connection between the client and server is not established.
Non-local the client connections are successful.
Non-local the server connections are established successfully.
Where am I mistaken?
Please help me.

devyte, 2019-08-23:
@d-a-v is the expert here. 

The lwip2 libs are included precompiled with the core. That means that just changing lwipopts won't have an effect on the core rebuild. You have to first recompile lwip2 with your changes (there's a readme), then make sure the newly built libs made it to the core replacing the originally included ones (I'm not sure if they get copied over automatically, or you need make install, or manual cp over). And only then rebuild the core.

Be aware that the localhost interface is disabled on purpose, because it hasn't been tested. That means that enabling it could work. Or it could cause current wifi functionality to break. Or something else. 

Closing since this isn't a core issue.

devyte, 2019-08-23:
BTW, if you do get it to work, we'd love to hear about what you did and how it behaves. I'm thinking of a mini writeup for other users who could want to follow your footsteps.

closed 2019-08-23 by devyte

igvassor-git, 2019-08-24:
Thank. I will try.