esp8266/Arduino #6441: [Question] Light sleep and delay, millis behavior

cziter15, 2019-08-24:
After studying esp documentaion about automatic light sleep, few questions came to my mind.

Esp8266 goes light sleep within delay call and periodically wake up to check DTIM messages. Documentation says that with light sleep, CPU is pending and system clock is disabled. Only RTC remains enabled. It means that hardware timers will not work in light sleep mode. Right?

How does it affect delay and millis functions? Are their "times" based on RTC or system clock?

Regards,
Chris

matthijskooijman, 2019-10-05:
It seems that millis (and also micros) use `system_get_timer`:

https://github.com/esp8266/Arduino/blob/5d20137339442982172dd6d84488f0c2bb639930/cores/esp8266/core_esp8266_wiring.cpp#L175
https://github.com/esp8266/Arduino/blob/5d20137339442982172dd6d84488f0c2bb639930/cores/esp8266/core_esp8266_wiring.cpp#L198

I assume these are not updated during sleep (given there is also a `system_get_rtc_time()` function in the SDK). If you need a continuing clock, maybe you can use `system_get_rtc_time()` directly?

devyte, 2019-10-05:
Please read the [Espressif NONOS SDK API Reference document 2c](https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf) for details.
Also, please be aware that this is not a general support forum for such questions, but an issue tracker meant to track issues in the core. 
Closing due to not an issue.

closed 2019-10-05 by devyte