esp8266/Arduino #6442: Why HTTP_UPDATE_FAILD Error during OTA updation?

wounder123, 2019-08-24:
SERIAL WINDOW:ARDUINO

OTA RECEIVE OK
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 200
[httpUpdate]  - len: 316720
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 2793472
[httpUpdate]  - current Sketch Size: 349632
[httpUpdate] runUpdate flash...
ssl->need_bytes=15936 > 6859
failed to grow plain buffer
[httpUpdate] peekBytes magic header failed
[update] Update failed.

code::::

// Check for a firmware update
  LOG("Checking for firmware, same URL.\r\n");
  Result = ESPhttpUpdate.update(HTTPSURL,Fingerprint);

  // if successful, ESP will restart
  LOG("Oh noes, update failed!\r\nResult:");
  LOG(Result);
  LOG("\r\n");

  switch(Result) {
    case HTTP_UPDATE_FAILED:
        sprintf((char *)Buffer, "Update failed, error %d: %s\r\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        LOG((char *)Buffer);
        break;
    case HTTP_UPDATE_NO_UPDATES:
        LOG("No update available.\r\n");
        break;
    case HTTP_UPDATE_OK:
        LOG("You shouldn't see this message because I should be rebooting :)\r\n");
        break;
  }  

Note:
LOG is #define as serial.println()

d-a-v, 2019-08-24:
`ssl->need_bytes=15936 > 6859`

Not enough RAM

closed 2019-08-24 by devyte

wounder123, 2019-08-25:
Please tell me how to solve this issues