esp8266/Arduino #6443: ESP8266 version 2.5.2 versus version 2.4.2

JRAlvarez2017, 2019-08-24:
I have several applications using ESP8266 libraries like ESP8266WiFi, ESP8266WebServer, and WebSocketsServer, and all of them work very well with the ESP8266 version 2.4.2. However if I update to version 2.5.2 none of them work. All with the same problem: the phone (or other ESP8266 in STA mode) is no capable to connect to the ESP8266 WiFi AP.
Are there unsolved WiFi connectivity issues with the version 2.5.2?

Jason2866, 2019-08-26:
No

jandrassy, 2019-08-26:
what are your Tools menu settings?

Jason2866, 2019-08-26:
[image]

jandrassy, 2019-08-26:
@Jason2866, you have the problem?

jandrassy, 2019-08-26:
@JRAlvarez2017, what are your Tools menu settings?

d-a-v, 2019-08-27:
Did you try the third flash option "Erase all flash" (once) ?
With generic board and wemos d1 mini you must set 4MB flash size.

Jason2866, 2019-08-27:
Try a lower upload speed
[image]

For many D1 mini (clones) this speed is impossible.
Try 115200

jandrassy, 2019-08-27:
Crystal frequency 40 MHz? set 26MHz