esp8266/Arduino #6444: v2.5.1 and above causes ESP to crash on `server.send ( 200, "text/html"...`

AnumSheraz, 2019-08-24:
#### NOTE:
This issue occurs only on version 2.5.1 and 2.5.2. 
It works perfectly fine on v2.5.0

#### PROBLEM
I have a simple HotSpot mode code running on ESP, 

```
void send_gateway_node_config_html()
{
	Serial.println("Sending main page...");
	String s =PAGE_GatewayNode_Config;
	server.send(200, "text/html", s); 
}

void EnterHotSpotMode() {
    //entring hotspot mode
    ESP.eraseConfig();
    WiFi.disconnect();
    WiFi.setAutoConnect(false);
    WiFi.setAutoReconnect(false);  
    WiFi.mode(WIFI_AP_STA);
    nodeName = String(ACCESS_POINT_NAME) + String("-") + String(WiFi.softAPmacAddress().c_str());
    WiFi.softAP(nodeName.c_str() , ACCESS_POINT_PASSWORD);
    Serial.print("Connect to ssid " + nodeName + " to configure Node on IP: ");   
    Serial.println(WiFi.softAPIP());
    ConfigureWifi();
    server.on ( "/favicon.ico",   []() { Serial.println("favicon.ico"); server.send ( 200, "text/html", "" );   }  );
    server.on ( "/", send_gateway_node_config_html  );
    server.on ( "/admin/values", send_gateway_node_config_values_html );
    server.on ( "/admin/connectionstate", send_connection_state_values_html );    
    server.on ( "/style.css", []() { Serial.println("style.css"); server.send ( 200, "text/plain", PAGE_Style_css );  } );
    server.on ( "/microajax.js", []() { Serial.println("microajax.js"); server.send ( 200, "text/plain", PAGE_microajax_js );  } );
    server.onNotFound ( []() { Serial.println("Page Not Found"); server.send ( 400, "text/html", "Page not Found" );   }  );
    server.begin();
    Serial.println( "HTTP server started" );  
}
```

The page `PAGE_GatewayNode_Config` requires microAJAX.js code on `windows.onload` call 

```
window.onload = function ()
{
	load("style.css","css", function() 
	{
		load("microajax.js","js", function() 
		{
					setValues("/admin/values");
					setTimeout(GetState,3000);
		});
	});
}

```
This seems to crash on esp as soon as I request `GET /` . Here is the decoded stack trace

```
Exception 3: LoadStoreError: Processor internal physical address or data error during load or store
PC: 0x4020d65c
EXCVADDR: 0x40241d4d

Decoding stack results
0x4020793c: Print::println(char const*) at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/cores/esp8266/Print.cpp line 190
0x40201842: std::_Function_handler ::_M_invoke(const std::_Any_data &) at /Users/anum_sheraz/Documents/Arduino/projects/device_core_test/device_core_test.ino line 128
0x40207c98: String::invalidate() at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/cores/esp8266/WString.h line 267
0x402045d0: FunctionRequestHandler::canHandle(HTTPMethod, String) at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 20
0x401000a9: std::function ::operator()() const at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/xtensa-lx106-elf/include/c++/4.8.2/functional line 2465
0x402051c2: FunctionRequestHandler::handle(ESP8266WebServer&, HTTPMethod, String) at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266WebServer/src/detail/RequestHandlersImpl.h line 44
0x40207fe8: String::String(String const&) at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/cores/esp8266/WString.cpp line 41
0x4020524d: ESP8266WebServer::_handleRequest() at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 599
0x402054a4: ESP8266WebServer::handleClient() at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/libraries/ESP8266WebServer/src/ESP8266WebServer.cpp line 308
0x402027f7: RunDeviceSetupMode() at /Users/anum_sheraz/Documents/Arduino/projects/device_core_test/device_core_test.ino line 220
0x40202848: loop() at /Users/anum_sheraz/Documents/Arduino/projects/device_core_test/device_core_test.ino line 256
0x40208d74: loop_wrapper() at /Users/anum_sheraz/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.1/cores/esp8266/core_esp8266_main.cpp line 125

```

Note that ESP doesn't only crashes on this microAJAX call, I tried to load only a simply `Hello World ` web page and it crashed on that too :( 

#### Platform
Hardware:	[ESP8266 - LOLIN (WIMOS) D1 R2 & mini]
Core Version:	2.5.1 & 2.5.2 
Development Env:	[Arduino IDE 1.8.9]
Operating System:	[MacOS]

#### Settings in IDE

[image]

fredcazaux, 2019-08-28:
Get something similar here...

Everything OK on 2.5.0, but not on 2.5.1 and 2.5.2... In my code, it hangs randomly when calling `server.handleClient();` 
When looping, several calls pass OK (like between 5 to 20 times) and then hangs.

Libraries used : 
`#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <WiFiUdp.h>
#include <EEPROM.h>
#include <Timezone.h>   // https://github.com/JChristensen/Timezone
#include <ESP8266WebServer.h>
#include <ArduinoOTA.h>`

Thanks !

jandrassy, 2019-08-28:
2.5.0 uses SDK 3 pre-release. in 2.5.1 they reverted to 2.2.1 because of problems. In 2.5.2 you can select the SDK 3 in Tools menu for Generic esp8266 'board'.
with switch of the SDK it is recommended to, erase the flash, but it is possible that the problems go away after some time after the esp8266 executes RF calibration, which happens after some count of resets.
additionally you could try to reduce the RF TX power with WiFi.setOutputPower

devyte, 2019-11-10:
@AnumSheraz The "C" in MCVE stands for Complete. Your posted code is missing the html and other .js that contain calles functions. Without that, this issue will be closed.
Can you update your post to be complete?

AnumSheraz, 2019-11-11:
> @AnumSheraz The "C" in MCVE stands for Complete. Your posted code is missing the html and other .js that contain calles functions. Without that, this issue will be closed.
> Can you update your post to be complete?

update the answer :)

devyte, 2019-12-18:
@AnumSheraz a lot has changed since this was reported, including several relevant fixes. Is this issue still valid in 2.6.3?

devyte, 2020-02-02:
The provided app is still incomplete. Closing due to lack of feedback.

closed 2020-02-02 by devyte