esp8266/Arduino #6445: ESP8266 WiFi.begin() isn't working

Rafat001, 2019-08-25:
- Hardware:				ESP8266
- Development Env:		Arduino IDE 1.8.9
- Operating System:		Windows

### Settings in IDE

- Module:  			NodeMCU 1.0 (ESP-12E Module)
- Flash Size: 		        4MB
- lwip Variant:                  v2 Lower Memory
- CPU Frequency:		80MHz
- Upload Speed:              115200

### Problem Description

I did a project for IoT attendance system using ESP8266 and RDM6300. However, I can connect to wifi using WiFiMulti.addAP() function for my home WiFi but can't connect using WiFi.begin().

I tried WiFi.begin() instead of WiFiMulti.addAP() function to connect to my mobile hotspot but the esp8266 is failing to connect. In fact, WiFi.begin() also failed to connect to the same wifi router which was previously connected by WiFiMulti.addAP().

###Sketch

```cpp

#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClient.h>

#include <rdm6300.h>

#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

#define SCREEN_WIDTH 128 // OLED display width, in pixels
#define SCREEN_HEIGHT 64 // OLED display height, in pixels

// Declaration for an SSD1306 display connected to I2C (SDA, SCL pins)
Adafruit_SSD1306 display(SCREEN_WIDTH, SCREEN_HEIGHT, &Wire, -1);

#define RDM6300_RX_PIN 2
#define READ_LED_PIN 14

ESP8266WiFiMulti WiFiMulti;
Rdm6300 rdm6300;

String RFID[] = {"5958666", "5958741"};

String SID[] = {"161-115-111", "161-115-115"};

String NAME[] = {"Rafat", "Saney"};

int flag[2] = {0, 0};

void setup() {

  Serial.begin(115200);
  display.begin(SSD1306_SWITCHCAPVCC, 0x3C);
  pinMode(READ_LED_PIN, OUTPUT);
  digitalWrite(READ_LED_PIN, LOW);
  rdm6300.begin(RDM6300_RX_PIN);

  Serial.println();
  Serial.println();
  Serial.println();
  OLED("Connecting...");
  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFi.begin("abcd", "abcdabcd");
  Serial.print("Connecting");
  
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("Connected");
  OLED("Connected");
  Serial.println(WiFi.localIP());

}

void loop() {
  // wait for WiFi connection
  if ((WiFi.status() == WL_CONNECTED)) {
label:
    OLED("Place Your ID");
    String tag;
    int check = 0;
    if (rdm6300.update())
    {
      tag = String(rdm6300.get_tag_id());
      Serial.println(tag);
      for (int i = 0; i < 1; i++)
      {
        if (RFID[i] == tag) // checking validity of RFID
        {
          tone(READ_LED_PIN, 392);
          delay(100);
          noTone(READ_LED_PIN);
          check = 1;
          if (flag[i] == 1) // Flag=1 meaans already attendance counted
          {
            Serial.println("Attendance Already Counted");
            OLED("Already Counted");
            delay(2000);
            goto label;
          }
          OLED("Welcome "+NAME[i]);
          flag[i] = 1;
          WiFiClient client;
          HTTPClient http;

          Serial.print("[HTTP] begin...\n");
          String url = "http://metrounisas.000webhostapp.com/action.php?ID=" + SID[i];
          if (http.begin(client, url ) ) { // HTTP

            Serial.print("[HTTP] GET...\n");
            // start connection and send HTTP header
            int httpCode = http.GET();

            // httpCode will be negative on error
            if (httpCode > 0) {
              // HTTP header has been send and Server response header has been handled
              Serial.printf("[HTTP] GET... code: %d\n", httpCode);

              // file found at server
              if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
                String payload = http.getString();
                Serial.println(payload);
              }
            } else {
              Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
            }

            http.end();
          } else {
            Serial.printf("[HTTP} Unable to connect\n");
          }
        }
      }
      if (check == 0) {
        tone(READ_LED_PIN, 261);
        noTone(READ_LED_PIN);
        Serial.println("Invalid RFID");
        OLED("INVALID ID");
        delay(2000);
      }
    }
  }

  delay(1000);
}
void OLED(String txt){
  display.clearDisplay();
  display.setTextSize(1);
  display.setTextColor(WHITE,BLACK);
  display.setCursor(30, 30);
  display.println(txt);
  display.display();
  delay(2000);
}

```

### Debug Messages

```
Exception (3):
epc1=0x4020970d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4003bfc9 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffbc0 end: 3fffffc0 offset: 01a0
3ffffd60:  00000000 00000004 3ffffe40 3ffffef0  
3ffffd70:  4010156c 0000049c 000003fd 40209837  
3ffffd80:  0000049c 00001000 000003fd 4010156c  
3ffffd90:  40106980 3ffe85ad 3ffffe40 402354f1  
3ffffda0:  401004a5 402355db 3fff0c1c 0000049c  
3ffffdb0:  000003fd 3ffffef0 3fff0c1c 402355be  
3ffffdc0:  ffffff00 55aa55aa 00000035 00000020  
3ffffdd0:  00000020 00000048 0000004a aa55aa55  
3ffffde0:  000003ff 40235aa8 3fff0c1c 3fff0c1c  
3ffffdf0:  000000ff 3fff0c1c 0000049c 00000000  
3ffffe00:  401004a5 00000001 3fff0c2c 40235cc2  
3ffffe10:  00000008 3fff0c1c 000000ff 3ffffef0  
3ffffe20:  3fffff10 3fff0c53 00000035 00000020  
3ffffe30:  3fff0cdc 3fffff51 00000001 40235d72  
3ffffe40:  3ffffef0 40242be0 00000000 0000000c  
3ffffe50:  3fff101c 3fffff10 3fff6f54 40235d41  
3ffffe60:  3fff0c1c 40235da8 3ffe84dc 3ffe85ae  
3ffffe70:  402019ca 3ffe85ae 3ffe85b3 40201927  
3ffffe80:  61666152 00000074 3ffffe60 40204666  
3ffffe90:  4000e243 00000030 3ffffe60 402046c4  
3ffffea0:  616d7374 2e2e3730 3ffffe00 40204901  
3ffffeb0:  3ffffdf0 0693b740 000035af fffffffe  
3ffffec0:  00000090 3fffc6fc 00000003 40201dd4  
3ffffed0:  3ffee50c 000000f2 000000f2 40206bf8  
3ffffee0:  8ab9b001 00a0a64e 00000000 40206057  
3ffffef0:  78666152 3fff0c00 0000049c 00000000  
3fffff00:  00000a2e 0000007f 3fff0c1c 402356c6  
3fffff10:  61666172 66617274 32317461 40209500  
3fffff20:  00000001 3fff0c1c 00000000 402099e8  
3fffff30:  000003fd 402351a3 00000000 40101538  
3fffff40:  00000001 001ce8e7 3ffeea78 402351be  
3fffff50:  40235100 00000001 00000081 00000002  
3fffff60:  00000004 00000000 3ffee544 00000001  
3fffff70:  402083ea 000003e8 000003e8 3ffee870  
3fffff80:  3ffee544 00000000 3ffee750 40201194  
3fffff90:  00000000 00000000 ffefeffe feefeffe  
3fffffa0:  3fffdad0 00000000 3ffee840 40207830  
3fffffb0:  feefeffe feefeffe 3ffe8520 40101809  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

```

everslick, 2019-09-06:
Please reduce your test sketch to the minimum that still shows the error (remove OLED code). Additionally decode the stack trace with ExceptionDecoder, otherwise it is meaningless.

devyte, 2019-11-09:
A lot has been fixed since this was reported, including several fixes that are relevant.
Closing.
If you think the problem is still relevant, please open a new issue and follow the template instructions (pay close attention to the "M" part of MCVE).

closed 2019-11-09 by devyte