esp8266/Arduino #6446: WiFiClientSecure always fails to connect

aryghm, 2019-08-25:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X]I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it. (there is no stack dump)
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266]
- Core Version:			[latest git hash or date]
- Development Env:		[other]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [other] (serial upload only)

### Problem Description

Hey guys,
Im trying to send an email with my esp8266 with nodemcu LoLin v2 programmer board and my code worked fine in another sketch but I moved my functions and variables to another code which was my project's main code which is about 3 or 4 thousands long and it has one ISR function (Im suspicious about this one) and the code havent worked, I posted the code that has problems not the whole code obviously, and I've put the output for you too.

any help is appericiated, sorry if Im stupid :) 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <EEPROM.h>
#include <iostream>
#include <string>
#include <sstream>
#include <map>
#include <algorithm>
#include <vector>
#include <utility>
#include <functional> 
#include <cctype>
#include <locale>
#include <WiFiUdp.h>
#include <SPI.h>
#include <Wire.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

//variables and funciton declarations...

//define deug false, Serial log printing is disabled
//set to true for debugging
//WARNING: FALSE FOR RELEASE.
#ifndef DEBUG

#define DEBUG true 

#endif

[... 318 lines cut ...]
```

### Debug Messages
output
```

SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc

SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with Samaneh, channel 4
dhcp client start...
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
scandone
failed to connect to mail server
ssl_error: Unknown error code.

```

Jeroen88, 2019-08-25:
You should use one of the security mechanisms of Bearssl, see [here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino) for exmaples. If you are going to use server certificates you should also ensure that the ESP time is set.

earlephilhower, 2019-08-25:
If you enable debug messages you'll get the warning that no validations had been specified.  Check the bearssl_validation example.  Closing.

closed 2019-08-25 by earlephilhower

aryghm, 2019-08-26:
Hey guys thanks for response Im coding through the example you gave me, and I have a little problem in the example a const char[] is passed to the BearSSL::WiFiClientSecure::setFingerprint() function but in my code it gets a pointer of type const uint8_t, what is the problem here ?

thanks,

d-a-v, 2019-08-26:
`uint8_t` and `char` are both 8bits (and exactly the unsigned same in this core),
use `(const char*)your_uint8_t` or `reinterpret_cast<const char*>(your_uint8_t)`

aryghm, 2019-08-26:
> 
> 
> `uint8_t` and `char` are both 8bits (and exactly the unsigned same in this core),
> use `(const char*)your_uint8_t` or `reinterpret_cast<const char*>(your_uint8_t)`

Thanks it worked, but still I cant connect to smtp server through insecure and fingerprint option

aryghm, 2019-08-26:
> 
> 
> You should use one of the security mechanisms of Bearssl, see [here](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/BearSSL_Validation/BearSSL_Validation.ino) for examples. If you are going to use server certificates you should also ensure that the ESP time is set.

Hey, 
Well I tested 3 methods that were used in the example in order to use the bearssl client:
1. Insecure
2. Self-Signed
3. Fingerprint
The problem havent solved yet I think my problem is somewhere else although Im gonna test the other methods that were used in the example too but I think that the issue should not be closed

aryghm, 2019-08-26:
> 
> 
> If you enable debug messages you'll get the warning that no validations had been specified. Check the bearssl_validation example. Closing.

Do you mean setting debug level and debug serial port from "enable debug messages" or something else? cause I've already set those but still no change.

d-a-v, 2019-08-26:
Show you debug messages with this the following debug option enabled:
`Tools>Debug Level>(line before NDEBUG)`

d-a-v, 2019-08-27:
You should try with v2.5.2 of this arduino core.
You are using v2.4.2.

aryghm, 2019-08-27:
> 
> 
> Show you debug messages with this the following debug option enabled:
> `Tools>Debug Level>(line before NDEBUG)`

Hello, I did it still no change, those new lines are for the NTP time sync function and are added by myself no debug messages from BearSSL or something like that

Debug messages:
```
SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc

SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

connected with Samaneh, channel 7
dhcp client start...
ip:192.168.1.101,mask:255.255.255.0,gw:192.168.1.1
scandone
Waiting for NTP time sync: ..
Current time: Tue Aug 27 08:20:26 2019
failed to connect to mail server
ssl_error: Unknown error code.
```

aryghm, 2019-08-27:
> 
> 
> You should try with v2.5.2 of this arduino core.
> You are using v2.4.2.
Changing core version seems to take a long time for me as I have busy loops all over my code,

I'll let you know when I've done it

aryghm, 2019-08-28:
> 
> 
> You should try with v2.5.2 of this arduino core.
> You are using v2.4.2.

Hey uhhh.... I have a bit of problem with updating to 2.5.2 core version, whenever I run a sketch on my esp8266 nodemcu lolin v3 board I get this in my serial monitor repeatedly.

```
ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v8b899c12
~ld

```

Any ideas?

Thanks.

aryghm, 2019-08-30:
I reuploaded it and configurations are set to wipe the sketch only I believe.

> 
> 
> Did you wipe the flash?

aryghm, 2019-08-30:
> 
> 
> Did you wipe the flash?

although I did set configs to wipe all flash contents and still no change

RE-Novice, 2020-07-06:
Hello,

I am new here and may be stupid enough to write a comment for a cold case.
Trying to send secure mail using "https://github.com/gpepe/esp8266-sendemail" I encounter a problem when using the latest board (2.7.1). It seems to work up to 2.4.2. I can send unsecure mails.
Am I doing something wrong? Do I need to adjust something?
Is the mail sketch outdated?

Thanks a lot.
     Ralf

PS: I that the right way to address such things?

devyte, 2020-07-06:
@RE-Novice look at examples on how to use wificlient and bearssl in general.
This type of question should be asked at a community forum like esp8266.com or stackoverflow, not here.

RE-Novice, 2020-07-09:
Sorry about that. I am a novice (here and with programming). Don't know much about SSL/Bear, etc.. Just want to send mails and need SSL as all popular provider require it.
Choosing the AxTLS option in WiFiClientSecure solved the problem. Don't know what the downside is.
Thanks anyway
     Ralf

devyte, 2020-07-09:
AxTLS has been pulled out in latest master branch. It has been deprecated for a long while due to being unstable and the underlying lib being stale without activity for many years.
The bssl WiFiClientSecure has almost the same api and usage is almost the same as the axtls WiFiClientSecure. Again, look at the examples to figure out how to use it properly.

RE-Novice, 2020-07-10:
OK devyte, you convinced me (but only because I figured something out fairliy quickly). I believe I use BearSSL now (reverted to the original *.h).
client->setInsecure();
That line did the trick.
I think, I should be fine now.
Hopefully this helps others who as unexperienced as me.

Thanks a lot for checking and being insistant.
     Ralf