esp8266/Arduino #6447: Wire + SoftwareSerial

OpreaFlorin, 2019-08-25:
I am using Arduino Uno R3 board and Adafruit FONA 800 Shield - Voice/Data Cellular GSM for Arduino together.
I am sending commands to SIM800 and I am receiving responses.
To do this I am using a SoftwareSerial  (pins 3 and 4). So far so good.
This module is designed to be a slave into a I2C network where another Arduino Uno R3  is the master (only one master, only one slave). Server-slave examples works perfectly . The problem appears when I communicate with FONA 800 Shield using SoftwareSerial  and in the samme time communicate with master by using Wire.

This is the master code:
```
#include <Wire.h>

void setup() {
  Serial.begin(115200);
  Wire.begin(); // join i2c bus (address optional for master)
}

byte x = 0;

void loop() {
  Wire.beginTransmission(8); // transmit to device #8
  Wire.write(x);        // sends five bytes
  Wire.endTransmission();    // stop transmitting
  x++;
  delay(1);
}
```
For the slave, the code is simplified : 
```
void setup() {
  Wire.begin(8);                // join i2c bus with address
  Wire.onReceive(receiveEvent); // register event
}

// function that executes whenever data is received from master
// this function is registered as an event, see setup()
void receiveEvent(int howMany) {
	while(Wire.available()) Wire.read();
}

void loop() {
	// This is where I communicate with FONA 800 using SoftwareSerial
}
```
The problem is that the input from SoftwareSerial is altered when I2C is working. I am using SDA/SCL pins and a common GND for I2C and pin 3-4 for FONA. When a communication  takes place in I2C, the interrupt is messing up the input from SoftwareSerial , receiving altered chars (garbage).
If I send response to master (write), the output communication with SoftwareSerial is also destroyed, altering characters. (attering = changing, not adding or cutting characters)

I switch the FONA on Arduino Mega, using different pins - same result. 
Only using a Serial1 instead of a SoftwareSerial makes the those problems disappear.

If a decrease the speed for SoftwareSerial, the problems appears more rarely. So, the conclusion is that i cannot use Wire in conjunction with SoftwareSerial because SoftwareSerial will be messed up. I believe that under the hood Wire and SoftwareSerial are using someting common that makes interfere with each other.
If you can see the I2C communication is intense to accentuate this behaviour, Also, I rarely miss a byte sent by master.
How can I help you improve this behaviour?

devyte, 2019-08-25:
This is the ESP8266 core repo, unrelated to the UNO.
Closing.

closed 2019-08-25 by devyte

OpreaFlorin, 2019-08-25:
Sorry. Can you point me the right place to post it? I found a lot of places with Wire library ...

jandrassy, 2019-08-25:
> Sorry. Can you point me the right place to post it? I found a lot of places with Wire library ...

https://forum.arduino.cc/