esp8266/Arduino #6448: ESP8266 does not start when Pin 16 is connected to ground

hasenradball, 2019-08-26:
If I write an code where a Ticker is used and I reset the ESP while the pin16 is connected to ground (for deep sleep functionality) the ESP will not start.

Is this a normal behaviour when using the Ticker functionality?

What is the difference between the Ticker attach_ms() and attach_ms scheduled()?
Is scheduled run only once?

Maybe a small readme woulde be nice :-)

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266
- Core Version:			2.5.2
- Development Env:		[Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			dio
- Flash Size: 		        4MB
- lwip Variant:                  ck|nodemcu
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:	       SERIAL
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

jandrassy, 2019-08-26:
if io 16 is connected to reset and you connect io 16 to ground, then reset is connected to ground and the esp8266 is in reset

hasenradball, 2019-08-26:
Sorry wrong explanation.
`GPIO16` is connected to `RESET` Pin.

hasenradball, 2019-08-26:
But when I restart via reset button why does the chip make a loop with restarts when I connect GPIO16 to RESET?
This is not the case without Ticker!?!

hasenradball, 2019-08-26:
Is there an other behaviour to the `GPIO16` Pin when using Ticker during startup compared not using Ticker?

closed 2019-08-26 by hasenradball