esp8266/Arduino #6451: Documentation request: Ticker, attach_ms and attach_ms_scheduled

d-a-v, 2019-08-26:
Hint: [`Ticker/TickerFunctional.ino`](https://github.com/esp8266/Arduino/tree/master/libraries/Ticker/examples/TickerFunctional) example
This [conversation](https://gitter.im/esp8266/Arduino?at=5d63f68007d1ff39f8889a92)

tom-ch1, 2023-10-16:
The linked conversation does not help. It points to a matrix room with discussions of various things. I'd love to see the discussion here, so that anybody can follow it

tom-ch1, 2023-10-16:
Here's the documentation of Ticker: [https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#ticker](https://arduino-esp8266.readthedocs.io/en/latest/libraries.html#ticker)

> It is currently not recommended to do blocking IO operations (network, serial, file) from Ticker callback functions. Instead, set a flag inside the ticker callback and check for that flag inside the loop function.

Here's the source code of [attach_scheduled](https://github.com/esp8266/Arduino/blob/master/libraries/Ticker/src/Ticker.h#L63):

> // callback will be called at following loop() after ticker fires

As there is no documentation other than the source code and the examples, one would deduce that using `attach_scheduled` resolves the warning given in the recommendation

Is that correct?

According to the Example `TickerFunctional.ino`, there is no need for any funcion call inside `loop()`, and nevertheless, the callback is called inside `loop()`

Is that correct?

mcspr, 2023-10-16:
> As there is no documentation other than the source code and the examples, one would deduce that using attach_scheduled resolves the warning given in the recommendation

> Is that correct?

Right, `_scheduled` are the exception. Note the code flow - scheduled funcs are either called after loop() normally returns, or when yield() suspends loop() to run SDK tasks
https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/cores/esp8266/core_esp8266_main.cpp#L247-L251
https://github.com/esp8266/Arduino/blob/1662248b394740d82824f504a021d4dad3d314b6/cores/esp8266/core_esp8266_main.cpp#L262-L263

> As there is no documentation other than the source code and the examples, one would deduce that using attach_scheduled resolves the warning given in the recommendation
>
> Is that correct?

attach\_ms callback would execute whenever SDK decides it is time for the timer to expire and **in that context**. For attach\_ms\_scheduled we effectively do `schedule_function([user_callback]() { user_callback(); })`