esp8266/Arduino #6463: MDNS Not working properly after this commit https://github.com/esp8266/Arduino/commit/f9009b8a5e141235d3380cc6c6c06223008a1099

arihantdaga, 2019-08-29:
### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ X] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			         ESP-12
- Core Version:			85f1ea7c78a583b7a414f016a71d14c1c91099d5
- Development Env:		Arduino IDE
- Operating System:		MacOS

### Settings in IDE

- Module:  			NodeMcu 1.0
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v2 Lower Higher Bandwidth]
- Upload Using:		SERIAL

### Problem Description
MDNS isn't working continuously after this commit https://github.com/esp8266/Arduino/commit/f9009b8a5e141235d3380cc6c6c06223008a1099. 
Problem Scenario -
After calling mdns.begin() once, it says ok. Now if a new device which is running a service discovery application, Joins the wifi network, it cannot find the .local service and cannot connect to the esp node using .local hostname. 
I am attaching a screenshot from my service discovery browser, which publishes a remove event for this esp node. 

This was working fine previous to this commit, i have verified that. 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
  ESP8266 mDNS responder sample

  This is an example of an HTTP server that is accessible
  via http://esp8266.local URL thanks to mDNS responder.

  Instructions:
  - Update WiFi SSID and password as necessary.
  - Flash the sketch to the ESP8266 board
  - Install host software:
    - For Linux, install Avahi (http://avahi.org/).
    - For Windows, install Bonjour (http://www.apple.com/support/bonjour/).
    - For Mac OSX and iOS support is built in through Bonjour already.
  - Point your browser to http://esp8266.local, you should see a response.

*/

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiClient.h>

#ifndef STASSID
#define STASSID "MYSSID"
#define STAPSK  "MYPASS"
#endif

unsigned long last_done = millis();
const char* ssid = STASSID;
const char* password = STAPSK;

// TCP server at port 80 will respond to HTTP requests
WiFiServer server(80);

void setup(void) {
  Serial.begin(115200);
  delayInit(2);

  // Connect to WiFi network
  WiFi.mode(WIFI_STA);
  connectToWifi();

  // Set up mDNS responder:
  // - first argument is the domain name, in this example
  //   the fully-qualified domain name is "esp8266.local"
  // - second argument is the IP address to advertise
  //   we send our IP address on the WiFi network
  if (!MDNS.begin("esp8266_special_xx")) {
    Serial.println("Error setting up MDNS responder!");
    while (1) {
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");

  // Start TCP (HTTP) server
  server.begin();
  Serial.println("TCP server started");

  // Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
}

void loop(void) {
  MDNS.update();

  // Check if a client has connected
  WiFiClient client = server.available();
  if (!client) {
    return;
  }
  Serial.println("");
  Serial.println("New client");

  // Wait for data from client to become available
  while (client.connected() && !client.available()) {
    delay(1);
  }

  // Read the first line of HTTP request
  String req = client.readStringUntil('\r');

  // First line of HTTP request looks like "GET /path HTTP/1.1"
  // Retrieve the "/path" part by finding the spaces
  int addr_start = req.indexOf(' ');
  int addr_end = req.indexOf(' ', addr_start + 1);
  if (addr_start == -1 || addr_end == -1) {
    Serial.print("Invalid request: ");
    Serial.println(req);
    return;
  }
  req = req.substring(addr_start + 1, addr_end);
  Serial.print("Request: ");
  Serial.println(req);
  client.flush();

  String s;
  if (req == "/") {
    IPAddress ip = WiFi.localIP();
    String ipStr = String(ip[0]) + '.' + String(ip[1]) + '.' + String(ip[2]) + '.' + String(ip[3]);
    s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>Hello from ESP8266 at ";
    s += ipStr;
    s += "</html>\r\n\r\n";
    Serial.println("Sending 200");
  } else {
    s = "HTTP/1.1 404 Not Found\r\n\r\n";
    Serial.println("Sending 404");
  }
  client.print(s);

  Serial.println("Done with client");
}

```

d-a-v, 2019-08-29:
Would you be able to test with #6261 ?

arihantdaga, 2019-08-29:
@d-a-v Tried it today with #6261 . It has the same problem.

arihantdaga, 2019-09-04:
@d-a-v . I checked it latest today. Its working fine with this pull request.

d-a-v, 2019-09-04:
@arihantdaga with or without #6261 ?

arihantdaga, 2019-09-04:
With #6261

PR #6261 (2019-09-05): MDNS: fix random crash on startup
MDNS:
- AP address is used by default when STA is also present 
- enforce selected interface
- igmp: force selected interface

Arduino menu:
- mDNS debug option

*edit*: 
Fixes #6463
closed 2019-09-05 by earlephilhower