esp8266/Arduino #6467: Windows build busted with missing \Python3\ directory

earlephilhower, 2019-08-29:
Opening for @Mikekgr.

The Python3 merge broke Windows due to a renaming of the installed python directory.

See for full details:  https://github.com/esp8266/Arduino/pull/6378#issuecomment-526284744

Either the python.zip file for Windows needs to be moved on the toolchain dir to \python3\ (simplest), or the platform.txt needs to not use {python} as the subdir.

earlephilhower, 2019-08-30:
@mikekgr, please give #6472 a try.  You will need to re-execute `get.py` and it will download a properly named python3-xxx.zip file and install into the appropriate spot.  No renaming should be needed.

mikekgr, 2019-08-30:
Dear @earlephilhower , I happily confirm that it is working fine now. Many many thanks !

earlephilhower, 2019-08-30:
Merged and confirmed.  Closing.

closed 2019-08-30 by earlephilhower