esp8266/Arduino #6469: HTTP_UPDATE_FAILD Error

wounder123, 2019-08-30:
HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 302 Found'
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 30 Aug 2019 08:57:11 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'Location: https://xyz.co.in/ota/WIFI/WIFI.ino.bin'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 231'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 302
[HTTP-Client][handleHeaderResponse] size: 231
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 302
[httpUpdate]  - len: 231
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 3866624
[httpUpdate]  - current Sketch Size: 306560
[httpUpdate] HTTP Code is (302)
[HTTP-Client][end] still data in buffer (231), clean up.
[HTTP-Client][end] tcp stop
:ur 1
:del
HTTP_UPDATE_FAILD Error (-104): Wrong HTTP Code
pm open,type:2 0

why this error come in serial window(Arduino)?

/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

   #include <Arduino.h>

   #include <ESP8266WiFi.h>
   #include <ESP8266WiFiMulti.h>

    #include <ESP8266HTTPClient.h>
    #include <ESP8266httpUpdate.h>

    #define USE_SERIAL Serial

    #ifndef APSSID
    #define APSSID "JioFi2_0FC2A2"
    #define APPSK  "rbmibqpqz5"
    #endif

   ESP8266WiFiMulti WiFiMulti;

   void setup() {

    USE_SERIAL.begin(115200);
   // USE_SERIAL.setDebugOutput(true);

  USE_SERIAL.println();
  USE_SERIAL.println();
  USE_SERIAL.println();

  for (uint8_t t = 4; t > 0; t--) {
    USE_SERIAL.printf("[SETUP] WAIT %d...\n", t);
    USE_SERIAL.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(APSSID, APPSK);
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    ESPhttpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = ESPhttpUpdate.update( "http://xyz.co.in/_ota/WIFI/WIFI.ino.bin");
    // Or:
    //t_httpUpdate_return ret = ESPhttpUpdate.update(client, "server", 80, "file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        USE_SERIAL.printf("HTTP_UPDATE_FAILD Error (%d): %s\n", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        USE_SERIAL.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        USE_SERIAL.println("HTTP_UPDATE_OK");
        break;
    }
  }

}

void loop() {
  
}

Jeroen88, 2019-08-30:
Your server sends a HTTP code 302:

> HTTP-Client][handleHeaderResponse] RX: 'HTTP/1.1 302 Found'

Update will start if your server sends a HTTP code OK: 200. Possibly the URL you are using is being redirected. Try:

``
ESPhttpUpdate.followRedirects(true);
``

See also issue #6304

wounder123, 2019-08-30:
after compiling code : 
 exit status 1
'class ESP8266HTTPUpdate' has no member named 'followRedirects'

why??

 i am using arduino 1.8.7,core version 2.5.0

Jeroen88, 2019-08-30:
Version 2.5.2 has it, see [here](https://github.com/esp8266/Arduino/blob/d04f7689541ca5cf1fb00d73ca067c2aa651ebd6/libraries/ESP8266httpUpdate/src/ESP8266httpUpdate.h#L77)

wounder123, 2019-08-30:
still have return 302.??
Serial window:::

:abort
[HTTP-Client][handleHeaderResponse] RX**: 'HTTP/1.1 302 Found'**
[HTTP-Client][handleHeaderResponse] RX: 'Date: Fri, 30 Aug 2019 16:40:12 GMT'
[HTTP-Client][handleHeaderResponse] RX: 'Server: Apache'
[HTTP-Client][handleHeaderResponse] RX: 'Location: https://palmtec.co.in/m66_ota/WIFI/WIFI.ino.bin'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Length: 231'
[HTTP-Client][handleHeaderResponse] RX: 'Connection: close'
[HTTP-Client][handleHeaderResponse] RX: 'Content-Type: text/html; charset=iso-8859-1'
[HTTP-Client][handleHeaderResponse] RX: ''
[HTTP-Client][handleHeaderResponse] code: 302
[HTTP-Client][handleHeaderResponse] size: 231
[httpUpdate] Header read fin.
[httpUpdate] Server header:
[httpUpdate]  - code: 302
[httpUpdate]  - len: 231
[httpUpdate] ESP8266 info:
[httpUpdate]  - free Space: 3862528
[httpUpdate]  - current Sketch Size: 308016
[httpUpdate] HTTP Code is (302)
[HTTP-Client][end] still data in buffer (231), clean up.
:c0 1, 450
[HTTP-Client][end] tcp stop
:ur 1
:del