esp8266/Arduino #6475: Use of #ifdef not processed consistently

TD-er, 2019-09-01:
I am not really sure if this is something I am lacking in my knowledge of how Arduino does things with generating the .ino.cpp file, or maybe it is a bug.
Either way I am having build issues which look like `#ifdef` statements are not processed like they should (or at least as how I think they should be processed)

For example, in the main `project.ino` file, I had a forward declaration of a function and a function wrapped in an `#ifdef` like this:

```c++
#ifdef USES_BLYNK
// Blynk_get prototype
boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );

int firstEnabledBlynkController() {
  return -1;
}
#endif
```

This does result in the generated file `project.ino.cpp` which has the whole list of function forwarde declarations:

```c++
void bla1();
#ifdef USES_BLYNK
boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );
void otherbla();
void otherbla2();
....
```

It does not include the #endif in the forward declarations.
And thus the build does fail with lots and lots of errors since all other forward declarations are not found for obvious reasons.

If I move this part to a .h file, it all seems to work just fine.

Not sure if it has anything to do with it, but this define is not defined at the call of the compile, but is set dynamically based on other defines in one of the first included .h files.

d-a-v, 2019-09-01:
They recently updated arduino-builder.
Are you using a recent nightly ?
Does it work differently with latest stable arduino ide ?

They have their own pre-preprocessor that sometimes has to be tamed or workarounded

jandrassy, 2019-09-01:
is the #endif after `int firstEnabledBlynkController();`?

TD-er, 2019-09-01:
The #endif was not directly after the forward declaration, but after the function, just like in the example
And I am not using ArduinoIDE, but PlatformIO (latest stable)
This was while building core 2.5.2 for this example.

jandrassy, 2019-09-01:
in project.ino.cpp is the #endif after int firstEnabledBlynkController();?

TD-er, 2019-09-01:
No, just like I posted:

```c++
#ifdef USES_BLYNK
// Blynk_get prototype
boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );

int firstEnabledBlynkController() {
  return -1;
}
#endif
```

I wrapped the lines used here:
https://github.com/letscontrolit/ESPEasy/blob/298d45e984a1ff6eda6e797899ac42e8f5991468/src/ESPEasy.ino#L92-L103

TD-er, 2019-09-01:
This then generates the ESPeasy.ino.cpp like this:

```c++
# 1 "c:\\users\\gijs\\appdata\\local\\temp\\tmppyht7z"
#include <Arduino.h>
# 1 "C:/GitHub/TD-er/ESPEasy/src/ESPEasy.ino"

#include <Arduino.h>

#ifdef CONTINUOUS_INTEGRATION
#pragma GCC diagnostic error "-Wall"
#else
#pragma GCC diagnostic warning "-Wall"
#endif

#ifdef PLUGIN_SET_GENERIC_ESP32
  #ifndef ESP32
    #define ESP32
  #endif
#endif
# 86 "C:/GitHub/TD-er/ESPEasy/src/ESPEasy.ino"
#include "ESPEasy-Globals.h"

#include "_Plugin_Helper.h"

#include "_CPlugin_Helper.h"

#ifdef USES_BLYNK

boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );
int firstEnabledBlynkController();
void preinit();
void sw_watchdog_callback(void *arg);
void setup();
void RTOS_TaskServers( void * parameter );
void RTOS_TaskSerial( void * parameter );
```

TD-er, 2019-09-01:
> They recently updated arduino-builder.
> Are you using a recent nightly ?
> Does it work differently with latest stable arduino ide ?
> 
> They have their own pre-preprocessor that sometimes has to be tamed or workarounded

What is "recently"?
I got report last week from one of the users that he also had some issues building it in Arduino IDE.
His words: It does include way more than needed and should be included.

I have the same feeling, like we're now hitting the max. sketch size on almost all builds now. (regardless of what is included/excluded using defines)
Also we had some very strange build issues when trying to exclude some structs using define, but then the compiler is complaining about not knowing about some of these structs in functions that should not be linked/used anyway.
Those functions were using the excluded classes as an argument, so had to be forward declared.

If this is a bug in the way how #ifdef defines are being processed to generate the CPP, then it all starts to make sense now.

jandrassy, 2019-09-01:
so `Blynk_get` is not followed by void `otherbla();` in .ino.cpp but there is the `int firstEnabledBlynkController();`

TD-er, 2019-09-01:
I tried to make it not too specific and thus used other names. But apparently I made it more confusing by doing this.

Sorry 'bout that :(

These two comments should give a proper indication of what's happening:
https://github.com/esp8266/Arduino/issues/6475#issuecomment-526909957
https://github.com/esp8266/Arduino/issues/6475#issuecomment-526910126

In short, there is a forward declaration and a function definition (without forward declaration) and both are wrapped in the #ifdef  
The #endif statement does not appear after the forward declaration of both functions in the generated .ino.cpp file.

I guess it can be solved (for now) to explicitly forward declare these functions, but still there does seem to be a bug (or inconsistency) which does not copy the #ifdef wrappers for forward declarations in the .ino.cpp file.

jandrassy, 2019-09-01:
Arduino builder doesn't put #ifdefs and lines exclude with #ifdefs into ino.cpp

TD-er, 2019-09-01:
@jandrassy What script is then generating the forward declarations into the .ino.cpp file?
If I look into my code, there is no explicit forward declaration of `int firstEnabledBlynkController();` but a line like that does show up in the .ino.cpp file.

And it seems that generation process is causing issues here.
Or I'm using it incorrect by not explicitly declaring the forward declaration, that's also a possibility.

TD-er, 2019-09-01:
Just tested by adding this to ESPEasy.ino:
```c++
#include "_CPlugin_Helper.h"

#ifdef USES_BLYNK
// Blynk_get prototype
boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );

int firstEnabledBlynkController();
#endif

//void checkRAM( const __FlashStringHelper* flashString);

#ifdef CORE_POST_2_5_0
/*********************************************************************************************\
 * Pre-init
\*********************************************************************************************/
void preinit() {
```

Does generate:
```c++
#include "_CPlugin_Helper.h"

#ifdef USES_BLYNK

boolean Blynk_get(const String& command, byte controllerIndex,float *data = NULL );

int firstEnabledBlynkController();
#endif

#ifdef CORE_POST_2_5_0
void preinit();
```

So for that function it is now working properly.
I do see however that the next function (`preinit()`) is the next to cause issues, since that #ifdef is also not handled correctly.

jandrassy, 2019-09-01:
https://github.com/arduino/arduino-builder
but I think platformio doesn't use this

TD-er, 2019-09-01:
I will have a look.  Also will try and see what the output is of this Arduino-builder.
Maybe @ivankravets can tell us more?

ivankravets, 2019-09-02:
Could someone provide a simple PlatformIO project to reproduce this issue? In any case, we recommend using direct `*.cpp` instead of `*.ino` to avoid any issues. we do some "virtual preprocessing" and convert INO to CPP. However, it's a legacy mode and is better if the developer personally controls project files. If I remember, Arduino IDE should support original C/C++ files too.

uzi18, 2019-09-02:
@ivankravets how to debug such problem in platformio code?

TD-er, 2019-09-02:
Not really a very simple project at hand here, but I guess this should be enough to have in the project.ino file.

Just use the https://github.com/esp8266/Arduino/blob/74819a763bfb6e9890a57411dcea4aba221a778d/libraries/esp8266/examples/Blink/Blink.ino

and add some function like this in the .ino file:

```c++
#ifdef DO_NOT_USE
void thisShouldNotAppearInTheBinary() {
  delay(1000);
}
#endif
```

As long as you don't have the `DO_NOT_USE` flag set, you should not see it in the generated .ino.cpp file.
Well at least, that's what I expect should happen.

Things will fail to build if you have some object used as parameter which is only defined when a flag is set. For example:

```c++
#ifdef DO_NOT_USE
typedef int32_t delaytype;
void thisShouldNotAppearInTheBinary(delaytype timer) {
  delay(timer);
}
#endif
```

ivankravets, 2019-09-02:
It seems like our bug. Please file an issue here https://github.com/platformio/platformio-core/issues and put a link to this issue. Thanks!

TD-er, 2019-09-02:
@ivankravets Do you have any idea when this issue may have been introduced, or was it present for a long time?

I have got the idea my builds increased in size dramatically the last week or weeks.
If this has been an issue for a long time, then I must continue to look for other possible explanations.

uzi18, 2019-09-02:
@TD-er have reports about 1/3 (now it is max 10%) more in about year or more, but we can check this with released bins.

earlephilhower, 2019-09-27:
Not sure there's anything here to do, it's all on the PIO side and it looks like they've identified several workarounds.  Closing the Arduino Core side of this.

closed 2019-09-27 by earlephilhower