esp8266/Arduino #6481: Send audio streaming from NodeMCU to Firebase

chathuravithakshana, 2019-09-03:
I need to send audio stream as .wav format from NodeMCU to Firebase, so that later it can be access to Firebase and streaming stored .wav to Android app. So first I need to know how to send audio stream continuously from NodeMCU to Firebase? Any help is highly appreciate.

d-a-v, 2019-09-03:
You can use `WiFiClient` to establish a TCP connection to a TCP server, and `client.write()` to it.
Or `WiFiUdp`  for UDP connections.
Check examples in `ESP8266WiFi` library on how to use them, and official arduino documentation on the base the class `Client`.
This is not an issue, so I suggest going to https://esp8266.com or https://gitter.im/esp8266/Arduino for more help.

closed 2019-09-03 by d-a-v

chathuravithakshana, 2019-09-03:
thank u @d-a-v