esp8266/Arduino #6482: I belive you got a bug...

offcircuit, 2019-09-03:
Hi,

I dont know if this is a bug

SoftwareSerial.h

line 113

#ifndef ESP32
ssize_t m_swsInstsIdx = -1; <---- this should be size_t m_swsInstsIdx
#endif

I found this at

file
C:\Users\offcircuit\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h

dowloaded from

http://arduino.esp8266.com/stable/package_esp8266com_index.json

error

In file included from sketch\Nextion.h:7:0,

             from sketch\Nextion.cpp:1:
C:\Users\offcircuit\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h:113:2: error: 'ssize_t' does not name a type

ssize_t m_swsInstsIdx = -1;

^

exit status 1
Erro ao compilar para a placa LOLIN(WEMOS) D1 R2 & mini.

Thanks,
Helder Rodrigues

d-a-v, 2019-09-03:
cc @dok-net

earlephilhower, 2019-09-03:
`ssize_t` is legal and a signed-size type, `size_t` is unsigned, so it's not a matter of a typo there.

This sounds like just a case of a missing `#include <sys/types.h>` in `SioftwareSerial.h`

earlephilhower, 2019-09-15:
@dok-net, a user should not be using library manager to get SWSerial.  It's part of the GIT repo as a submodule, so it's actually checked out in the git core tree.  Use the web interface, you can see the actual git rev, and that it's coming from plreup's repo, not yours.  So the change would need to be in his git repo, not your fork.

The user reported this path as the error:
` C:\Users\offcircuit\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.2\libraries\SoftwareSerial\src/SoftwareSerial.h` 
So that's the correct location.

dok-net, 2019-09-15:
Dear @earlephilhower, names can be misleading... the current developer and one active maintainer of https://github.com/plerup/espsoftwareserial/, is... well, look closer... hint... you've just kind of p*** him off ;-)

earlephilhower, 2019-09-15:
Ah, I looked at the URL and owner, not at the commit log.  Sorry, mate!

Anyway, they're not using the lib manager to install the repo, it's the version included in the release.zip from esp8266 as far as I can see.  Maybe they're using it in a way that isn't standard, so it is included before an Arduino.h or something, so ssize_t isn't defined yet.

If the ssize thing was fixed after 2.5.2, then this can be closed.  But I don't see sys/types.h included anywhere on my quick perusal.  Just adding it to the main header and bumping the git reference in the 8266 tree here (w/a PR here) would take care of it, no?

dok-net, 2019-09-15:
ssize_t was dropped some time ago, probably in conjunction with the acceptance of the attachInterruptArg() PR.

earlephilhower, 2019-09-16:
Ah, got it.  Then, @offcircuit , you need to pull the latest GIT head for the ESP8266 Arduino repo, then do a "git submodule init; git submodule update" because it looks like you have a repo that's not quite git head and not quite an official release 2.5.2.

Closing, there's nothing to do here.

closed 2019-09-16 by earlephilhower