esp8266/Arduino #6488: Static IP does not persist after reboot

jpboucher, 2019-09-05:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266-wroom2]
- Core Version:			[latest git hash]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Size: 		[2MB (1MB SPIFFS)]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

In the ESP8266WiFiSTAClass::begin() method, if the _persistent variable is true, saved wifi configurations will be loaded after reboot. The problem is that the _useStaticIp variable will be set on false after reboot even if you used a static IP prior to rebooting. This means that the DHCP will start and overwrite past static IP settings even if it's supposed to be persistent.

Right now the _useStaticIp variable will only be set using ESP8266WiFiSTAClass::config() which needs IP settings as parameters which is not something you want to call again as your settings were supposed to stay persistent.

jandrassy, 2019-09-05:
the SDK doesn't support static IP configuration persistence. so if you want to use static IP turn of WiFi connection persistence and setup the connection in `setup()`

jpboucher, 2019-09-06:
This would explain the behavior. The SDK also seems to automatically start the DHCP on boot if there's an address in flash memory. (Please confirm this) That is problematic since I am trying to swap from DHCP to static between each boot. Since a previous boot on DHCP with persistence wrote on flash memory even if I try later to go back to static it won't work because the DHCP starts and uses my old address. 

Of course I could wait and then force a change to a static address but that's probably not the optimal solution.

d-a-v, 2019-09-06:
> Or is there any way to disable DHCP on bootup somehow...

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi/examples/EarlyDisableWiFi

So you can do all the work yourself as soon as `begin()` is called, in non-persistent mode, with the help of EEPROM or LittleFS.

jpboucher, 2019-09-06:
@d-a-v Thanks, this is very useful. Like jandrassy said, the issue is a limitation on the espressif SDK side. I have been able to find a workaround so I will close this issue.

closed 2019-09-06 by jpboucher