esp8266/Arduino #6491: The macro __FILE__ does changed?

mikekgr, 2019-09-07:
Dear Sirs,
I am using Arduino 1.8.8 at Windows OS having ESP8266 core Git master updated.
I notice that recently the macro __FILE__ does not include the full path + the filename. That was the case recently. Now the command:
Serial.printf(" Source File Name: %s\n",__FILE__);
gives:  Source File Name: testSketch.ino
before was:   Source File Name: C:\ArduinoIDE\sketchbook\testSketch\testSketch.ino
Do you know if this is an error or change in core plans?
Can I revert back the __FILE__ macro contains the full path as well as the filename?
Thanks and Best Regaards
Mike Kranidis

d-a-v, 2019-09-07:
It's a feature (a "change in core plans") !
It is also a step towards reproducible generation of flash binaries,
it also offers smaller flash occupation with debug messages,
privacy for pre-generated core libraries,

What is your use case ?

mikekgr, 2019-09-07:
Thanks for the fast answer.
My use case is to know exactly from were my source file coming from. Because I have same naming in Arduino Projects but in different locations ex external USB disks vs internal. Is there anything I can do in order to have the previous behavior? 
Thanks and Best Regards,
Mike Kranidis

d-a-v, 2019-09-07:
This feature has been awaited for a long time
(try to google for [`gcc __FILE__ full path`](https://www.google.com/search?q=gcc+__FILE__+full+path), check also #4520).
Thanks to @earlephilhower who did this job by patching gcc.

> Is there anything I can do in order to have the previous behavior?

I'd use a personal git repository to handle everything: versioning, backuping, sharing common code between two closely-looking source code and a string to differentiate them, ...

mikekgr, 2019-09-07:
@d-a-v Dear Sir, I totally understand the need to patch the gcc in order to strip out the path part of the filename.
I understand your suggestion regarding the personal git, I have it already. This change is not convenient to me but it is not a big deal.
Thanks a lot for the answer and the suggestion.

Best Regards,
Mike Kranidis

earlephilhower, 2019-09-11:
Looks like this has been settled.   Closing.

closed 2019-09-11 by earlephilhower

WeaVaer, 2020-12-01:
Well,
provided __FILE__ macro has changed stripping path info for the aforementioned reasons,
the need to have a macro giving full path to the source code still persists for various previously mentioned reasons here or in other blog entries for different people. I humbly am among them.
Something named like __PATH__ maybe could be very useful in our situation..
It shouldn't be too difficult provided the stripping could be done effortlesly with expertise in the __FILE__ macro to provide a macro which gives us the path (or a macro that gives full path + filename just like before).
Or show a method to get the full path of compiled source code into the sketch to be utilized during runtime.
We need at least one of this proposed solutions come true.
?

WeaVaer, 2020-12-01:
@earlephilhower 
Could you please point out exactly where in source you applied the gcc patch,
so I can revert it back to standard as it was back in older versions of the core ?
I need filenames and line numbers or a diff file if you prefer.

This task is crucial for me to sustain a few distributed products which does not work anymore because of this change since it was highly dependent on \_\_FILE\_\_ having full path.

Thank you