esp8266/Arduino #6495: Something wrong with WiFi.config(...) and WiFi.status()

ruggi99, 2019-09-08:
# Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[2.5.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio|dio|other] (Don't know)
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

When I upload this sketch to my NodeMCU I see that WiFi.status() always returns WL_CONNECTED after a call to WiFi.config(...) and it doesn't if WiFi.config(...) is commented out. I use "Erase Sketch + WiFi settings" (option from arduino IDE's menu) each time I flash this sketch. 
Even if i put a delay(1000) after WiFi.config(...) it doesn't work. Is it a bug?
Couldn't try the sketch with the latest git because it fails (it says it cannot execute "tools/python3/python3" if I understood correctly)

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup() {
  Serial.begin(9600);
  Serial.println()
  Serial.println("Hello!");
  
  WiFi.mode(WIFI_STA);
  WiFi.config(IPAddress(102, 168, 1, 50), IPAddress(192, 168, 1, 1), IPAddress(255, 255, 255, 0));
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("Connected!");
  } else {
    Serial.println("Non connected!");
  }
}

void loop() {

}

```

### Debug Messages

```

Hello!
Connected!
```

5chufti, 2019-09-09:
I don't see any WiFi.begin(ssid,pwd) anywhere in your sketch?

ruggi99, 2019-09-09:
Great observation. Why should WiFi.status() return WL_CONNECTED if ESP isn't really connected to a wifi network? In my opinion it's a bug, isn't it? I repeat: I erase Sketch and wifi settings each time I upload this sketch and if Wifi.config(...) function is commented out, WiFi.status() function doesn't return WL_CONNECTED. Suggestions?

jandrassy, 2019-09-09:
the esp8266 connected to WiFi with persistent ssid and password
https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html?highlight=persistent#persistent

mbm60, 2019-09-09:
@ruggi99 Problem solved in latest git version.

ruggi99, 2019-09-10:
@mbm60 Thank you. Could you point me the PR?

mbm60, 2019-09-10:
PR #6194.

ruggi99, 2019-09-10:
Ok thank you. I searched the issue before opening this one but I didn't find it, my fault. Closing

closed 2019-09-10 by ruggi99