esp8266/Arduino #6497: I2C >> Bit Error when compiling for 160 MHz

detamend, 2019-09-09:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			Wemos D1 mini pro (ESP8266)
- Core Version:			2.5.2
- Development Env:		Arduino IDE
- Operating System:		Windows 10 x64

### Settings in IDE

- Module:  				Wemos D1 mini pro
- Flash Mode:			not sure (using IDE)
- Flash Size: 		4MB
- lwip Variant:   v2 Higher Bandwidth
- Reset Method:		not sure
- Flash Frequency:	not sure (IDE)
- CPU Frequency:		80Mhz AND 160MHz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Trying to let two ESPs talk to each other via I2C - easy, right? Obviosly not, when boths CPUs running at 160 MHZ.

SENDER SKETCH

```cpp

#include <Wire.h>

#define SDA_PIN 4
#define SCL_PIN 5

#define sender 0x42
#define reciver 0x08

void setup(){
  Serial.begin(115200);
  Wire.begin(SDA_PIN,SCL_PIN,sender);
}

void loop(){
  Wire.beginTransmission(reciver);
  Wire.write("aaAA");
  Wire.endTransmission();
  delay(1000);
}

```

RECIVER SKETCH

```cpp

#include <Wire.h>

#define SDA_PIN 4
#define SCL_PIN 5

#define sender 0x42
#define reciver 0x08

void setup(){
  Serial.begin(115200);
  Wire.begin(SDA_PIN,SCL_PIN,reciver);
  Wire.onReceive(receiveEvent);
}

void receiveEvent(size_t bytesRecived){
  (void) bytesRecived;
  String command;
  byte c;

  while(Wire.available()){
    c=Wire.read();
    Serial.print(c);
    Serial.print("\t");
    Serial.println(c,BIN);
    command+=(char)c;
  }
  Serial.println(command);
  Serial.println("---------");
}

void loop(){
  delay(1);
}

```

Compiled with both CPUs are running at 80 MHz, the reciver gets "aaAA" from the sender.

If I compile both scripts at 160 MHz CPU speed, the reciver gets 

```
?aAA
```

instead.

To get an idea, whats been send/recived, I dumped every recived by in the reciver script:

```
225	11100001
97	1100001
65	1000001
65	1000001
?aAA
```

So the first bit of the data should have been 0 - but gets recived as 1. This sets the MSB of the first byte to 1, changing the expected 97 to a 225.

I use two 2k2 resistors as pullups on the SCL/SDA lines. Sadly no logic analyzer around, so I can't pinpoint the sender or reciver as the culprit.

The boards are connected via a triple base board, only two CPU Modules are plugged in.

Any help would be greatly appreciated! I know: simply go back to 80 MHz... but I really like to get this setup working.

detamend, 2019-09-10:
Got my trusty Oscilloscope going and compared the signals for CPU@80MHz and CPU@160MHz. I'm sending to Adress 0x08 and transmitting a single Character: "S". S in binary ready 01010011.

On 80MHz I recive 01010011 (as expected).
On 160MHz I recive 11010011 (fist bit should be 0).

Besides the error, the timing is quite different. Running 80MHz, one cycle is roughly 23us. @160MHz a typical cycle is 17us.

On the screenshots of the oscilloscope it' s pretty obvious, that the first bit of 01010011 is not low when the CPU is running at 160MHz. 

[image]

Jason2866, 2019-09-10:
Maybe the same issue as https://github.com/esp8266/Arduino/issues/5762

detamend, 2019-09-10:
> 
> 
> Maybe the same issue as #5762

I read that issue - definitly related.

TD-er, 2019-10-03:
Maybe this one is related?
https://github.com/esp8266/Arduino/pull/6592

The PR description has a lot of links referring to the original issue and a detailed readme page describing what the original author tries to fix.

detamend, 2019-10-14:
@TD-er - definitly related. Just replaced the core_esp8266_si2c.cpp with the one from the PR - now works like a charm.

Great Job, guys!

devyte, 2019-11-09:
#6592 is merged and released.
Closing.

closed 2019-11-09 by devyte

laercionit, 2020-07-11:
I am using CORE 2.7.2 I have the same problem, anyone else?