esp8266/Arduino #6500: nothing happens after uploading sketch on ESP8266

pouriyae, 2019-09-10:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-8266]
- Core Version:			[2.5]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[512k]
- lwip Variant:   [v2prebuilt mss536]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

i tried to upload the sketch for relay on ESP8266 and after uploading done nothing changed on my module! i also tried to upload through FT232RL and uno r3 but same thing happens all the time!the version of module is latest and connections are listed below.
the power suply is FT232RL and data comes from UNO R3
VCC->3v3 and RST and EN
GND->GND and IO0
RX->RX
TX->TX

please help me to solve this issue

cziter15, 2019-09-11:
RX->TX
TX->RX

Have fun :)

pouriyae, 2019-09-11:
> RX->TX
> TX->RX
> 
> Why? Because if you are sending/transmitting, you are doing it **from** TX pin, when you receive it comes **to** RX pin. Have fun!

this is what i get when i connect RX->TX l TX->RX
_________________________________________________
Sketch uses 260287 bytes (52%) of program storage space. Maximum is 499696 bytes.
Global variables use 34168 bytes (41%) of dynamic memory, leaving 47752 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

jandrassy, 2019-09-11:
what do you do with io 0?

pouriyae, 2019-09-11:
> what do you do with io 0?

IO0 is connected to GND while flashing and after that disconnect it from GND!

jandrassy, 2019-09-11:
> > what do you do with io 0?
> 
> IO0 is connected to GND while flashing and after that disconnect it from GND!

it is underpowered.
the Uno has a stronger 3.3 pin then the FT232RL. power the esp8266 from the Uno 3.3 v pin

devyte, 2019-09-11:
The espcomm errors are documented.
Closing.

closed 2019-09-11 by devyte

Slayer4477, 2020-02-18:
change tx pin to gpio ,,, i solve this problem with ota upload

void setup() {
  Serial.begin(9600);
  while (!Serial);
WiFi.hostname("xxxxx");      // DHCP Hostname (useful for finding device for static lease)
  ArduinoOTA.setHostname("xxxx");
  ArduinoOTA.setPort(8266);
  ArduinoOTA.setPassword("");
  ArduinoOTA.onStart([]() {

    /////////////////change pin to gpio////////
    //GPIO 1 (TX) swap the pin to a GPIO.
    pinMode(1, FUNCTION_3);
    //GPIO 3 (RX) swap the pin to a GPIO.
    pinMode(3, FUNCTION_3);
    ///////////////////////
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_SPIFFS
      type = "filesystem";
    }

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    ////////////////change pin to rx tx again//////////
    //GPIO 1 (TX) swap the pin to a TX.
    pinMode(1, FUNCTION_0);
    //GPIO 3 (RX) swap the pin to a RX.
    pinMode(3, FUNCTION_0);
    ///////////////////////
    Serial.println("\nEnd");
  });
  ArduinoOTA.begin();
  delay(1000);
}