esp8266/Arduino #6501: ESP8266HTTPclient http.getstring() causing delays since version 2.5.1

MacSass, 2019-09-10:
### Basic Infos
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.

#### Platform
- Hardware:			[ESP-12]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE
- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- Upload Using:		[SERIAL]

### Problem Description
I have a fairly simple sketch that uses ESP8266HTTPClient to pull a XML from my AVM Fritz Box router. It downloads a challenge file that is needed in ongoing process to calculate a response and login to the router ...

Since core version 2.5.1 this single HTTP request takes 5000ms longer than with previous versions. In fact the http.getString() is what causes the 5000ms delay. Same code compiled with up to version 2.5.0 the same request only takes about 300ms.

This issue does not apply to all HTTP requests, pulling the same XML from my website, the additional 5000ms time delay is not seen. 

Comparing version I found there was a change in honoring the http.timout delay and I suspect that the changes made cause the http.getstring() to wait for the timeout instead of coming back directly after the file has been read.
Honor timeout in HTTPClient #6056
[](https://github.com/esp8266/Arduino/pull/6056)
Unfortunately I´m not savy enough to identify the exact issue. Also, as it does not apply to all http requests, I suspect it might only happen if the webserver does not send a standard C string with correct terminating characters.
As said - that is / was no issue up to version 2.5.0

I also tried to reduce the http.timout() to e.g. 1000ms, which reduced the overall time, but still http.getstring always has to wait for the timeout, which was not the case prior to version 2.5.1.

### Sample Sketch

```cpp
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>

const char* ssid = "mySSID";
const char* password = "myWifiPassword";

HTTPClient http;

void setup() {
  Serial.begin(9600); 
  delay(10);
  if (WiFi.status() == WL_CONNECTED) {
    Serial.print("Already connected... IP address: ");
    Serial.println(WiFi.localIP());    
  } else {
    Serial.print("Connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
  
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    } 
  } 
}

void loop() {
  Serial.println("Challenge: " + getSID());   
  delay(60 * 1000);
}

String getSID() {
  unsigned long startMillis;
  startMillis = millis(); 
  //http.setTimeout(1000);
  http.begin("http://fritz.box/login_sid.lua");
  int retCode = http.GET();
  if (retCode != 200) {
    Serial.println("Get Challengd failed! " + String(retCode));
    return "false";
  }
  String result = http.getString();  // <<-- this is taking 5000ms extra since version 2.5.1
  String challenge = result.substring(result.indexOf("<Challenge>") + 11, result.indexOf("<Challenge>") + 19);
  http.end();
  Serial.print("Challenge retrieval took: ");
  Serial.println(millis() - startMillis);
  return challenge;
}

```
If you require any more information, please let me know.

TD-er, 2019-09-10:
One of the changes in the PR you linked is the change of waiting for the number of bytes available (old) to number of bytes expected.

So I guess the Fritzbox is giving less than the expected amount of data? (or not fast enough?)

MacSass, 2019-09-10:
Thanks for the fast feedback - that could well be the case ...
I took a quick look at the header returned with curl, but I did not see any lenght provided by the Fritz router at all ...
Would that imply a bad behaviour on the Fritz router side? From reading up, it looks like the content lenght header is not mandatory.

So this might be the root cause of the issue - is there any way that this might be fixable in the library code?
As content lenght header ist not mandatory it might be causing the same issue on other sites and I would guess it would be hard to get the router manufacture fix that in router firmware if the header is not mandatory.

TD-er, 2019-09-10:
Chunked transfer will end with a 0-length chunk.
So must be terminated with something like this:
```
0\r\n
\r\n
```

See: [developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Transfer-Encoding](https://developer.mozilla.org/en-US/docs/Web/HTTP/Headers/Transfer-Encoding)

MacSass, 2019-09-11:
Hello, thanks for the info.
I did a network trace of what the Fritz router sends and from what I see it seems to send well formatted chunks:
[image]

Currently my suspicion is that the change made from 2.5.0 to 2.5.1 broke the handling of a chunked response when the timout-handling was introduced.

As said - this is just a suspicion and I don´t have enough understanding if the library code to really be able to prove or event fix that ... 

I would appreciate your help to verify my suspicion - if there is anything more how I could help with that, please let me know.

Thanks - MacSass

TD-er, 2019-09-11:
Well, let's see "you" as the more common interpretation :)
I'm also just a user of this library

d-a-v, 2019-09-11:
What would be helpful is a way to reproduce: an MCVE and an URL

MacSass, 2019-09-11:
Hi,
I have posted what I considered an MCVE above - is this not a good example?

An URL is a bit harder, as this is from calling up the router internal URL. But anybody having a Fritz box router (which is very popular in Europe at least) should be able to reproduce calling up the URL
http://router-ip/login_sid.lua

I´ll dig around to see if I can find another public URL that uses chunked transfer encoding and will try duplicating with that ...

TD-er, 2019-09-11:
ESPeasy does serve its web pages (and JSON) as chunked transfer :)

Jeroen88, 2019-09-12:
Also [jigsaw](https://jigsaw.w3.org/HTTP/) offers several (very useful!) public test servers including [chunked](https://jigsaw.w3.org/HTTP/ChunkedScript)

MacSass, 2019-09-15:
Hello,
I follow your reasoning and agree that from what I have presented so far it looks like the right conclusion. 
I think I need to dig a bit deeper to understand what is going on, as I get well formed chunked response from the router when I call up the URL from Firefox or using curl ...
Also I would really suprised if the router return bad formed response, as usually AVM Fritz is pretty good in what they are doing ... but of course it can still be on their side.

I will do some more investigation and in case I come to the conclusion we still have an issue in the code, I´ll do a re-open. Until then I can always go back to 2.5.0 version, which works perfectly for me.

Thanks for your time - hope to have some more details and a better undestanding of what is going on soon.

Regards - MacSass

MacSass, 2019-12-08:
Hello, 
unfortunately I did not have the time to dig deeper into the issue, but I´m happy to report that this has been fixed as of 2.6.0.
With 2.6.0 and up, I´m getting the nice short response times again like with version prior to 2.5.1, only 2.5.1 and 2.5.2 are causing the issue.
As said - I did not have the time to follow the changes or dig deeper into the root cause - just wanted to state that as 2.6.0 everything is great again ...

Thanks - MacSass

hojnorek, 2021-12-29:
The workaround for this is to set timeout to 1 just before getString.

d-a-v, 2021-12-29:
@hojnorek this is a long-time-closed issue.
If you still get problems, please open a new issue with details and mcve.