esp8266/Arduino #6502: 2.5.3 release

cziter15, 2019-09-11:
Due to important fix #6241 maybe it is worth to release current git version as 2.5.3-beta or something, so it can be reached via Arduino Boards Manager?

Jason2866, 2019-09-11:
A good reason too https://github.com/esp8266/Arduino/issues/6436#issuecomment-528192690

earlephilhower, 2019-09-11:
Pushing betas out on the Arduino channel causes difficulty, because the ide will take it as an "update" and Dolls will install it without looking.

That said, 2.6 official is getting close so it's not going to be a great wait...

Jason2866, 2019-09-11:
Mhh, dont call it beta. 
2.5.3 will be better than 2.5.2. Since you fixed a lot of issues!
So Dolls will get a better core ;-)

devyte, 2019-09-11:
We've discussed this internally, and the decision is to go with 2.6.0 next.
Closing.

closed 2019-09-11 by devyte

d-a-v, 2019-09-29:
Please have a look to experimental snapshot page https://d-a-v.github.io/