esp8266/Arduino #6503: Suggestion: Removing esp8266 when installing from github

dimitre, 2019-09-11:
Just an observation: when I uninstall esp8266 from an Arduino it stops working (quits on loading) until I erase the folder ~/Library/Application Support/Arduino15 in macOS.

Suggestion: instructions could be added on Readme too.
Cheers!

Roman1820, 2019-10-06:
I tried to install arduino-master.zip in IDE 1.8.10 for Windows and got the error that the chosen file has no correct libraries. Regardless whether the folder ~/Arduino15 exists or not.

closed 2019-11-10 by devyte

JazzTp, 2019-12-25:
I have at the moment a superficial knowledge of the Arduino IDE inner "mechanisms", apology if this feedback is useless or if it should be reported elsewhere.

Arduino 1.8.10: I lost a few hours with **errors** (even compiling the bare minimum sketch with empty setup() and loop() functions), **both after following the instructions to install via boards manager _and_ from git**, until I came here and **realized** that **I had some other core supporting esp8266** (which I had installed for other boards with which the IDE is doing fine).

After **moving away the folder hardware/SparkFunBoard_Addons_Jan_16/esp8266/** this core is apparently finally running fine (git install at the moment, just because I was trying out everything including having it under portable/ or not).

**Maybe the Arduino IDE boards manager could check, or check in some other manner, if support for the same hardware is being provided by more than one of the installed cores, before starting to compile anything, and possibly even provide the option to chose among them, or invite to remove all but one.**

earlephilhower, 2019-12-25:
@JazzTp, your best bet is to post your suggestion to the https://github.com/arduino/Arduino repo.  There's nothing we can do here about the IDE itself.

jandrassy, 2019-12-25:
@earlephilhower, esp8266 boards vendors copy the esp8266 Arduino core because they can't refer it as they can refer for AVR or SAMD boards the cores by Arduino
https://github.com/esp8266/Arduino/issues/5769

JazzTp, 2019-12-27:
> @JazzTp, your best bet is to post your suggestion to the https://github.com/arduino/Arduino repo. There's nothing we can do here about the IDE itself.

Yes, thank you, going to give it a try.

PR #6741 (2019-11-10): Mention erasing arduino15 contents before installing git
Fixes #6503