esp8266/Arduino #6506: ArduinoOTA not working when Serial speed is set to 9600

ruggi99, 2019-09-12:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other] (Don't know)
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu] (Don't know)
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

I'm using latest git version and when I upload the example sketch BasicOta.ino from the ArduinoOTA library, the sketch works fine, I can update sketch Over The Air, ecc. The problem comes when I change the serial speed from 115200 (default in the example) to 9600. When I change it, ESP8266 runs always in stack errors. I've not tried with others serial speeds.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch (BasicOTA.ino)

```cpp

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(9600);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
[... 346 lines cut ...]
0x4020c810: esp8266::MDNSImplementation::MDNSResponder::_readRRAnswer(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRAnswer*&) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 435
0x40100fac: free(void*) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1930
0x40215f5d: operator delete(void*) at /workdir/repo/gcc/libstdc++-v3/libsupc++/del_op.cc line 48
0x402089d0: esp8266::MDNSImplementation::MDNSResponder::_parseQuery(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_MsgHeader const&) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 282
0x4021460e: __ssputs_r at /home/earle/src/esp-quick-toolchain/repo/newlib/newlib/libc/stdio/nano-vfprintf.c line 233
0x4020bf31: esp8266::MDNSImplementation::MDNSResponder::_udpReadBuffer(unsigned char*, unsigned int) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/UdpContext.h line 293
0x4020ca14: esp8266::MDNSImplementation::MDNSResponder::_readMDNSMsgHeader(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_MsgHeader&) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1092
0x40209c7b: esp8266::MDNSImplementation::MDNSResponder::_parseMessage() at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 129
0x401007f5: _umm_free(void*) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1380
0x40100fac: free(void*) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\umm_malloc\umm_malloc.cpp line 1930
0x40220340: mem_free at core/mem.c line 237
0x4020a265: esp8266::MDNSImplementation::MDNSResponder::_process(bool) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 76
0x4020a2be: esp8266::MDNSImplementation::MDNSResponder::_callProcess() at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Helpers.cpp line 176
0x40219e78: pbuf_alloc_LWIP2 at core/pbuf.c line 284
0x4020e1f7: std::_Function_handler    (esp8266::MDNSImplementation::MDNSResponder*)> >::_M_invoke(std::_Any_data const&) at c:\users\ruggero\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x401000e1: std::function ::operator()() const at c:\users\ruggero\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2/functional line 2465
0x40203688: UdpContext::_s_recv(void*, udp_pcb*, pbuf*, ip4_addr const*, unsigned short) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src/include/UdpContext.h line 509
0x4021a68c: udp_input at core/udp.c line 404
0x40206ad1: malloc(size_t) at C:\Users\Ruggero\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\heap.cpp line 95
0x4021f5c0: ip4_input at core/ipv4/ip4.c line 1461
0x40220320: mem_malloc at core/mem.c line 210

```

Can someone try?

d-a-v, 2019-09-13:
We can add a mention in serial documentation that printing lots of data (all debug options enabled for example) at low speed like 9600bauds can have bad side effects because serial write is indeed blocking.

https://arduino-esp8266.readthedocs.io/en/latest/reference.html#serial
https://github.com/esp8266/Arduino/blob/master/doc/reference.rst#serial

PRs are welcome

earlephilhower, 2019-09-15:
Actually, @d-a-v, could this be a problem in UART.  He's got a stack dump, indicating a crash, and not a WDT reset in the first entry.  Could UART.c have issues when its send FIFO overflows?

devyte, 2019-09-15:
> Core Version: [latest git]

but log shows 2.5.2. Which is it?

ruggi99, 2019-09-15:
> but log shows 2.5.2. Which is it?

I've taken latest git from GitHub Desktop and overwritten 2.5.2 version. That why you use 2.5.2 version in the log

devyte, 2019-09-15:
That's not a supported installation procedure. I don't know if the end result is correct, latest git pulls things from various places, and some of them have received stability fixes very recently. You could potentially have a Frankenstein build. At rhe very least, the logs are misleading. 
I suggest you do a cleanup of your installation (delete everything), then install a fresh latest git core, then retest.

ruggi99, 2019-09-16:
I've done a cleanup of my installation and installed correctly using this instructions: https://arduino-esp8266.readthedocs.io/en/latest/installing.html#using-git-version.
Then I retried all and the problem is the same. When I select 9600 speed on the sketch, with 2.5.2 it runs without problems and with the very latest git it breaks in stack dumps.

d-a-v, 2019-09-16:
@ruggi99 Does it break always, and not only with debug enabled ?

ruggi99, 2019-09-16:
No, only with all debug enabled and serial speed 9600 and only with latest git. Otherwise it will run without problems. Like @jandrassy said, it could be lots of debug made with a very slow speed.

ruggi99, 2019-09-17:
I've done some tests. If I remove debug only from ESP8266mDNS library, the sketch runs without problems and same if I remove debug only from UdpContext.h. The problem appears only when this two debug options (with the others) are active at the same time. I think it's only related to serial speed

d-a-v, 2019-09-17:
Debug modes have always been modifying nominal behavior, and mDNS probably has now more debug messages since 2.5.2, to help chasing bugs.
You actually found a situation where debug mode and low serial speed break expected behaviour.
Do you think it is fair only mentioning in documentation that low speed serial link in debug mode is not advised ?

ruggi99, 2019-09-18:
Yes, I think it would be enough. Because those who use debug mode and latest gits also know how to disable debug functions in each single library

earlephilhower, 2021-01-05:
#7799 should fix the underlying cause here (too much UART chatter causing WDT).  Nothing to document as it's working now.

closed 2021-01-05 by earlephilhower