esp8266/Arduino #6517: ESP8266AVRISP Don't reset arduino

notsolowki, 2019-09-14:
I'm pretty sure i have it correctly wired, the i have pins D5/D6/D7 wired to the ICSP header of the arduino mega. I have wired pin 5 aka D1 to reset of the arduino. I confirmed i can manually reset arduino but when i start to flash with avrdude i get timeout and the arduino never resets?

jandrassy, 2019-09-14:
do you use level conversion?

notsolowki, 2019-09-14:
yes 3.3v is used. im using the example right from this git. I tried another command this is what i get now

`C:\Users\User\Desktop\new firmwares\AllInOne>avrdude -C avrdude.conf -v -c arduino -p atmega2560  -P net:10.0.0.143:328 -U flash:w:AllInOne.ino.mega.hex:i -F

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "avrdude.conf"

         Using Port                    : net:10.0.0.143:328
         Using Programmer              : arduino
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10     8    0 no       4096    8      0  9000  9000 0x00 0x00
           flash         65    10   256    0 yes    262144  256   1024  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4744608
         Firmware Version: 0.2
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000 (retrying)
avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
avrdude: Expected signature for ATmega2560 is 1E 98 01
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

C:\Users\User\Desktop\new firmwares\AllInOne>
`

notsolowki, 2019-09-14:
You think it only supports stk500v1? there is not much documentation on this project. Or is this something else? i don't think i have anything wired wrong?

jandrassy, 2019-09-14:
protocol 'arduino' should be OK.
ground is connected between boards?

notsolowki, 2019-09-14:
> protocol 'arduino' should be OK.
> ground is wired between boards?
> do you have `const uint8_t reset_pin = D5;?

im not sure how to use avrdudes protocol. I just used what worked for esplink minus the baudrate parameter. i do have ground between boards. i can in the espcode pull reset high to reset. i have tried D1 AND D2 AKA pin 5 and pin 4. I just tried with another type of arduino "uno" and its still failing?

notsolowki, 2019-09-14:
Just to clarify i have wired.

esp  arduino
D5 -- D13
D6 -- D12
D7 -- D11
D1 -- RESET
G-G
EACH POWERED BY USB,

jandrassy, 2019-09-14:
11, 12, 13 is not ISP on Mega. use ICSP header

notsolowki, 2019-09-14:
> 11, 12, 13 is not ISP on Mega. use ICSP header

Sorry i posted so soon. i did use ICSP header on mega. just to make sure i tried to use arduino due/nano now.  i know the spi pins are right they are the same ones i used to wire the nrf cards. I only have esp on spi

jandrassy, 2019-09-14:
on this picture two more connection are wired: reset of esp8266 and one pin to pin wire.

[image]

notsolowki, 2019-09-14:
i just wired it according to this,

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP i am using a nodemcu but that should be okay right? i called a reset in the code and it worked perfect but trying to send a sketch dont do anything except print programming mode from debug console.

notsolowki, 2019-09-14:
if i try it with a different baudrate i get this

`C:\Users\User\Desktop\new firmwares\doseModule1\doseModule1>avrdude -C avrdude.conf -p atmega328p -c arduino -P net:10.0.0.143:328 -b57600 -U flash:w:"doseModule1.ino.standard.hex" -D -F
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
avrdude: AVR device initialized and ready to accept instructions
avrdude: Device signature = 0x666c61
avrdude: Expected signature for ATmega328P is 1E 95 0F
avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.
`

i notice the reset dont happen. and the device id keeps changing

jandrassy, 2019-09-14:
> i just wired it according to this,
> 
> https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266AVRISP i am using a nodemcu but that should be okay right? i called a reset in the code and it worked perfect but trying to send a sketch dont do anything except print programming mode from debug console.
> 
> What if i dont use a level shifter is there a chance it could work and not destroy anything?

D pin to io pin maping is same on Wemos as on NodeMcu

I did it without level conversion. But only for 328p (nano)

notsolowki, 2019-09-14:
> D pin to io pin maping is same on Wemos as on NodeMcu
> 
> I did it without level conversion first. But only for 328p (nano)

AAre you saying you had this working? i have my reset pin connected to d1 of the nodemcu and reset on the arduino. in the nodemcu code i use pin 5 default.

jandrassy, 2019-09-14:
https://raw.githubusercontent.com/wiki/jandrassy/arduino-firmware-wifilink/images/wemos-nano-spi.jpg
https://raw.githubusercontent.com/wiki/jandrassy/arduino-firmware-wifilink/images/wemos-nano-ls-spi.jpg

jandrassy, 2019-09-14:
there are maybe simpler ways to remote upload to Mega. 
https://github.com/jandrassy/ArduinoOTA/blob/master/README.md

notsolowki, 2019-09-14:
my nano is wired like the first picture. Except i am using a nodemcu v0.1 Can you produce the same result? I don't think i'm going to be able to dig into ESP8266AVRISP.h and reverse engineer the whole process

jandrassy, 2019-09-14:
> my nano is wired like the first picture. Except i am using a nodemcu v0.1 Can you produce the same result? I don't think i'm going to be able to dig into ESP8266AVRISP.h and reverse engineer the whole process and im not sure what ESP8266mDNS.h is used for

I don't have time now to test it. It worked in Jan 2018 when I tested it.

the sketch is intended for upload from IDE so it must propagate the MDNS record used by Arduino IDE to show a network port

[image]

notsolowki, 2019-09-14:
Do i have to be running optiboot? I don'nt see a network options under ports on my end. Why is my board not resetting? Do i need to change avrdude.conf?

jandrassy, 2019-09-14:
ignore the left part of the picture. it is not important what board is selected. if IDE finds an Arduino MNDS record it shows the port

I think the board is resetting. Why do you think it is not? You could try to reset it manually in the right moment

notsolowki, 2019-09-14:
> ignore the left part of the picture. it is not important what board is selected. if IDE finds an Arduino MNDS record it shows the port
> 
> I think the board is resetting. Why do you think it is not? You could try to reset it manually in the right moment

Typically when i reset the board even with the hardware pin the bootloader like flashes and then it flashes 3 more times for my serial prints. This don't happen when control by the Nodemcu so i'm pretty sure its not resetting. i also tried a million times to reset it manually as iv'e done this before with other projects.

notsolowki, 2019-09-14:
One strange part is how my device signature changes after every attempt to flash with avrdude. I Am using ESP 2.5.2

notsolowki, 2019-09-14:
Do you know of a program that actually works. and one that i can use the uploading code with my main program. i need to be able to flash and communicate with my arduinos without reflashing the esp. thats what led to me this broken program

jandrassy, 2019-09-14:
> Do you know of a program that actually works. and one that i can use the uploading code with my main program. i need to be able to flash and communicate with my arduinos without reflashing the esp. thats what led to me this broken program

for Mega I again can only recommend my ArduinoOTA library
https://github.com/jandrassy/ArduinoOTA/blob/master/README.md

notsolowki, 2019-09-14:
> for Mega I again can only recommend my ArduinoOTA library
> https://github.com/jandrassy/ArduinoOTA/blob/master/README.md

Should it be easy to run my main program loop inside your code. i want to switch to programmer role and do ota. then back to my main loop. and what about arduino nanos ota should this work with those?

jandrassy, 2019-09-14:
> > > Do you know of a program that actually works. and one that i can use the uploading code with my main program. i need to be able to flash and communicate with my arduinos without reflashing the esp. thats what led to me this broken program
> > 
> > 
> > for Mega I again can only recommend my ArduinoOTA library
> > https://github.com/jandrassy/ArduinoOTA/blob/master/README.md
> 
> Should it be easy to merge my main program with your code? i want to switch to programmer role and do ota. then back to my main loop. and what about arduino nanos ota should this work with those?

Now I remember why I didn't recommend ArduinoOTA when you asked on forum. You have Nanos in your setup. Every Nano has and esp?

If you use WiFiLink firmware, it can OTA flash ATmega328p and the Mega can update itself with WiFiLink and my ArduinoOTA library. And the WiFiLink library has standard Arduino WiFi classes WiFiCliient and WiFiServer. WiFiLink firmware is Arduino esp8266 sketch.

notsolowki, 2019-09-14:
i sent you a pm on arduino

jandrassy, 2019-09-14:
https://github.com/jandrassy/arduino-firmware-wifilink

closed 2019-09-14 by notsolowki