esp8266/Arduino #6519: ESP8266AVRISP Broken for me

notsolowki, 2019-09-14:
I uploaded Arduino_Wifi_AVRISP.ino to a NodeMCU v1 and wired it to a Arduino Nano, according to your instructions. I tried to use avrdude to send a sketch to the arduino and i can see in the debug windows for the esp that it acknowledged my incoming request but nothing happens. The arduino don't restart and it seems that it is not communicating at all on the SPI lines. I would really like to use this program. I tripple checked my wiring except i am trying without a level shifter.

devyte, 2019-09-15:
When you opened this iddue you were presented with sn issue template with specific instructions snd requesting specific information. Please open a new issue, fill out the requested fields, and then copy the above into the section Problem Description.

Closing due to non-compliant issue.

closed 2019-09-15 by devyte

notsolowki, 2019-09-15:
Seriously, What details did i miss. What else could you possibly need to know

jandrassy, 2019-09-15:
I tested it and it works. For Nano and for Mega.
I changed 2 things only to be able to upload from IDE. In the example sketch I replaced the mdns record initialisation, because the original triggers SSH upload in IDE. I used the  `MDNS.enableArduino(port);` used usually for esp8266 ArduinoOTA. 
And in AVR platform.txt I replaced tools.avrdude.upload.network_pattern to 
```
tools.avrdude.upload.network_pattern="{cmd.path}" "-C{config.path}" -p{build.mcu} -c{upload.protocol} "-Pnet:{serial.port}:328" "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i"
```
Additionally for Mega test I changed in AVR boards.txt
 `mega.menu.cpu.atmega2560.upload.protocol` from `wiring` to `arduino`. But that only for upload directly from IDE.

[image]

notsolowki, 2019-09-15:
So the only way to upload the sketch is through IDE? What about avrdude on its own? And i really thank you for your help.  I wouldn't mind removing MDNS from the sketch if i can and still use avrdude without IDE

jandrassy, 2019-09-15:
> So the only way to upload the sketch is through IDE? What about avrdude on its own? And i really thank you for your help. I wouldn't mind removing MDNS from the sketch if i can and still use avrdude without IDE

IDE executes the command I put into platform.txt. Nothing else.

notsolowki, 2019-09-15:
> > So the only way to upload the sketch is through IDE? What about avrdude on its own? And i really thank you for your help. I wouldn't mind removing MDNS from the sketch if i can and still use avrdude without IDE
> 
> IDE executes the command I put into platform.txt. Nothing else. Nothing special.
Please forgive me i have not looked in AVR Platforms.txt. Do i need to change this like you showed? I think i just copied avrdude.conf from the core folders into my sketch location and i call it from there. Please forgive me knowledge

jandrassy, 2019-09-15:
I think it is the same command you used before

`avrdude -C avrdude.conf -v -c arduino -p atmega2560 -P net:10.0.0.143:328 -U flash:w:AllInOne.ino.mega.hex:i -F`

notsolowki, 2019-09-15:
Here is a picture how I have my boards wired. Maybe this wont work with a nodemcu?

[image]

[image]

notsolowki, 2019-09-15:
Do You see anything wrong with the wiring.I tried to use ICSP head on the mega. According to the nano pinouts this should be the correct wiring? my NodeMCU Don't have a 5v output. This is driving me crazy

jandrassy, 2019-09-15:
the command 
`avrdude -Cavrdude.conf -patmega328p -carduino -Pnet:192.168.1.112:328 -Uflash:w:/tmp/arduino_build_251810/sketch_sep15a.ino.with_bootloader.hex:i`

The wiring looks good. Does the esp8266 sketch connect to WiFi? What does it print to Serial Monitor?

notsolowki, 2019-09-15:
> the command
> `avrdude -Cavrdude.conf -patmega328p -carduino -Pnet:192.168.1.112:328 -Uflash:w:/tmp/arduino_build_251810/sketch_sep15a.ino.with_bootloader.hex:i`
> 
> The wiring looks good. Does the esp8266 sketch connect to WiFi? What does it print to Serial Monitor?

Yes the esp connects fine and then i run the command this is what happens,

`C:\Users\User\Desktop\new firmwares\AllInOne>avrdude -C avrdude.conf -patmega328p -carduino -Pnet:10.0.0.110:2323 -Uflash:w:Module1.ino.standard.hex:i
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.`

In the serial output from the esp i get this,

`Arduino AVR-ISP over TCP
IP address: 10.0.0.110
Use your avrdude:
avrdude -c arduino -p <device> -P net:10.0.0.110:2323 -t # or -U ...
[AVRISP] connection pending
[AVRISP] programming mode
[AVRISP] now idle
`

jandrassy, 2019-09-15:
why port 2323? it should be 328

notsolowki, 2019-09-15:
> why port 2323? it should be 328. you changed the example sketch?

yea just because i have a batch program to send the avrdude command it made it easier to change in the sketch. I have the link showing up in ide right now but it failed to upload. i'm trying to locate the AVRplatform.txt file

notsolowki, 2019-09-15:
i tried to find the file you named. i only found platform.txt so i go to the line you show and mine is much longer. so i replace it like this?

tools.avrdude.upload.network_pattern="{cmd.path}" "-C{config.path}" -p{build.mcu} -c{upload.protocol} "-Pnet:{serial.port}:328" "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i" -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return}

jandrassy, 2019-09-15:
> i tried to find the file you named. i only found platform.txt so i go to the line you show and mine is much longer. so i replace it like this?
> 
> tools.avrdude.upload.network_pattern="{cmd.path}" "-C{config.path}" -p{build.mcu} -c{upload.protocol} "-Pnet:{serial.port}:328" "-Uflash:w:{build.path}/{build.project_name}.with_bootloader.hex:i" -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return}
> 
> my error message in IDE says "Failed to reset board" also i have gone back to the default port 328

yes I replaced that line (commented it out with #)

notsolowki, 2019-09-15:
Okay after i do this i get the exact same error message as avrdude gave me last time,

```avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.```

i i got the same debug output from the esp.
Arduino AVR-ISP over TCP
IP address: 10.0.0.110
Use your avrdude:
avrdude -c arduino -p <device> -P net:10.0.0.110:328 -t # or -U ...
[AVRISP] connection pending
[AVRISP] programming mode
[AVRISP] now idle

jandrassy, 2019-09-15:
> Okay after i do this i get the exact same error message as avrdude gave me last time,

because it is not different. still the same avrdude command.

notsolowki, 2019-09-15:
> > Okay after i do this i get the exact same error message as avrdude gave me last time,
> 
> because it is not different. still the same avrdude command.

Is there something wrong with my command?  I don't know where i'm going wrong

jandrassy, 2019-09-15:
why do you select the "Generic ESP8266 Module" and not NodeMcu in Tools menu?

notsolowki, 2019-09-15:
> why do you select the "Generic ESP8266 Module" and not NodeMcu in Tools menu?

I reflashed the esp Using NodeMCU1.0 this isnt the first time i tried that. still no luck. So add the -F command and that dont help either. device name 0x00000 no connection? this is driving me crazy

notsolowki, 2019-09-15:
I know your probably busy but if you have a NodeMcu to try this with that would be great

jandrassy, 2019-09-15:
> I know your probably busy but if you have a NodeMcu to try this with that would be great

I have many different esp8266 boards and modules, but not a NodeMcu. I flashed the Wemos as NodeMcu and the sketch still works.

notsolowki, 2019-09-15:
I uninstalled and reinstalled IDE and all of its boards. Still no luck, Now even after modifying the platforms.txt i don't see network options under ports. I tried multiple different library versions for esp8266 and nothing worked. At what point does the AVR get reset by the esp. what conditions = reset?

notsolowki, 2019-09-28:
I just tried with the wemos mini and i get the same results. what is wrong with this program it does not work

notsolowki, 2019-09-28:
I get no device signature no reset, nothing happens except the print in the serial console from the esp.

beicnet, 2021-07-02:
@notsolowki I have a same issue, could you solve yours?

notsolowki, 2021-07-02:
It could be so many things Firewall/antivirus/router settings/dns settings. 

I just ended up using avrdude directly from command prompt and the ip address of the device

beicnet, 2021-07-02:
@notsolowki I would also use avrdude directly from command prompt, but getting almost the same error as you had.

What command you used?

I'm using this one:
`avrdude -v -C avrdude.conf -c arduino -p atmega328 -P net:192.168.0.230:328 -U flash:w:Blink.hex:i`

notsolowki, 2021-07-03:
Its been a long time but try this, this is what i used it includes a baudrate

avrdude -C avrdude.conf -p atmega2560 -cstk500v2 -P net:10.0.0.110:2323 -b115200 -V -U flash:w:"control_panel.ino.mega.hex" -D

change your port to whatever urs is

beicnet, 2021-07-03:
@notsolowki Thank you for your reply!

Unfortunately I'm still getting the same avrdude error output!

jandrassy, 2021-07-03:
`-P net:` doesn't work on Windows

beicnet, 2021-07-03:
@jandrassy Ohh, didn't know that, but is there any workaround or something?
And in this case ESP8266AVRISP will only work on Linux and Mac platforms?

notsolowki, 2021-07-03:
I dont know about that. I been using -p net on windows in a batch script for some time. Where did you get that idea @jandrassy

beicnet, 2021-07-03:
@notsolowki What version of avrdude are you using?

jandrassy, 2021-07-04:
I have this demo and people say they can\t use it on Windows because avrdude doesn't support -P net
but in this case it is emulation of Serial upload:
https://github.com/jandrassy/lab/blob/master/AvrDudeTelnet/AvrDudeTelnet.ino

beicnet, 2021-07-04:
So, @jandrassy and @notsolowki any working solution to program AVR over ESP8266 for Windows?

notsolowki, 2021-07-04:
It really has been a long time since I've used avrdude but it might have somthing to do with the bootlader your using. But take that wo tf h a grin of salt. Have u tried asking this on the arduino forum?

beicnet, 2021-07-04:
@notsolowki I'm using the default built-in bootloader from Arduino IDE v1.8.2 and ESP8266 Core v2.3.0, no, I'm didn't asked on the Arduino forum yet, because, I felt that on the source page you need to have all information with a few example pictures and/or compatibility issues if exists.

notsolowki, 2021-07-04:
It looks like I was running esplink on the esp8266. That must be why the port is 2323 because I was using the telnet server built into esplink. One u flash esplink make sure avrdude baudrate and esplink baudrate in the web interface match

beicnet, 2021-07-04:
> Is the arduino being reset when you try to flash?

No, it's doing nothing.

I'm getting the same error debug output as you did:

Avrdude output:
```
avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude: ser_drain(): read error: The parameter is incorrect.
avrdude: ser_drain(): read error: The parameter is incorrect.

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14
avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1

         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10
avrdude done.  Thank you.
```

and the output from the ESP8266 (WeMos D1 mini):
```
Arduino AVR-ISP over TCP
IP address: 192.168.0.230
Use your avrdude:
avrdude -c arduino -p <device> -P net:192.168.0.230:328 -t # or -U ...
[AVRISP] connection pending
[AVRISP] programming mode
[AVRISP] now idle
```

Rippatec, 2021-07-06:
Hi,
I'm getting the same problem, and the same messages from AVRdude and the ESP as beicnet (above)
Same AVRdude error and same o/p from my ESP (WeMos mini)
I have tried a node-MCU board and the WeMos mini. Both give the same results.
Also, I looked at the RST and SPI pins from the ESP with a scope and none of them ever change. At the least I would expect the RST line to activate to enable ISP programming but it stays high.
Target has a 10K pullup resistor on RST.
My command line is:
avrdude -c arduino -p t1634 -P net:192.168.3.20:328 -e (for a chip erase).

beicnet, 2021-07-06:
Hi @Rippatec please post your message on ESP forum too...
[https://www.esp8266.com/viewtopic.php?f=29&t=22930](https://www.esp8266.com/viewtopic.php?f=29&t=22930)