esp8266/Arduino #6520: wemos d1 pro mini periodic exceptions (unstable)

afagard, 2019-09-15:
I've got an off-brand [Wemos D1 mini pro off Amazon](https://www.amazon.com/Aideepen-WEMOS-Mini-Pro-Connector/dp/B07333L9VR).

I made a script that works fine on a regular Arduino but results in period exceptions that are always different on the wemos. So I thought, ok, maybe my code is the problem, maybe some library I was using wasn't "compatible". I then tried various example sketches that were installed along with this repo. Notably, the "hello server" one. And am facing the same issues, although the exception is almost always 0.

Raw trace:

```
Exception (0):
epc1=0x4023a054 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffa40 end: 3fffffc0 offset: 01a0
3ffffbe0:  400005e1 000188c3 40102252 00000100  
3ffffbf0:  4023a054 00000033 00000010 00000001  
3ffffc00:  4010465f 04000102 00000000 00000001  
3ffffc10:  fbf8ffff 04000002 3feffe00 00000100  
3ffffc20:  0000001a 00000018 04000102 40104640  
3ffffc30:  3fffc100 3ffee510 00000000 00000000  
3ffffc40:  400043e6 00000030 00000016 ffffffff  
3ffffc50:  400044ab 3fffc718 3ffffd40 08000000  
3ffffc60:  60000200 08000000 00000000 00000000  
3ffffc70:  0000ffff 00042035 00002035 003ff000  
3ffffc80:  4024a5d0 00000020 003ff000 00000030  
3ffffc90:  3ffe8b50 40104304 3ffef19c feefeffe  
3ffffca0:  00000001 40103527 3ffeccd0 feefeffe  
3ffffcb0:  4010398a feefeffe 4023536f 00000001  
3ffffcc0:  ffffffff 00000000 3ffe8ca1 00000008  
3ffffcd0:  402353be 3ffec2c0 3ffeed34 00000001  
3ffffce0:  40104659 40104656 3ffeed34 3ffec2c0  
3ffffcf0:  400005e1 00000000 00000020 40100f26  
3ffffd00:  4023a054 00000033 00000010 3fff01c0  
3ffffd10:  4010465f 04000102 00000000 00000001  
3ffffd20:  fbf8ffff 04000002 3feffe00 00000100  
3ffffd30:  0000001a 00000018 04000102 40104640  
3ffffd40:  3fffc100 3ffee510 00000000 00000000  
3ffffd50:  00000001 40103527 3ffecd18 3ffef0fc  
3ffffd60:  4010398a 0000002e 3ffef12f 4022a0cb  
3ffffd70:  3ffed630 40229c9c 4022c2fc 00000000  
3ffffd80:  00000005 00000000 00000020 40100f26  
3ffffd90:  3ffe8b55 4010431f 3ffec5e0 4021a050  
3ffffda0:  40101c3d 3ffec5e0 00000020 40100f26  
3ffffdb0:  ffffffdc 0044f24f 3ffecfb0 40101e12  
3ffffdc0:  3ffe93f4 00000000 00000000 00000022  
3ffffdd0:  ffffffdc 0044f24f 40102252 00000100  
3ffffde0:  7fffffff 3ffe93f4 3ffe93f4 00000001  
3ffffdf0:  40104659 40104656 0000118d 00000000  
3ffffe00:  400005e1 00040000 7fffffff 00000000  
3ffffe10:  402082cc 00000030 00000010 00000022  
3ffffe20:  40208372 3ffee510 00000000 00000001  
3ffffe30:  00004bc6 00000000 00000000 fffffffe  
3ffffe40:  00000000 3fffc6fc 00000000 3ffee4f8  
3ffffe50:  00000000 3ffee510 3ffee620 00000000  
3ffffe60:  402025eb 00000030 0000001e ffffffff  
3ffffe70:  40204284 3fffff70 4020ef38 3ffee508  
3ffffe80:  00000000 00000000 00000000 fffffffe  
3ffffe90:  ffffffff 3fffc6fc 00000025 3ffee410  
3ffffea0:  00000000 3fffdad0 3ffee620 00000030  
3ffffeb0:  3fffdab0 00000000 3fffd9d0 3ffee620  
3ffffec0:  00000000 00000000 00000001 3ffee5ec  
3ffffed0:  00000000 4bc6a7f0 5645a1ca 00000000  
3ffffee0:  00000000 00000000 4bc6a7f0 00000000  
3ffffef0:  4023cbf6 3fffff58 40100200 000011a7  
3fffff00:  00000000 4c957f2d 3fffdad0 3ffee620  
3fffff10:  0014697b 00000000 4c957f2d 5851f42d  
3fffff20:  3fffdad0 40212d4d 40212d93 3ffee620  
3fffff30:  3fffdad0 3ffee510 3ffee550 40100123  
3fffff40:  4023ddbb 3ffee510 3ffee510 4020759d  
3fffff50:  3ffeed34 40201ea8 3ffee410 4020e9fc  
[... 59 lines cut ...]
0x4020a53d: esp8266::MDNSImplementation::MDNSResponder::_write16(unsigned short, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1184
0x4020aa1d: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSAnswer_PTR_IP4(IPAddress, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1419
0x40100200: millis() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_wiring.cpp line 186
0x40100123: esp8266::polledTimeout::timeoutTemplate   >::expiredOneShot() const at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/PolledTimeout.h line 216
0x40207618: esp8266::MDNSImplementation::MDNSResponder::_updateProbeStatus() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1073
0x40201ea8: ESP8266WiFiSTAClass::status() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 613
0x4020e9fc: WiFiClient::operator bool() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi\src\WiFiClient.cpp line 350
0x40208372: esp8266::MDNSImplementation::MDNSResponder::_process(bool) at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 84
0x4020655a: esp8266::MDNSImplementation::MDNSResponder::update() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 1124
0x4020ca89: esp_schedule() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 95
0x4020cb1e: loop_wrapper() at C:\Users\Alex\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266\core_esp8266_main.cpp line 127
```

Could this be hardware related failure? I am hoping so, otherwise this ESP ecosystem is kind of a pida given the strack traces (at least to me) don't seem that informative. Especially if they have a tenancy to change with the same code making it difficult to debug.

**Things I've tried:**

- Checked to make sure no pins are being pulled low/high that shouldn't be
- Other HTTP-related example sketches provided with this repo.
- Using `esptool.py` I've cleared the flash.
- I'm installed a fresh copy of Arduino IDE as well as this repo via board manager (v2.5.2)

**Power:** My system is supplied with 12 volts from an adapter (also tried computer PSU), which is then stepped down to 5 volts and then fed in to the Wemos D1 5V in.

_I ordered a LOLIN Wemos D1 mini pro off Amazon so I have something to test against but it isn't here yet._

devyte, 2019-09-15:
When you opened this issue you were presented with an issue template with specific instructions and requiring specific info. You completely ignored it.
Closing due to non-compliance. 

BTW, you haven't tried the most important thing of all: latest git.

closed 2019-09-15 by devyte

jandrassy, 2019-09-15:
https://github.com/esp8266/Arduino/issues/6366#issuecomment-525145409

afagard, 2019-09-15:
> [#6366 (comment)](https://github.com/esp8266/Arduino/issues/6366#issuecomment-525145409)

I read that comment, that is why I double checked my power supply. It is 12v to a 5v buck converter, then direct to the wemos board. Surely that couldn't be the issue? It isn't powered off USB or some voltage divider from a TTL adapter.

> When you opened this issue you were presented with an issue template with specific instructions and requiring specific info. You completely ignored it.
> Closing due to non-compliance.
> 
> BTW, you haven't tried the most important thing of all: latest git.

Understandable. I apologize.

ChocolateFrogsNuts, 2019-09-24:
This....
https://github.com/esp8266/Arduino/issues/6366#issuecomment-534337146

Basically the flash chip on the off-brand D1 can't cope with 40mhz and needs to run slower.
Turns out this was the real problem I was having... RFI was just making it worse.

afagard, 2019-09-24:
> This....
> [#6366 (comment)](https://github.com/esp8266/Arduino/issues/6366#issuecomment-534337146)
> 
> Basically the flash chip on the off-brand D1 can't cope with 40mhz and needs to run slower.
> Turns out this was the real problem I was having... RFI was just making it worse.

Yea. I returned the off-brand one and got a genuine lolin d1 mini pro and haven't had a problem since ;)