esp8266/Arduino #6521: ESP8266AVRISP Not working

notsolowki, 2019-09-15:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			        [ESP-12|[NodeMCUv0.1/v1.0]
- Core Version:			[2.5.1]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[DOUT]
- Flash Size: 		[512KB NO SPIFFS]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[USB]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

I uploaded Arduino_Wifi_AVRISP.ino to a NodeMCU v1 and wired it to a Arduino Nano, according to your instructions. I tried to use avrdude to send a sketch to the arduino and i can see in the debug windows for the esp that it acknowledged my incoming request but nothing happens. The arduino don't restart and it seems that it is not communicating at all on the SPI lines. I would really like to use this program. I tripple checked my wiring except i am trying without a level shifter. i really need the lightest way to to uplaod a sketch over Wifi and this seemed like what i needed. please forgive my knowledge

Iv'e tried this with Arduino Mega2560, Arduino Nano and a NodeMCU V0.1/v1.0
I'm using ESP8266 V2.5.1
I'm using Arduino IDE V1.8.9

Sometimes i see that people abuse to use the level shifter when needed. is this just one of those times it is 100% required? I have never dealt with SPI communication. I figured i should still get a reset. i don't register a pulse with my oscilloscope. If i set avrprog.setReset(false); to true then the Arduino reset is activated and stays activated.

### [PROGRAM](https://stackoverflow.com/help/mcve) Sketch

```#include <SPI.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <ESP8266AVRISP.h>

#ifndef STASSID
#define STASSID "wIFINAME"
#define STAPSK  "PASSWORD!!"
#endif

const char* host = "esp8266-avrisp";
const char* ssid = STASSID;
const char* pass = STAPSK;
const uint16_t port = 328;
const uint8_t reset_pin = 5;

ESP8266AVRISP avrprog(port, reset_pin);

void setup() {
  Serial.begin(115200);
  Serial.println("");
  Serial.println("Arduino AVR-ISP over TCP");
  avrprog.setReset(false); // let the AVR run

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    WiFi.begin(ssid, pass);
    Serial.println("WiFi failed, retrying.");
  }

 // MDNS.begin(host);
 // MDNS.addService("avrisp", "tcp", port);

  IPAddress local_ip = WiFi.localIP();
  Serial.print("IP address: ");
  Serial.println(local_ip);
  Serial.println("Use your avrdude:");
  Serial.print("avrdude -c arduino -p <device> -P net:");
  Serial.print(local_ip);
  Serial.print(":");
  Serial.print(port);
  Serial.println(" -t # or -U ...");

  // listen for avrdudes
  avrprog.begin();
}

void loop() {
 
  static AVRISPState_t last_state = AVRISP_STATE_IDLE;
  AVRISPState_t new_state = avrprog.update();
  if (last_state != new_state) {
    switch (new_state) {
      case AVRISP_STATE_IDLE: {
          Serial.printf("[AVRISP] now idle\r\n");
          // Use the SPI bus for other purposes
          break;
        }
      case AVRISP_STATE_PENDING: {
          Serial.printf("[AVRISP] connection pending\r\n");
          // Clean up your other purposes and prepare for programming mode
          break;
        }
      case AVRISP_STATE_ACTIVE: {
          Serial.printf("[AVRISP] programming mode\r\n");
          // Stand by for completion
          break;
        }
    }
    last_state = new_state;
  }
  // Serve the client
  if (last_state != AVRISP_STATE_IDLE) {
    avrprog.serve();
  }

  
}

```

### Debug Messages

```
Debug messages go here
```
[AVRISP] connection pending
[AVRISP] programming mode
[AVRISP] now idle

stfschaefer, 2019-09-16:
What was you command line for avrdude?
Do you invoked avrdude with parameters for using network instead of USB or COM-Port?

jandrassy, 2019-09-16:
@stfschaefer https://github.com/esp8266/Arduino/issues/6519

notsolowki, 2019-09-28:
wow great support from the devs!!. why dont you just take your broken program down

devyte, 2019-09-28:
@notsolowki oh, you seem to be confused. This is not a support forum at all. This is an issue tracker, meant to track issues reported by the community, and fixes implemented also by the community. That means that you, as the primary interested party in this feature, are expected to pursue this, either by investigating and fixing it yourself, or by finding someone who will. Or you can do nothing and wait, and either someone else will pick up your slack, or the issue will timeout and eventually be closed due to lack of interest. 
Our job as maintainers, and be mindful that it's a voluntary one, is to arbitrate and review proposed fixes. 
Please make sure to understand the difference moving forward.

notsolowki, 2019-09-28:
Do you feel better now. Only if you put as much effort into helping the community as you did arguing with people

notsolowki, 2019-09-28:
the only way i got it to work is by setting up a virtual com port on windows and upload using programmer in arduino ide. I found reference to this method here,

https://www.puhy.cz/blog/vzdalene-nahravani-programu-do-arduina-pres-wifi/

So this is avrdudes fault? why can i just use avrdude commandline on windows?

notsolowki, 2019-09-28:
Maybe you should update the instructions and warn windows users of the bug with avrdude and that they need to setup a virtual com port for some unknown reason

liebman, 2019-09-28:
@notsolowki Why don't you submit a pull request with the updates you would like to see in the documentation?

notsolowki, 2019-09-28:
why exactly does avrdude net option not work properly in windows?

devyte, 2019-09-28:
There you go, you found an article with instructions. How about you translate it and propose a guide to be included in our docs? Document PRs are the easiest of all.

notsolowki, 2019-09-28:
The guide is helpful if you using windows but its not a very desirable as you have to install more 3rd party software. It would be better to just type the below command from command prompt.

avrdude -C avrdude.conf -patmega328p -cstk500v1 -b19200 -Pnet:10.0.0.110:328 -Uflash:w:Module1.ino.standard.hex:i

Why exactly do i need to use a virtual com driver. Why didn't i need to use it with esplink. It seems there should be a better solution

devyte, 2019-09-28:
Firewall comes to mind

notsolowki, 2019-09-28:
i just tried that, I manually added it to incoming and outgoing rules to allow the connection, same results.

jandrassy, 2019-09-28:
supposedly avrdude's -Pnet: doesn't work on Windows

notsolowki, 2019-09-28:
I screwed something up. i pulled the wires during a flash and now my device shows up as 0x0000 and dont seem like the bootloader is working? can anything fuse wise on the atmega328p cause the device signature not to show up?

jandrassy, 2019-09-28:
with ISP flashing the bootloader is deleted

Pablo2048, 2019-09-28:
Avrdude network transfer, according to the documentation, is not implemented on Win32 platform - see https://www.nongnu.org/avrdude/user-manual/avrdude_4.html section ```-P port``` . I don't follow why are you asking questions about avrdude functionality in Arduino ESP8266 core instead of using "How to get help or report bugs" from https://www.nongnu.org/avrdude/ .

devyte, 2019-09-28:
I don't see a core issue here. Closing.
@notsolowki if you do make a PR with doc updates, please reference this issue in it.

closed 2019-09-28 by devyte