esp8266/Arduino #6523: Example Sketch httpUpdateSecure.ino Errors on Compile

lbussy, 2019-09-15:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [X] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:  ESP8266
- Core Version:  v2.2.3 [cd6f5cc] 9/2/2019
- Development Env:  PlatformIO
- Operating System:  Windows

### Settings in IDE

- Module:  Wemos D1 mini r2
- Flash Mode:  qui
- Flash Size:  4MB
- lwip Variant:  v1.4
- Reset Method:  ck
- Flash Frequency: 40Mhz
- CPU Frequency:  80Mhz|160MHz]
- Upload Using:  SERIAL
- Upload Speed:  460800

### Problem Description

Compiling provided httpUpdateSecure.ino sketch from the repo examples results in an error.

### Sketch

[Example sketch from repo](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266httpUpdate/examples/httpUpdateSecure/httpUpdateSecure.ino).

### Debug Messages

```
src\main.cpp: In function 'void setup()':
src\main.cpp:56:89: error: no matching function for call to 'BearSSL::CertStore::initCertStore(fs::FS&, const char*, const char*)'
     int numCerts = certStore.initCertStore(SPIFFS, PSTR("/certs.idx"), PSTR("/certs.ar"));
                                                                                         ^
src\main.cpp:56:89: note: candidate is:
In file included from C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClientSecureBearSSL.h:30:0,
                 from C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClientSecure.h:41,
                 from C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiServerSecure.h:20,
                 from C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:41,
                 from src\main.cpp:1:
C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/CertStoreBearSSL.h:62:9: note: int BearSSL::CertStore::initCertStore(BearSSL::CertStoreFile*, BearSSL::CertStoreFile*)
     int initCertStore(CertStoreFile *index, CertStoreFile *data);
         ^
C:\Users\Xxxxx\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/CertStoreBearSSL.h:62:9: note:   candidate expects 2 arguments, 3 provided
*** [.pio\build\d1_mini\src\main.cpp.o] Error 1
```

earlephilhower, 2019-09-15:
You have an old or inconsistent version of the core.  GIT head is 2.6.0-dev, the latest release installable from the IDE is 2.5.2.  2.2.3 would be ancient, and wouldn't have BearSSL included at all.

Our CI system builds every example using both PIO and the standard Arduino IDE, so please clear out your version and do a git pull on head or reinstall 2.5.2 from whatever method PIO uses.

closed 2019-09-15 by earlephilhower

lbussy, 2019-09-15:
Thanks, Earl, I'll dig around.  Not sure what's up as this was a clean environment but I'll do the needful.