esp8266/Arduino #6524: Print::print(const __FlashStringHelper *) is very inefficient

nomis, 2019-09-15:
The implementation calls `pgm_read_byte()` to read every character. This is inefficient because it will read every block of 4 bytes multiple times.

It calls calls `write(c)` for every character. This is very inefficient at the network level if `TCP_NODELAY` is turned on because suddenly all flash strings are sent out as one byte per packet.

It should use `strncpy_P()` with a small buffer instead.

earlephilhower, 2019-09-15:
A PR for this would be quite welcome.  Just make sure the buffer isn't too large, as the stack space is pretty limited on this device.  I think 64 bytes or so is what is used elsewhere.

TD-er, 2019-09-29:
I was looking into the code to see if this suggested change could also benefit my own code.

But what I don't understand is how it could be faster to use `strncpy_P()`, since that code is also using `pgm_read_byte()`
https://github.com/esp8266/Arduino/blob/74819a763bfb6e9890a57411dcea4aba221a778d/cores/esp8266/pgmspace.cpp#L151-L172

Not sure if the code has changed, as it appears to be some external code now?

nomis, 2019-09-29:
I didn't realise none of the implementations were efficient :(

It would still be faster to use `strncpy_P()` if the output is a socket with `TCP_NODELAY`, otherwise it creates a lot of 1 byte packets.

d-a-v, 2019-09-29:
Try with nagle / without `NODELAY`, and \with `client.flush()` after your prints.

earlephilhower, 2019-09-29:
@TD-er , that's a very old version.  The newlib-xtensa ones are significantly optimized and pretty close to RAM ones when possible:  https://github.com/earlephilhower/newlib-xtensa/blob/xtensa-2_2_0-lock-arduino/newlib/libc/sys/xtensa/string_pgmspace.c.

In any case, larger chunks would be faster due to lower overhead.  PRs always appreciated. :)

nomis, 2019-09-29:
> Try with nagle / without `NODELAY`, and with `client.flush()` after your prints.

`flush()` is part of `Stream` (input) not `Print` (output). You do not want to call it on a `Client` because it will discard all currently available input.

d-a-v, 2019-09-29:
You do want do call it because it forces data out, per arduino API.
https://www.arduino.cc/en/Reference.StreamFlush

And this is the way `::flush()` is implemented in this repo's TCP (check `WiFiClient`, `ClientContext`).

And `Stream::` inherits from `Print::`. So while `Print::` is output only, `Stream::` is I/O.

nomis, 2019-09-30:
`flush()` is implemented here as "wait for the send buffer to empty" not "flush the send buffer into packets".

Look at the ESP32 implementation if you don't believe that it gets implemented as flushing input.

`Print` is output only, so it should have the `flush()` function. It's not present so it's effectively an input function.

d-a-v, 2019-09-30:
Quoting `Serial.flush()` ([ref](https://www.arduino.cc/reference/en/language/functions/communication/serial/flush/))
> ***Waits for the transmission of outgoing serial data to complete***. (Prior to Arduino 1.0, this instead removed any buffered incoming serial data.)

Quoting `Stream::flush()` ([ref](https://www.arduino.cc/en/Reference.StreamFlush))
> flush() clears the buffer ***once all outgoing characters have been sent***. 

While it is unclear what buffer it is about, `Serial.flush()` documentation explains it well.

In official arduino avr implementation, `::flush()` is declared in `Print::` like in this core.
Now concerning implementation in this core, `WiFiClient::flush()` follows Serial's one.

About esp32 implementation, I don't agree [with it](https://github.com/espressif/arduino-esp32/blob/caa391ab3420db0316b8ad74ba13d8fc3ac66a9e/libraries/WiFi/src/WiFiClient.cpp#L467), but comments state that we have the good implementation. @me-no-dev might have more insights about that. I think this is an inheritance from Arduino 1.0.

> Print is output only, so it should have the flush() function. It's not present so it's effectively an input function.

Well then. We can speak for hours, or you can just try whether `WiFiClient.flush()` fits your needs, or PR/fix like you proposed in OP.

devyte, 2019-11-09:
@earlephilhower @d-a-v what is the status here? is there anything to improve, or is the current code already reasonable?

nomis, 2019-11-10:
@devyte The function still needs to be modified to write in blocks of 64 bytes.

It's not something I'm likely to get around to doing because I've implemented a general outgoing buffer in my application so that separate print calls can be merged too.

earlephilhower, 2019-11-11:
@devyte current code is functionally correct, but does only 1 byte-sized writes when it could potentially be doing larger chunks.  That's an optimization, yes, but not critical.  If there's downtime it's worth doing, but it's not a high priority IMO.

PR #6893 (2019-12-10): Use 128B chunks instead of 1B writes in Print::print(FlashStringHelper)
Fixes #6524

Should help with speed of output when printing large flash strings to
things like a file or a TCP connection.

Use a 128 byte chunk in a temp buffer to send data using write(),
reducing the # of write calls by ~127x.
commit 759ba27b (2019-12-10): Use 128B chunks instead of 1B writes in Print::print(FlashStringHelper) (#6893)

Fixes #6524

Should help with speed of output when printing large flash strings to
things like a file or a TCP connection.

Use a 128 byte chunk in a temp buffer to send data using write(),
reducing the # of write calls by ~128x.
closed 2019-12-10 by devyte