esp8266/Arduino #6528: [can't compile] > tools/python3 no such file ../ What happens?

derrgis, 2019-09-16:
### Basic Infos

- [x] I have tested that the issue is present in current master branch (aka latest git).

#### Platform

- Hardware:			       [ESP01]
- Core Version:			[latest git]
- Development Env:		[Arduino IDE]
- Operating System:		[MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[qio]
- Flash Size: 		[1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

Not able to compile with persistant console error after from scratch installation.

### sketch from examples > Blink.ino
```

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

void loop() {
  digitalWrite(LED_BUILTIN, LOW); 
  delay(1000);        
  digitalWrite(LED_BUILTIN, HIGH); 
  delay(2000);    
}

```

### Debug Messages

```
Arduino : 1.8.9 (Mac OS X), Carte : "Generic ESP8266 Module, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), ck, 26 MHz, 40MHz, QIO (fast), 1MB (FS:64KB OTA:~470KB), 2, nonos-sdk 2.2.1+100 (testing), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

fork/exec /Users/.../Documents/Arduino/hardware/esp8266com/esp8266/tools/python3/python3: no such file or directory
Erreur de compilation pour la carte Generic ESP8266 Module

```

earlephilhower, 2019-09-16:
After updating git, you need to run tools/get.py to pull the latest tools.  GIT head now required python3 to be installed (and it's part of the tools JSON) because Python2 is EOL on Jan 1 2020.

derrgis, 2019-09-16:
Great, am going to have a look at that, Many thanks 😃

earlephilhower, 2019-09-16:
Closing.  We also have Mac builds now as a part of CI, so the tools update should get you going.  See https://travis-ci.org/esp8266/Arduino/jobs/584763541

closed 2019-09-16 by earlephilhower

derrgis, 2019-09-17:
Sorry to come again. Before to go further here is what happens.
Today I've upgrade my Python3 version on my Mac OS.10.11.6 via homebrew wihtout problem.

After the command `cd esp8266/tools` and `python3 get.py` as described for the download binary tools [https://arduino-esp8266.readthedocs.io/en/latest/installing.html#boards-manager](url)

I get this and I can't understand if that comes from my particular computer setup or from the esp distribution.

it seems not easy, where the best to go now ?

```
python3 get.py
Platform: x86_64-apple-darwin
Downloading python3-macosx-placeholder.tar.gz
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1317, in do_open
    encode_chunked=req.has_header('Transfer-encoding'))
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1229, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1275, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1224, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1016, in _send_output
    self.send(msg)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 956, in send
    self.connect()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/http/client.py", line 1392, in connect
    server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 412, in wrap_socket
    session=session
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 853, in _create
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/ssl.py", line 1117, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "get.py", line 131, in <module>
    main()
  File "get.py", line 128, in main
    get_tool(tool)
  File "get.py", line 80, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 247, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 543, in _open
    '_open', req)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1360, in https_open
    context=self._context, check_hostname=self._check_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 1319, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)>
```

earlephilhower, 2019-09-17:
Ah, I ran into that, too, on the CI integration.  Homebrew installs python, but doesn't actually install the needed certificates into the Mac keyring or whatever.  I don't understand their reasoning, but w/o it the HTTPS functionality of Python3 is broken.

I followed the directions and added below, which actually installs the certs and makes things work:
`cd "/Applications/Python 3.7/" && sudo "./Install Certificates.command"`

You'll need to do the same, possibly changing the path to the one you installed Python3 into.  I don't remember if I found that out from the homebrew install logs or StackOverflow, but it was quite a puzzler for me (last Mac I used had a PowerPC chip in it!).

derrgis, 2019-09-17:
Wow great ! it helps go further. Indeed there was a certificate issue.
The command works as expected.

Unfortunately I get a permission denied on dist/python3-macosx-placeholder.tar.gz
and I unable to find where is that file. Do you have an idea?
Here's what I get now 

`python3 get.py
Platform: x86_64-apple-darwin
Downloading python3-macosx-placeholder.tar.gz
Traceback (most recent call last):
  File "get.py", line 131, in <module>
    main()
  File "get.py", line 128, in main
    get_tool(tool)
  File "get.py", line 80, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/urllib/request.py", line 257, in urlretrieve
    tfp = open(filename, 'wb')
PermissionError: [Errno 13] Permission denied: 'dist/python3-macosx-placeholder.tar.gz'`

earlephilhower, 2019-09-17:
That's a FS permission issue on your computer.  Not sure what Mac permissions look like, or where you've installed the git repo (could it be in a protected spot, or installed using `sudo git clone...`?), but assuming it's in the right spot a judicious use of `sudo python3 get.py` should allow it to write to that directory no matter what.

derrgis, 2019-09-17:
Pfiouuu you saved my day.
Please come see me if you come in Staint-Etienne, france, I owe you a beer :)))

devyte, 2019-09-17:
if @earlephilhower were to call in all the beers he's owed, he'd drown :joy:

jasonw-nc, 2020-03-23:
Leaving this here for others to find.
This was a perplexing issue, I've used Arduino 1.8.9 successfully for many months and just decided to upgrade, which broke uploading to my nodemcu with the above python error.  I looked here for an answer, found this supremely complex and confusing response and decided to look further.  
I just simply installed the latest stable release of python 3 from python.org (only 30MB) and it's working just fine again.  The installer even notified the user at the end of the install regarding the cert stuff, which is specific to Mac users.  Python 3 installer came with a packaged autoscript to clean that up too.  Much, much easier route to take...just my .02.

GregJPowers, 2020-04-22:
> Leaving this here for others to find.
> This was a perplexing issue, I've used Arduino 1.8.9 successfully for many months and just decided to upgrade, which broke uploading to my nodemcu with the [...]

Thanks so much for putting this SOLUTION HERE! The other way was just too complicated. Your solution was something I could. Just do not miss the addition step that is called out on the last screen. You MUST got the the Finder folder and run the certificate script to make everything work. Thanks again it was MAGIC !!!!!!!!

pingiboi, 2020-04-28:
Hello, I have a similar problem derrgis but I managed to succesfully install the certifications in the directory where python 3.8 is (I have python 3.8). I have since written in the terminal `cd esp8266/tools` and `python3 get.py` and gotten this to work succesfully as well. However I still get an error message trying to compile in the Arduino IDE. I now get the error message :

`env: python3: No such file or directory
exit status 127
Error compiling for board Generic ESP8266 Module.`

aleonnet, 2020-05-02:
> Hello, I have a similar problem derrgis but I managed to succesfully install the certifications in the directory where python 3.8 is (I have python 3.8). I have since written in the terminal `cd [...]

The problem is that the symlink for python3 is incorrect. To correct it:

$ cd /Users/[your-username]/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/
$ rm python3
$ which python3
$ ln -s [path-returned-by-the-above-command] python3

https://forum.arduino.cc/index.php?topic=649739.msg4401959#msg4401959

TASSDevon, 2020-05-03:
> The problem is that the symlink for python3 is incorrect. To correct it:
>
> $ cd /Users/[your-username]/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/
> $ rm python3
> $ which python3
> $ ln [...]

This is the only solution that worked for me, thanks!

luxxtek, 2020-05-05:
Indeed the only solution that works.
Great !

kchrs, 2020-05-11:
> The problem is that the symlink for python3 is incorrect. To correct it:
>
> $ cd /Users/[your-username]/Library/Arduino15/packages/esp8266/tools/python3/3.7.2-post1/
> $ rm python3
> $ which python3
> $ ln [...]

I do not have Python 3 installed but 2. So what? Should I install Python 3 and do the steps? Thanks

aleonnet, 2020-05-11:
Python 2 is gone... I suggest you to install Python 3.

kchrs, 2020-05-11:
> Python 2 is gone... I suggest you to install Python 3.
> […](#)

OK! Thanks!

NieVystrel, 2020-06-11:
In a desperate way to avoid the problem mentioned in the post, i remove Python (previously installed in default route), RE installed in C:\Users\{USERNAME}\Documents\Arduino\hardware\esp8266com\esp8266\tools (where repo resides) and run a GIT BASH from here, wrote "python3 get.py" and no results...deleting the number 3 DID THE JOB! Try this line "python get.py" and maybe youll get the binary tools in a lucky strike like me hehehe. Never give up guys! Regards.