esp8266/Arduino #6531: EEPROM location was changed

byq13, 2019-09-18:
In the thirst place I have update to v2.5.2 and it was working ok, but when I have downloaded the newest branch from GIT, then I have discovered problem with EEPROM. 
In the newest branch EEPROM location was changed.
My problem is that I have put authentication key there. Now I have to have access to it.

EEPROM function is working fine. In v2.52 and with the newest. It is just not the same part of the flash memory.

How can I have access to old EEPROM location with the newest Arduino framework from GIT?

ESP-12S with 4MB, - 4M2M=4MB (FS:2MB OTA:~1019KB)

d-a-v, 2019-09-18:
Current eeprom location:

`(((uint32_t)&_FS_end - 0x40200000) / SPI_FLASH_SEC_SIZE)`
https://github.com/esp8266/Arduino/blob/5d609fd29475c44c7b752b6cbf4128527f4977d3/libraries/EEPROM/EEPROM.cpp#L44

`_FS_end` in current master:
https://github.com/esp8266/Arduino/blob/5d609fd29475c44c7b752b6cbf4128527f4977d3/tools/sdk/ld/eagle.flash.4m2m.ld#L18

In 2.5.2:
https://github.com/esp8266/Arduino/blob/8b899c12fb5044a6a51b92501bfcdf86a3d98dd8/tools/sdk/ld/eagle.flash.4m2m.ld#L18

By calliing `EEPROM.begin((((uint32_t)0x405FB000 - 0x40200000) / SPI_FLASH_SEC_SIZE)`, you might recover what you had in 2.5.2.
You would then need to close and `EEPROM.begin()` again to save it to the new location.

Sorry for this change, it should be documented for next release.

d-a-v, 2019-09-18:
You can also get try to add 4096 to the offset of your data, without calling `.begin(newaddress)` like above.

byq13, 2019-09-18:
Working. Thank you.
I would like to stay with old data in its place.

For what old address will be reserved  in next framework version? 
OLD  eeprom @0x405FB000 (4KB) 
NOW: _FS_end = 0x405FA000  -> EEPROM = 8KB.

d-a-v, 2019-09-18:
It should not change anymore.
It was changed because of an oddness in flash mapping (#5989).

byq13, 2019-09-18:
May I assume, that now we have 8 KB of EEPROM?
Or EEPROM will be on the end of FS and on old EEPROM space will be something else?
At the moment is noones land.

Many thanks for help. It is working perfectly.

devyte, 2019-09-18:
@byq13 #5989 addressed a bug in the calculation of the SPIFFS end address. In a cormer case, it could theoretically corrupt the filesystem. I now realize the fix shifts the location of tbe EEPROM, so after an update the EEPROM data wouldn't be seen by the class.
What you may want to do is copy the EEPROM data from the old address to the new correct address. This would need to be done only once, and only when updating from before that fix to after the fix.

Notes:
-retro - mark #5989 as breaking and include the above description on how to address it.

earlephilhower, 2019-09-18:
For 2.6, I'll fix the EEPROM sector to point back to the same address as it was, by making boards.txt.py calculate it using the old (busted) formula and removing the implicit calculation in eeprom.cpp.  This will make EEPROM backwards compat w/o any changes for users.  For 3.0 we may move it to the last sector after the FS, but IMO that's probably not worth the trouble and breakage.

liebman, 2019-09-18:
@earlephilhower If this was in 2.5.2 maybe a config option?  Someone who has built with 2.5.2 may have the same issue if you change the calculation back......

earlephilhower, 2019-09-18:
Thanks for checking that.  This bug report was very timely and caught a subtle problem!

earlephilhower, 2019-09-19:
Please try PR #6537 .  It should restore the location so you can read old data correctly.

PR #6537 (2019-09-20): Restore EEPROM address to prior released location
When the FS_END was adjusted to end on a full block (i.e. rounded down)
to avoid filesystem issues, but _FS_end was changed.  The EEPROM library
used _FS_end to implicitly calculate the start of the EEPROM data, so
this means after the _FS_end fix, EEPROM data written with prior
releases would "disappear."

Avoid the issue by explicitly calculating the EEPROM start location in
the linker, using the same formula as prior release.

Fixes #6531
commit 69f3e81f (2019-09-20): Restore EEPROM address to prior released location (#6537)

When the FS_END was adjusted to end on a full block (i.e. rounded down)
to avoid filesystem issues, but _FS_end was changed.  The EEPROM library
used _FS_end to implicitly calculate the start of the EEPROM data, so
this means after the _FS_end fix, EEPROM data written with prior
releases would "disappear."

Avoid the issue by explicitly calculating the EEPROM start location in
the linker, using the same formula as prior release.

Fixes #6531
closed 2019-09-20 by earlephilhower