esp8266/Arduino #6532: Connection drops for batch of clients at the exact same time

kinitzki, 2019-09-18:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[2.5.0]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows & MacOS]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[OTA|SERIAL (both)]
- Upload Speed:   [115200]

### Problem Description

I am struggling with a project that has ESP8266 (NodeMCUs) as clients communicating to VerneMQ running on a DigitalOcean droplet. We have about 200 things at different locations sending about 20msg per minute each. Yesterday 18 clients at one location suddenly dropped connection. A month before 40 clients showed the same behaviour at another location.

All clients were able to connect to MQTT again by simply restarting them.

Can you help us troubleshooting the issue? Receiving debugs messages / dumps is quite difficult due to the ESP8266 being remote. We could have them hooked up to a computer running Arduino IDE but to be honest we do not know what else to check other than the serial print - especially as when restarting the ESP8266, the connection is established again.

We did not manage to reproduce the behaviour locally. :/

Thank you very much in advance. Please let me know if you need some more specs.

#### Expected behavior
At several remote locations, clients connect to the (MQTT) server and publish data to diverse topics. Each client has a unique ID (MAC address). At each location, the clients connect via (dedicated) WIFI.

#### Actual behaviour
Clients connect to the MQTT server and publish data to diverse topics. Occasionally a batch of clients drops connection at the same time at a location and is not able to reconnect unless the devices are rebooted. This is not happening isolated at one single location but is random.

### Sketch [Cannot reproduce - therefore no MVCE, sorry]

```

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>
#include <Bounce2.h>
#include <WiFiUdp.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ESP8266Ping.h>

// --------------------- Sensor Settings

const char* MQTT_LOCATION = "REDACTED";
const char* PLACE = "REDACTED";
const char* SKETCH_VERSION = "14";

// --------------------- Network Settings

#define WIFI_SSID "REDACTED"
#define WIFI_PASSWORD "REDACTED"
[... 472 lines cut ...]
    strcat(url,"&PLACE=");
    strcat(url, PLACE);
    strcat(url,"&version=");
    strcat(url, SKETCH_VERSION);
    HTTPClient http;  //Declare an object of class HTTPClient
    Serial.println(url);
    http.begin(url);  //Specify request destination
    int httpCode = http.GET();   
    if (httpCode > 0) { //Check the returning code
      String payload = http.getString();   //Get the request response payload
      Serial.println(payload);
      if(payload == "restart"){
        Serial.print("Reboot now");
        ESP.restart();
      } else if(payload == "update"){
        updateOTA = true;
      } else if(payload == "update_PLACE"){
        updateOTAPLACE = true;
        updateOTA = true;
      }
    }
    http.end();   //Close connection
  }
}
```

d-a-v, 2019-09-18:
Are there some kind of logs from the AP ?

> Yesterday 18 clients at one location suddenly dropped connection. A month before 40 clients showed the same behaviour at another location.

Can you try to reconnect to MQTT when connection is lost ?
```
if(WiFiMulti.run() != WL_CONNECTED){
    Serial.println("WiFi not connected!");
    // TODO: reconnect to wifi, then if necessary, to MQTT
   ...
}
```

d-a-v, 2019-09-18:
Also you should try again with latest git master, #2330 has been closed by #6484.

kinitzki, 2019-09-18:
@d-a-v thank you very much. I really appreciate your reply!

I already asked for the AP logs and will post them here asap. 

The big issue for us is that we have no idea on how to reproduce it. We tried the typical test cases with Wifi / Internet / MQTT and switching between states...

What's kind of odd. We had two sensors at that location that remained online **this time** that have a way older sketch flashed (Version 5 instead of 14). Nevertheless, they also had similar issues at other occasions.

**Libraries used in current sketch (see above):**

```
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>
#include <Bounce2.h>
#include <WiFiUdp.h>
#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>
#include <ESP8266Ping.h>
```

**Libraries used in older sketch (the clients that stayed online):**

```
#include <ESP8266WiFi.h>
#include <Ticker.h>
#include <AsyncMqttClient.h>
#include <Bounce2.h>
#include <WiFiUdp.h>
#include <NTPClient.h>
```

Not sure though what versions the libraries had back then. :/

devyte, 2019-09-18:
#2330 is an issue in the ESP's ROM, and is all-invasive. It was addressed only very recently via a workaround with #6484. Its prime symptom was the ESP dropping out randomly. 
Your core 2.5.0 is rather old. The number of stability fixes merged since then that could be relevant is greater than I care to remember.
We don't really have the manpower to help you dig through your project code, let alone the mqtt lib, and in any case it doesn't make sense to do so for your core version. I suggest the following:

1. find a way to reproduce the issue with an ESP locally, i. e. one that isn't far away. If your issue is #2330, you'll have better luck with many ESPs plus some strain plus the same AP as where the issue happened originally. 
2. try with latest git, or at least some commit after #6443 
3. work with the asyncmqtt lib authors to reduce the issue, i. e. reproduce it without their lib (just our core code). Once you have that, we can pick it up.

Closing in the meantime.

closed 2019-09-18 by devyte