esp8266/Arduino #6535: Issue downloading the link copied to preferences as shown in boards manager

Sabakalsoom, 2019-09-19:
When i paste the link into the other boards URL preferences, board manager doesn't show ESP8266 and the following error shows up.

[image]

Please give a feedback on this issue.

d-a-v, 2019-09-19:
https://github.com/esp8266/Arduino#quick-links > Installing > Boards Manager
gives:
https://arduino.esp8266.com/stable/package_esp8266com_index.json

closed 2019-09-19 by d-a-v

Sabakalsoom, 2019-09-20:
> https://arduino.esp8266.com/stable/package_esp8266com_index.json

I did use the same link in the other boards manager URL, still the same error shows up.

d-a-v, 2019-09-20:
I assume clicking on the above link with the browser works and shows the json file.
Try reinstalling arduino from scrach, after having [properly removed all previous installations](https://forum.arduino.cc/index.php?topic=362507.0) then following [this process](https://arduino-esp8266.readthedocs.io/en/2.5.2/installing.html#boards-manager).

Sabakalsoom, 2019-09-20:
> I assume clicking on the above link with the browser works and shows the json file.
> Try reinstalling arduino from scrach, after having [properly removed all previous [...]

Got it right. Thank you!