esp8266/Arduino #6539: warning: "UDP_TX_PACKET_MAX_SIZE" redefined

kunosee, 2019-09-20:
With the current github version of the esp8266/Arduino Framework, i receive compile time warnings, when using mDNS and dns.

> warning: "UDP_TX_PACKET_MAX_SIZE"

I  can reduce it to the following minimal sketch:

```
#include <ESP8266mDNS.h> 
#include <Dns.h>

void setup() {
}

void loop() {
}

```

Before updating my Arduino IDE from 1.8.7 to 1.8.10 and pulling the most current esp8266-Framework from github, I can't remember this warning.

earlephilhower, 2019-09-30:
Dns.h comes from the Ethernet library which is incompatible with MDNS and most other things due to severe restrictions (24-byte max UDP packet sizes, for a start).

Are you sure that's what you're looking for?

I think you want:
````
#include <ESP8266mDNS.h>
#include <lwip/dns.h>
````

kunosee, 2019-10-08:
Thanks, that was it!