esp8266/Arduino #6546: Loosing last byte with readBytesUntil

danielkucera, 2019-09-22:
I have following code:
```
            client_len = serverClients[i].readBytesUntil(_EOF, client_buf, MAX_FRAME_LENGTH);
            client_buf[client_len++] = serverClients[i].read();
            if (client_buf[client_len-1] != _EOF){
              //something strange happened
              serverClients[i].write("uhm");
              continue;
            }
```
How should I distinguish if _EOF character or MAX_FRAME_LENGTH or timeout has been reached?
The implementation reads the terminating character but then it is lost (not written to output buffer):
https://github.com/esp8266/Arduino/blob/ad2b51e36fdd305e4e9a9556e54891a3a4be1848/cores/esp8266/Stream.cpp#L228

jandrassy, 2019-09-22:
this is Arduino API copied into every Arduino boards package. it is how it should work.
https://www.arduino.cc/reference/en/language/functions/communication/stream/streamreadbytesuntil/

danielkucera, 2019-09-22:
I'm not saying that it's implemented in a wrong way - it corresponds to the docs. I'm saying that maybe the design is wrong.

> How should I distinguish if _EOF character or MAX_FRAME_LENGTH or timeout has been reached?

devyte, 2019-09-22:
That's a question to be raised with the Arduino reference developers, not here. If you do get them to change the return value in the Arduino reference to allow distinguishing between the cases, please let us know and we'll update our implementation. 
Closing.

closed 2019-09-22 by devyte

jandrassy, 2019-09-23:
the complete set of the blocking functions of Stream is strange. but they can't be changes after so many years copied in so many cores and used in so many sketches