esp8266/Arduino #6551: eeprom no longuer working since 2.5.1

viknet365, 2019-09-23:
----------------------------- Delete above -----------------------------

### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[wemos d1 r1]
- Core Version:			2.5.1
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			[Wemos D1 R1|
- Flash Mode:			unknown
- Flash Size: 		[4MB]
- lwip Variant:   v2 Lower Memory
- Reset Method:		board button
- Flash Frequency:	[unknown]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600] (serial upload only)

### Problem Description

The eeprom.read functions no longer work on my board with my IDE since 2.5.1
it works if using core 2.5.0.
The following sketch read eeprom (emulated eeprom), then write random data to it. and commit at the end.
The sketch read 0 if compiled with core 2.5.1 and later
the sketch dump correct eeprom value if run with core 2.5.0

Other project using eeprom no longuer work either (in my environment).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <EEPROM.h>

int addr = 0;
int readMode = 1; 
int writeMode = !readMode;
int eepromLength = 100;

void setup()
{
  EEPROM.begin(eepromLength); 
  Serial.begin(9600);
  delay(2000);
  Serial.println();
  Serial.println();
  Serial.printf("Core version        : %s\n"     , ESP.getCoreVersion().c_str() );
  Serial.printf("SDK  version        : %s\n"     , ESP.getSdkVersion() );
  Serial.printf("eeprom length       : %i\n"     , eepromLength );
}

void loop()
{
  if (addr % 20 == 0)
  {
    Serial.println();
    Serial.print(addr);
    Serial.print(" : ");
    delay(250);
  }

  int val = random(50);

  if (!readMode ) EEPROM.write(addr, val);

  Serial.print(EEPROM.read(addr));
  Serial.print("  ");

  addr = addr + 1;
  if (addr == eepromLength)
  {
    Serial.println("");
    if (readMode )
      Serial.println("starting Writing");
    else
    {
      EEPROM.commit();
      Serial.println("EEPROM Commited");
      delay(10000);
      Serial.println("starting reading");
    }
    readMode = !readMode;
    addr = 0;
  }
  delay(100);
}
```

### Debug Messages

```
Core version        : 2_5_2
SDK  version        : 2.2.1(cfd48f3)
eeprom length       : 100

0 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
20 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
40 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
60 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
80 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
starting Writing

0 : 27  24  19  10  37  12  9  12  42  39  42  0  6  3  16  39  18  42  16  29  
20 : 40  40  47  13  30  25  46  16  15  21  23  11  41  25  19  30  41  37  3  27  
40 : 46  18  16  35  15  22  15  48  20  23  1  14  19  39  49  6  45  38  4  17  
60 : 30  0  48  26  27  31  17  19  30  33  6  5  27  37  21  38  36  5  36  38  
80 : 6  20  3  17  37  40  43  21  18  49  41  43  43  23  10  3  40  46  41  44  
EEPROM Commited
starting reading

0 : 27  24  19  10  37  12  9  12  42  39  42  0  6  3  16  39  18  42  16  29  
20 : 40  40  47  13  30  25  46  16  15  21  23  11  41  25  19  ⸮⸮⸮5⸮⸮⸮
```
reboot arduino
```
Core version        : 2_5_2
SDK  version        : 2.2.1(cfd48f3)
eeprom length       : 100

0 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
20 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
40 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
60 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0 
```
recompile using 2.5.0
```
Core version        : 2_5_0
SDK  version        : 3.0.0-dev(c0f7b44)
eeprom length       : 100

0 : 77  69  67  86  0  78  101  116  119  111  114  107  0  0  0  0  0  0  0  0  
20 : 0  0  0  0  0  0  0  0  0  0  0  0  97  110  101  119  102  117  99  107  
40 : 101  100  107  101  121  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
60 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
80 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  119  108  101  100  
starting Writing

0 : 41  9  26  4  45  30  48  30  6  6  34  1  45  6  16  3  12  30  26  20  
20 : 37  14  18  9  34  31  36  47  23  27  48  17  2  16  43  34  17  7  49  38  
40 : 33  32  11  37  26  26  47  22  20  13  33  49  49  31  47  9  13  9  18  46  
60 : 44  8  21  34  23  10  27  48  38  46  46  29  19  34  17  35  39  17  21  36  
80 : 39  19  29  26  9  36  37  38  40  30  30  10  46  0  39  3  46  24  41  42  
EEPROM Commited
starting reading

```

earlephilhower, 2019-09-23:
Thanks for the detailed description and MCVE.

Please try with GIT head.  EEPROM just had a few changes applied to it which fixed some issues with its calculated address.

The test you gave shows the correct behavior during runtime and after a reboot with git head:
````
Core version        : f3ca0900
SDK  version        : 2.2.2-dev(38a443e)
eeprom length       : 100

0 : 255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  
20 : 255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  
40 : 255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  
60 : 255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  
80 : 255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  255  
starting Writing

0 : 12  45  9  12  47  47  5  39  2  3  17  17  31  31  15  19  28  6  48  12  
20 : 16  47  10  23  3  42  2  3  44  2  43  3  5  49  28  32  11  47  26  40  
40 : 15  23  39  8  27  9  43  34  24  19  30  14  49  39  16  28  45  19  43  44  
60 : 46  40  30  21  11  8  23  14  30  32  6  41  22  0  32  30  44  1  21  26  
80 : 1  1  8  31  13  24  40  23  30  44  29  34  36  34  16  16  46  7  1  37  
EEPROM Commited
starting reading

0 : 12  45  9  12  47  47  5  39  2  3  17  17  31  31  15  19  28  6  48  12  
20 : 16  47  10  23  3  42  2  3  44  2  43  3  5  49  28  32  11  47  26  40  
40 : 15  23  39  8  27  9  43  34  24  19  30  14  49  39  16  28  45  19  43  44  
60 : 46  40  30  21  11  8  23  14  30  32  6  41  22  0  32  30  44  1  21  26  
80 : 1  1  8  31  13  24  40  23  30  44  29  34  36  34  16  16  46  7  1  37  
starting Writing

0 : 17  29  44  48  38  27  8  37  27  32  7  44  22  31  22  35  28  46  28  48  
20 : 9  36  26  1  28  11  2  48  45  42  25  12  30  46  36  12  31  35  41  21  
40 : 15  49  19  18  8  11  30  15  33  42  38  21  48  14  4  44  29  1  22  5  
60 : 38  43  49  4  33  46  26  23  35  40  39  12  20  20  35  0  6  26  43  32  
80 : 4  34  41  6  1  6  3  28  41  29  23  13  43  14  5  20  25  39  15  8  
EEPROM Commited
starting reading

0 : 17  29  44  48  38  27  8  37  27  32  7  44  22  31  22  35  28  46  28  48  
20 : 9  36  26  1  28  11  2  48  45  42  25  12  30  46  36  12  31  35  41  21  
40 : 15  49  19  18  8  11  30  15  33  42  38  21  48  14  4  44  29  1  22  5  
60 : 38  43  49  4  33  46  26  23  35  40  39  12  20  20  35  0  6  26  43  32  
80 : 4  34  41  6  1  6  3  28  41  29  23  13  43  14  5  20  25  39  15  8  
starting Writing

0 : 23  28  25  3  10  39  42  31  24  47  41  5  4  27  14  37  45  19  20  18  
20 : 4  48  9  24  4  28  11  21  2  32  26  46  13  18  38  7  41  3  35  41  
40 : 14  25  5  7  27  4  29  22  24  18  16  44  40  39  42  23  44  3  18  48  
60 : 35  49  29  21  37  36  15  14  3  35  37  24  42  47  5  13  25  33  0  44  
80 : 3  1  37  21  45  17  1  8  19  28  2  24  21  36  3  22  1  18  42  8  
EEPROM Commited
starting reading

0 : 23  28  25  3  10  39  42  31  24  47  41  5  4  27  14  37  45  19  20  18  
20 : 4  48  9  24  4  28  11  21  2  32  26  46  13  18  38  7  41  3  35  41  
40 : 14  25  5  7  27  4  29  22  24  18  16  44  40  39  42  23  44  3  18  48  
60 : 35  49  29  21  37  36  15  14  3  35  37  24  42  47  5  13  25  33  0  44  
80 : 3  1  37  21  45  17  1  8  19  28  2  24  21  36  3  22  1  18  42  8  

...reboot....

Core version        : f3ca0900
SDK  version        : 2.2.2-dev(38a443e)
eeprom length       : 100

0 : 23  28  25  3  10  39  42  31  24  47  41  5  4  27  14  37  45  19  20  18  
20 : 4  48  9  24  4  28  11  21  2  32  26  46  13  18  38  7  41  3  35  41  
40 : 14  25  5  7  27  4  29  22  24  18  16  44  40  39  42  23  44  3  18  48  
60 : 35  49  29  21  37  36  15  14  3  35  37  24  42  47  5  13  25  33  0  44  
80 : 3  1  37  21  45  17  1  8  19  28  2  24  21  36  3  22  1  18  42  8  
starting Writing

0 : 49  4  16  49  24  44  21  9  25  4  33  15  4  9  9  22  23  9  48  16  
20 : 21  35  21  10  29  28  45  36  31  38  19  26  2  22  23  15  3  29  47  32  
40 : 47  24  32  18  14  38  28  12  2  26  3  11  25  12  41  12  15  29  11  39  
60 : 23  43  13  3  9  0  9  49  6  28  16  43  42  35  32  46  3  18  32  35  
80 : 28  3  42  37  35  47  26  7  1  1  12  6  42  2  14  2  12  35  14  18  
EEPROM Commited
starting reading

0 : 49  4  16  49  24  44  21  9  25  4  33  15  4  9  9  22  23  9  48  16  
20 : 21  35  21  10  29  28  45  36  31  38  19  26  2  22  23  15  3  29  47  32  
40 : 47  24  32  18  14  38  28  12  2  26  3  11  25  12  41  12  15  29  11  39  
60 : 23  43  13  3  9  0  9  49  6  28  16  43  42  35  32  46  3  18  32  35  
80 : 28  3  42  37  35  47  26  7  1  1  12  6  42  2  14  2  12  35  14  18  
````

earlephilhower, 2019-09-24:
I think you have some other problem, unfortunately.  In #6531 @liebman and others saw the EEPROM location was *moved* vs. prior versions, but it was working for them (once they rewrote it).

So it may be a HW problem with your specific board or a SW one that others don't seem to be having.

What is the exact board and options used in the IDE?  Can you enable full debug output and rerun?  Do SPIFFS examples work and preserve data after a reboot?

earlephilhower, 2019-09-24:
Also, what is the flash chip on your board?  There are known issues with PUYA flash requiring a special define in the code to make them work, since they're out of spec.  This might be the same thing.

A quickie test would be to make `eepromLength=4096` in your example.  The Puya problem was with partial writes, but 4K is a full write.

viknet365, 2019-09-24:
thank you Earle for following this issue.
Making eepromLength=4096 does not change anything. I still get 0 (Though I had to change some timing in my sketch to achive a timely 4096 write).
Please find below the compile and download option.
[image]

here are the full debug trace that might help :
````
SDK:2.2.2-dev(38a443e)/Core:2.5.2-154-gf3ca0900=20502154/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-8-g7958710/BearSSL:89454af

Core version        : f3ca0900
SDK  version        : 2.2.2-dev(38a443e)
eeprom length       : 100

0 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
20 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
40 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
60 : scandone
no codebeat found, reconnect after 1s
reconnect
0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
80 : 0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  0  
starting Writing

0 : 35  15  5  20  5  27  9  23  0  45  40  28  28  39  38  21  26  41  24  36  
20 : 48  41  0  25  4  12  46  11  28  29  33  37  10  48  40  17  14  21  26  14  
40 : 46  27  17  11  48  41  49  49  21  46  11  19  41  2  10  5  35  31  15  42  
60 : 28  7  29  26  31  21  23  23  46  29  26  16  26  17  38  12  18  32  23  3  
80 : 3  17  34  45  16  1  4  31  9  47  15  24  33  32  3  35  14  23  37  28  
EEPROM Commited
scandone
no codebeat found, reconnect after 1s
reconnect
scandone
no codebeat found, reconnect after 1s
reconnect
scandone
no codebeat found, reconnect after 1s
reconnect
starting reading

0 : scandone
no codebeat found, reconnect after 1s
reconnect
35  15  5  20  5  27  9  23  0  45  40  28  28  39  38  21  26  41  24  36  
20 : 48  41  0  25  4  12  46  11  28  29  33  37  10  48  40  17  14  21  26  14  
40 : 46  27  17  11  48  41  49  49  21  46  11  19  41  2  10  5  35  31  15  42  
60 : 28  7  29  26  31  21  23  23  46  29  26  16  26  17  38  12  18  32  23  3  
80 : 3  17  34  45  16  1  4  31  9  47  15  24  33  32  3  35  14  23  37  28  
starting Writing

````

earlephilhower, 2019-09-24:
Sorry, but I've the exact same setup (wemos d1 mini, 4MB, no spiffs, 80mhz) and not able to reproduce this with your test.  On reboot I get the random values written before the reboot, always.

Do you have any other EEPROM libs in your path?  Full verbose build output would help check.  What it looks like to me is that there's some simulated EEPROM that's just using a bit of RAM (hence the reset to 0) and never actually writing to your flash.  If the flash itself had issues, sketch upload would be problematic.

Unless someone else can hit it, there's nothing we can do.  The actual flash writing routines are from the SDK, not our code, and EEPROM itself is a very simple library.

devyte, 2019-09-24:
> wemos d1 r1

Is that correct for your board? Most Wemos D1 today are r2.

earlephilhower, 2019-09-24:
Yes, a typo.  It's `LOLIN(WEMOS) D1 R2 & mini`

earlephilhower, 2019-09-24:
@viknet365 , my mistake.  @devyte was asking you to double-check your board.  It should probably be `LOLIN(WEMOS) D1 R2 & mini` and not the `WeMos D1 R1`...

viknet365, 2019-09-24:
hello, it is this kind of board:
https://alexnld.com/product/3pcs-wemos-x-8266-esp-wroom-02-development-board-d1-mini-nodemcu-wifi-internet-of-things-esp8266-module/
I made a picture of it 
[image]
 I am really not sure I have an attached SPI flash attached to it. JFFS sample never work with 2.5.0 or 2.5.2

earlephilhower, 2019-09-24:
Well, there you go.  According to the DOCS, it only has 2MB, not 4MB.  Try fixing your flash size to 2MB and see if you're still out of luck...

...snip...
> ESP-WROOM-02 currently integrates a 2-MB SPI flash. ESP-WROOM-02 supports these SPI modes: Standard SPI, DIO (Dual I/O), DOUT (Dual Output), QIO (Quad I/O) and QOUT (Quad Output). 

Also, check the flash speed in `boards.txt` since you're on such a strange board.  

The docs imply the flash only supports the slowest, 26MHz mode.  Uploading with esptool.py may auto-slowdown and work, but flash ops on-chip might not if your menu selection is at 40MHz.
...snip...
> ESP-WROOM-02 uses a 26-MHz crystal oscillator. The accuracy of the crystal oscillator should be ±10 PPM. 

And boards.txt:
https://github.com/esp8266/Arduino/blob/244dbd7713338a127be3f53ba2fa1c3915c3ba98/boards.txt#L4160-L4162

viknet365, 2019-09-24:
Yes it is now working if I use a generic ESP8266 board with 2MB or less memory.
Thanks a lot and really sorry for bothering the dev team with this.

Because new implementation ( since 2.5.1)  is a bit more "sensitive" maybe some safeguards could be added (if possible) to check that memory defined really exist.

Sketch that give memory info such as the one here:
https://arduino.stackexchange.com/questions/49096/esp8266ex-does-not-hold-written-eeprom-data-what-could-be-wrong?noredirect=1&lq=1
did not seems to clearly detect the mismatch.

Maybe one line in the github release explaining that mismatched memory option will no longer work could also be usefull.

PR #6556 (2019-09-26): Add EEPROM debug printouts, error check to example
Add debug printouts when EEPROM calls fail, even if the API doesn't
allow returning a success/failure code.

Adds error checking to the example to make it explicit that when you
call EEPROM::commit(), you need to look at the result code in your code.

Fixes #6551 , or would fix it if there was error checking in the MCVE.
commit 2b9fcdb5 (2019-09-26): Add EEPROM debug printouts, error check to example (#6556)

* Add EEPROM debug printouts, error check to example

Add debug printouts when EEPROM calls fail, even if the API doesn't
allow returning a success/failure code.

Adds error checking to the example to make it explicit that when you
call EEPROM::commit(), you need to look at the result code in your code.

Fixes #6551 , or would fix it if there was error checking in the MCVE.

* Clarify example error message
closed 2019-09-26 by devyte