esp8266/Arduino #6562: LWIP2 build broken due to ets_putc, ets_printf changes

earlephilhower, 2019-09-28:
Looks like recent ets_vprintf and ets_putc signature changes in the main repo (to fix UMM performance dumps, I think), make the LWIP2 build fail.

````
earle@server:~/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2$ make
make -C builder -f Makefile.arduino all
make[1]: Entering directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
for v6 in 0 1; do \
for mss in 536 1460; do \
for feat in 0 1; do \
	test \( $feat = 0 \) -a \( $v6 = 1 \) && continue; \
	featname= ;\
	test $feat = 1 && featname=-feat ;\
	make -f makefiles/Makefile.build-lwip2 \
		target=arduino \
		DEFINE_TARGET=ARDUINO \
		SDK=../.. \
		LWIP_ESP=../../lwip/include \
		LWIP_LIB=liblwip$((2+4*v6))-$mss$featname.a \
		LWIP_LIB_RELEASE=../../lib/liblwip$((2+4*v6))-$mss$featname.a \
		LWIP_INCLUDES_RELEASE=../include \
		TOOLS=../../../xtensa-lx106-elf/bin/xtensa-lx106-elf- \
		TCP_MSS=$mss LWIP_FEATURES=$feat LWIP_IPV6=$v6 \
		BUILD=build-$mss$featname-v$((4+2*v6)) \
		all ; \
done; done; done
make[2]: Entering directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
( \
echo "// script-generated, extracted from espressif SDK's lwIP arch/cc.h"; \
echo "#define LWIP_NO_STDINT_H 1"; \
grep -e LWIP_ERR_T -e ^typedef ../../lwip/include/arch/cc.h \
) > glue-lwip/lwip-err-t.h
makefiles/make-lwip-hash
make -f makefiles/Makefile.glue-esp BUILD=build-536-v4 V=0
make[3]: Entering directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
C  build-536-v4/glue-esp/lwip-esp.o
In file included from glue-esp/lwip-esp.c:46:0:
glue/esp-missing.h:30:5: error: conflicting types for 'ets_vprintf'
 int ets_vprintf (int (*print_function)(int), const char * format, va_list arg) __attribute__ ((format (printf, 2, 0)));
     ^
In file included from glue/glue.h:51:0,
                 from glue-esp/lwip-esp.c:35:
../../include/ets_sys.h:226:5: note: previous declaration of 'ets_vprintf' was here
 int ets_vprintf(fp_putc_t print_function, const char * format, va_list arg) __attribute__ ((format (printf, 2, 0)));
     ^
In file included from glue-esp/lwip-esp.c:46:0:
glue/esp-missing.h:32:5: error: conflicting types for 'ets_putc'
 int ets_putc(int);
     ^
In file included from glue/glue.h:51:0,
                 from glue-esp/lwip-esp.c:35:
../../include/ets_sys.h:237:6: note: previous declaration of 'ets_putc' was here
 void ets_putc(char);
      ^
makefiles/Makefile.rules:7: recipe for target 'build-536-v4/glue-esp/lwip-esp.o' failed
make[3]: *** [build-536-v4/glue-esp/lwip-esp.o] Error 1
make[3]: Leaving directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
makefiles/Makefile.build-lwip2:47: recipe for target 'liblwip2-536.a' failed
make[2]: *** [liblwip2-536.a] Error 2
make[2]: Leaving directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make[2]: Entering directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
make -f makefiles/Makefile.glue-esp BUILD=build-536-feat-v4 V=0
make[3]: Entering directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
C  build-536-feat-v4/glue-esp/lwip-esp.o
In file included from glue-esp/lwip-esp.c:46:0:
glue/esp-missing.h:30:5: error: conflicting types for 'ets_vprintf'
 int ets_vprintf (int (*print_function)(int), const char * format, va_list arg) __attribute__ ((format (printf, 2, 0)));
     ^
In file included from glue/glue.h:51:0,
                 from glue-esp/lwip-esp.c:35:
../../include/ets_sys.h:226:5: note: previous declaration of 'ets_vprintf' was here
[... 110 lines cut ...]
                 from glue-esp/lwip-esp.c:35:
../../include/ets_sys.h:226:5: note: previous declaration of 'ets_vprintf' was here
 int ets_vprintf(fp_putc_t print_function, const char * format, va_list arg) __attribute__ ((format (printf, 2, 0)));
     ^
In file included from glue-esp/lwip-esp.c:46:0:
glue/esp-missing.h:32:5: error: conflicting types for 'ets_putc'
 int ets_putc(int);
     ^
In file included from glue/glue.h:51:0,
                 from glue-esp/lwip-esp.c:35:
../../include/ets_sys.h:237:6: note: previous declaration of 'ets_putc' was here
 void ets_putc(char);
      ^
makefiles/Makefile.rules:7: recipe for target 'build-1460-feat-v6/glue-esp/lwip-esp.o' failed
make[3]: *** [build-1460-feat-v6/glue-esp/lwip-esp.o] Error 1
make[3]: Leaving directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
makefiles/Makefile.build-lwip2:47: recipe for target 'liblwip6-1460-feat.a' failed
make[2]: *** [liblwip6-1460-feat.a] Error 2
make[2]: Leaving directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
Makefile.arduino:3: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/earle/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip2/builder'
Makefile:3: recipe for target 'all' failed
make: *** [all] Error 2
````

mhightower83, 2019-09-28:
@d-a-v Sorry, I thought I would try and clean up an existing issue and resolve mine at the same time. I was trying to resolve the need for casting to (void *) in so many places, as was noted here:
https://github.com/esp8266/Arduino/blob/3cc64f7877c740df8f0c8915bc16c2655c6310e7/cores/esp8266/core_esp8266_postmortem.cpp#L111
I was a little afraid, I might be opening a can of worms. :slightly_frowning_face: I forgot about the lwIP2 build. I guess this is why it hasn't been done before now.

d-a-v, 2019-09-28:
No worry at all. I'll update.

PR #6373 (2019-09-29): time: import IANA timezone definitions, expose SNTP API
- `configTime("timezone", "ntp servers...")` added
- timezone definitions by country/cities (TZ.h)
- script to update timezone definitions
- updated example
- sntp additions (see below)

Closes #4637
Closes #5681
Fixes #6562
closed 2019-09-29 by devyte